rickysarraf/laptop-mode-tools #42: Allow user defined way to detect "ON_AC"

ikus060, 2015-04-02:
current implementation doesn't allow the user to define a special way to detect if the "computer" is running on battery or not. I intent to use `laptop-mode` when my servers is running on UPS battery. I need a simple hook in the script to tell if the server is on AC or on BATTERY.

I'm suggesting something very simple like a file or executable file containing or returning "ac" or "battery". But I'm open to suggestion.

ikus060, 2015-04-02:
I don't ave the change in place. I'm suggesting a new feature. I may have some free time to develop something rapidly.

When you are talking about module what are you expecting ? a new directory /etc/laptop-mode/modules.d/*.sh containing bash script ? The module are "called" or "sourced" from /usr/bin/laptop_mode ? 

May you provide a bit more detail about the "module" approach so when I start writing something it will be closer to what you expect ?

rickysarraf, 2015-04-02:
On Thursday 02 April 2015 05:19 PM, Patrik Dufresne wrote:
> I don't ave the change in place. I'm suggesting a new feature. I may
> have some free time to develop something rapidly.
>
> When you are talking about module what are you expecting ? a new
> directory [...]
You can have a look at the current list of modules here:
https://github.com/rickysarraf/laptop-mode-tools/tree/lmt-upstream/usr/share/laptop-mode-tools/modules

It is my opinion that you should do it as a module. Please look at the
current set of modules, on how it should be done.

If introducing this module requires some changes in the main script,
/usr/sbin/laptop_mode, I'd be more than willing to review and accept those.

All modules are sourced by the main script /usr/sbin/laptop_mode. In
line 1106, you can see the code that does the sourcing of the modules.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

rickysarraf, 2015-06-28:
Were you able to get through it ?

ikus060, 2015-06-28:
Nope. The modification was too intrusive for me.

ikus060, 2016-03-10:
Any update. It is possible to use laptop-mode for server ? Maybe I can force the "mode" to "on battery" vs "on ac".

rickysarraf, 2016-03-10:
Simply running it on a server is easy. Just set the following, and you should be able to run it as an "always on" setting.

```
#
# Enable laptop mode power savings, even when on AC power.
# This is useful when running as a headless machine, in low power mode
#
ENABLE_LAPTOP_MODE_ON_AC=0

```

Does that help ? It still is not what was requested in this report.

rickysarraf, 2016-03-10:
Actually, I thought a little more about it. I've had a similar use case recently.

The problem is, when the machine goes on UPS, there needs to be a mechanism for the server OS to know that. In the past, I've done that with NUTS. But now, I have a APC UPS and I haven't really investigated if my model has the necessary interface. There, though, is a project dedicated to APC UPS, http://www.apcupsd.org/

Now, based on implementation, this will vary. IIRC, back then when using NUT, there was a daemon listening for events from the UPS. The UPS would also give charge information. And then the user could specify a policy.

If the same is true with the implementations today (sorry, I really need to look at the rear of my UPS to see if it has any interface), all we need is the monitoring daemon to invoke Laptop Mode Tools.

And for that scenario, I can think of a new setting, like 

```
ENABLE_LMT_ON_UPS=1
```

and then an evaluation like:

```
if [ $ENABLE_LMT_ON_UPS = 1 ]; then
   # For forcing UPS use case
    ON_AC=0
fi
```

This setting aside, we still need someone to invoke LMT. And that is why a UPS Daemon will be needed. Because the Host kernel is completely unaware of these power state changes.

rickysarraf, 2016-03-10:
BTW, the NUT project is still active. http://networkupstools.org/

ikus060, 2016-03-10:
In fact, I'm running NUT to monitor my UPSs. So your idea look nice.

From the deamon (upsmon), how would I call laptop-mode to tell him it's running on UPS. Also when power get back, how to restore "previous" value ?

rickysarraf, 2016-03-10:
On Thu, 2016-03-10 at 04:35 -0800, Patrik Dufresne wrote:
> In fact, I'm running NUT to monitor my UPSs. So your idea look nice.
> From the deamon (upsmon), how would I call laptop-mode to tell him
> it's running on UPS. Also when power get back, how to restore
> "previous" value ?

We still have a problem, the server is constantly ON_AC. Does NUT
integrate into Linux's power_supply subsystem ?

I think before we explore our brute force approach, I need you to
provide me some details.

1) ls /sys/class/power_supply/

I'm interested if there is anything interesting when running on UPS.

2) sudo udevadm monitor

Run the command as root and then switch your server to UPS supply and
then back to AC. See if this guy reports any message.

If the above 2 do not work, then we'll explore the brute approach.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

ikus060, 2016-03-10:
Nothing in ` ls /sys/class/power_supply/`.

Nothing displayed in `sudo udevadm monitor` when switching to battery or to ac.

But I see a "Broadcast Message":
```
Broadcast Message from nut@fente                                               
        (somewhere) at 11:41 ...                                               
                                                                               
UPS myups@localhost on battery                                                 
                                                                               
                                         
Broadcast Message from nut@fente                                               
        (somewhere) at 11:42 ...                                               
                                                                               
UPS myups@localhost on line power     
```
I'm not sure if you can leverage this.

rickysarraf, 2016-03-10:
On Thu, 2016-03-10 at 08:46 -0800, Patrik Dufresne wrote:
> Nothing in ls /sys/class/power_supply/.
> Nothing displayed in sudo udevadm monitor when switching to battery
> or to ac.
> But I see a "Broadcast Message":
> Broadcast Message from [...]

Nope. That won't help.

So the server OS (kernel) has no information about the power state. The
only entity that has access to that information is NUT.

In Laptop Mode Tools, I'm thinking of adding a new config option for
UPS. Something like ENABLE_LAPTOP_MODE_ON_UPS

Like I had mentioned earlier, the purpose of this switch would be to
instruct LMT that ON_AC=1 should be overridden....

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

rickysarraf, 2016-03-11:
On Thu, 2016-03-10 at 22:52 +0530, Ritesh Raj Sarraf wrote:
> Nope. That won't help.
>
> So the server OS (kernel) has no information about the power state.
> The
> only entity that has access to that information is NUT.
>
>
> In Laptop Mode Tools, I'm thinking of adding [...]

Attached is a patch that you can test. Based on feedback, I'll see if
this should be added to the repository.

The core assumption here is that someone else, like NUT, should write
the exact expected value into the file. LMT's polling daemon, when
invoked, will read that value and then accordingly override the state.

NOte: This is not yet in the repo but in my local branch. Inclusion
depends on testing results.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

ikus060, 2016-03-11:
"attached" ? I don't see any attachment ?!

rickysarraf, 2016-03-12:
I don't why github has a half baked (email) interface.

And then there is this stupid error when attaching a patch: **"Unfortunately, we don’t support that file type.  Choose Files Try again with a PNG, GIF, JPG, DOCX, PPTX, XLSX, TXT, PDF, or ZIP."**

[0001-Add-new-UPS-module-to-read-power-state-from-3rd-part.patch.txt](https://github.com/rickysarraf/laptop-mode-tools/files/170383/0001-Add-new-UPS-module-to-read-power-state-from-3rd-part.patch.txt)

rickysarraf, 2016-03-12:
@ikus060 You'll need your UPS monitoring tool to write the state into /run/laptop-mode-tools/ups_state

The states that this module will understand are: "BATTERY" and "POWER".

rickysarraf, 2016-03-23:
@ikus060 I did not hear back from you. Did you try the patch ?
FYI: The same logic works for me. But I'm using it on a different setup not involving LMT

rickysarraf, 2016-05-29:
I did not hear back from you. And given that this has less to do with LMT, I'm closing it for now. If you have interesting thought on this, feel free to ping back.

closed 2016-05-29 by rickysarraf