rickysarraf/laptop-mode-tools #44: PowerTop reports more power savings

rickysarraf, 2015-07-03:
This bug report is only for the record, so that if any user files, they know that it is NOT a problem with Laptop Mode Tools, but rather with PowerTop itself.

The wireless report is correct because I've disabled it for my WIFI card.

But if you have powertop reporting of 'VM Writeback Timeout' or 'Audio Codec Power Management' (assuming you have the very common Intel HD Audio chipset), please do not file a bug report. Those reports are incorrect

[image]

ManuGithubSteam, 2018-02-09:
Thanks for the Info!
May i ask, why they are incorrect ? 
Thanks!

rickysarraf, 2018-02-10:
powertop has a very weird project page, with no tracker for bugs. IIRC, I had reported both the issues. Maybe it was just plain email. I don't know.

Here's a Debian report for the sound issue: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=691594

rickysarraf, 2018-02-10:
And for the VM Writeback Timout, see my comment on the same bug: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=691594#13