rickysarraf/laptop-mode-tools #47: systemd: cannot remove /var/run/laptop-mode-tools/enabled

bretter, 2015-09-09:
Since upgrading to version 1.68, I have been getting the following message as my computer is booting:

    cannot remove /var/run/laptop-mode/tools/enabled: No such file or directory

This is the status of laptop-mode.service as reported by systemd

    ● laptop-mode.service - Laptop Mode Tools
       Loaded: loaded (/usr/lib/systemd/system/laptop-mode.service; enabled; vendor preset: disabled)
       Active: active (exited) since Tue 2015-09-08 23:30:40 EDT; 28s ago
         Docs: man:laptop_mode(8)
               man:laptop-mode.conf(8)
               http://samwel.tk/laptop_mode
      Process: 463 ExecStopPost=-f /var/run/laptop-mode-tools/enabled (code=exited, status=1/FAILURE)
      Process: 469 ExecStart=init auto (code=exited, status=0/SUCCESS)
     Main PID: 469 (code=exited, status=0/SUCCESS)
       CGroup: /system.slice/laptop-mode.service

    Sep 08 23:30:39 morton systemd[1]: Starting Laptop Mode Tools...
    Sep 08 23:30:40 morton systemd[1]: Started Laptop Mode Tools.

rickysarraf, 2015-09-09:
That is interesting. What distribution are you on ?

Can you please provide some more logs? From what you've shared so far,
it looks like LMT was started successfully. And if it does start
successfully, then that file should be created.

That said, there was a last minute bug, that hit 1.68, but is fixed in
my git tree. Perhaps you may be hit by that one.

bretter, 2015-09-09:
I am on Arch Linux. Running systmed v 225.

Also, I there seem to be no errors when powering on under battery power.

I'll do some testing when I get back to a wall socket but if I remember correctly, it seemed that powering on under AC and then unplugging did not enable laptop-mode (running <<<sudo laptop_mode>>> returned without issuing '[unchanged]'

Let me know if there are other tests you'd like me to run to help narrow down the problem.

rickysarraf, 2015-09-09:
The [changed/unchanged] return is for the state of LMT. So, if you see that, it'd mean LMT was indeed started.

The only bug that crept in into 1.68 was specific to systemd. Under systemd, you can have different targets for ExecStart, ExecStop, ExecReload. In the very last moment, I dropped ExecReload (now reverted).

Now if you look at the main invoking udev component of LMT, it checks systemd and calls reload-or-restart. With reload absent in 1.68, it'd mean Restart, i.e. Stop/Start. Which is what you may have hit.
But my explanation is still theoretical. You need to verify it on your box.

rickysarraf, 2015-09-09:
If possible, can you try the current git tree?

rickysarraf, 2016-02-15:
This is been long with no more updates. Closing. Please feel free to re-open if you have more info.

closed 2016-02-15 by rickysarraf