rickysarraf/laptop-mode-tools #48: 1.68 restarts continuously on Arch with systemd

dluciv, 2015-09-10:
1.68 is restarting continuously on Arch Linux x64.

When rolling back to 1.67, it works ok. Users report 1.68 works ok on Gentoo, so possibly systemd integration issue.

Also when booting, prints "Stopping Laptop Mode Tools" (or something like it) several times.

I can provide more info if required, but it is better to know, what logging to turn on exactly.

rickysarraf, 2015-09-10:
Hello Dmitry @dluciv

This bug crept in because of a change in the very last minute of the 1.68 release. The change has already been reverted in the git tree.  8e19ce2cb16c28cc4505b6da4676d42565296983

The change is very minor, so I have not done a new release. Can you cherry pick it ?

dluciv, 2015-09-10:
Hello @rickysarraf
Thank you very much for your fast feedback!
No problem for me to pick it, it will likey take me about a day to get to my laptop and ensure it helps. So I will close the issue then.

dluciv, 2015-09-10:
Yes, https://github.com/rickysarraf/laptop-mode-tools/commit/8e19ce2cb16c28cc4505b6da4676d42565296983 helps. Thank you again!

closed 2015-09-10 by dluciv

dluciv, 2015-09-23:
After some time, another oddity happened.
It does not restart any more, but it does print following to log constantly (with period of 1 to few seconds):

    Laptop mode
    enabled, active [unchanged]

Nothing bad (at least easily visible) happens then, so I do not want to open new issue. But it irritates (regular unconditional garbage in all system ttys...), and also I am afraid it can potentially cause faster battery drain...

rickysarraf, 2015-09-23:
On Wed, 2015-09-23 at 00:26 -0700, Dmitry V. Luciv wrote:
> After some time, another oddity happened.
> It does not restart any more, but it does print following to log
> constantly (with period of 1 to few seconds):
> Laptop mode
> enabled, active [unchanged] [...]

Hello Dmitry,

Thank you for reporting this one too.

I think I know the problem you may be running into.

If I understand correct, you are seeing frequent messages on system
console. Right ?

If so, this has been fixed in fdd6b5d6a046e52453fea6215d383a1e1712f968

Can you please verify with the current git HEAD ?
I think if you can confirm the fix, I will do a release so that all
users/distributions can upgrade to it, rather than cherry picking
individual fixes.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

dluciv, 2015-09-23:
Hello, Ritesh!

Yes, it puts those messages to system log frequently so they appear on system console.

I will try to check it in about a day, then I will report how it goes.

dluciv, 2015-09-23:
BTW https://github.com/rickysarraf/laptop-mode-tools/commit/fdd6b5d6a046e52453fea6215d383a1e1712f968 looks like as laptop-mode-tools should post all this this to journal. Shold it really do it so frequently (once per several seconds)?

rickysarraf, 2015-09-24:
On Wed, 2015-09-23 at 08:44 -0700, Dmitry V. Luciv wrote:
> BTW fdd6b5d looks like as laptop-mode-tools should post all this this
> to journal. Shold it really do it so frequently (once per several
> seconds)?

It should not be very frequent

With the new systemd based polling feature, that'll poll every 150
seconds. That's when you'll see the messages in the journal.

If you are seeing anything more frequent, it could be specific to your
machine. Your machine may be generating spurious events.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

dluciv, 2015-09-25:
Yes, it now writes to systemd journal. And not so frequently as before. Do not know, what changed, may be my laptop was afraid of something that time... =)

HEAD works good overall, at least on my machine.

rickysarraf, 2015-09-26:
Great. Thanks Dmitry. So I'll do a minor upstream release to help all
downtream pick it up. I'll close this bug when I've completed the
release.

On Fri, 2015-09-25 at 01:28 -0700, Dmitry V. Luciv wrote:
> Yes, it now writes to systemd journal. And not so frequently as
> before. Do not know, what changed, may be my laptop was afraid of
> something that time... =)
> HEAD works good overall, at least on my machine.
-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."