rickysarraf/laptop-mode-tools #49: Does not start if powering on from battery power.

robjloranger, 2015-11-04:
I can't get lmt to take effect if i boot cold on battery power. Only if I {boot with ac / plug in the ac} and then remove it. Is this a bug or the desired effect? What can I do to change it? Thanks.

rickysarraf, 2015-11-05:
On Wed, 2015-11-04 at 06:11 -0800, rob wrote:
> I can't get lmt to take effect if i boot cold on battery power. Only
> if I {boot with ac / plug in the ac} and then remove it. Is this a
> bug or the desired effect? What can I do to change it? Thanks.

No. It should be able to sense the state and act accordingly.

What distribution are you on ? And what version of LMT ?

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

robjloranger, 2015-11-05:
I'm running Manjaro and LMT v1.66-1.
I'll start with testing from the latest source to see if that helps.

robjloranger, 2015-11-05:
Just tried installing and get a few errors, presumably from using with openrc.
1. Say's the service laptop mode does not exist
2. ACPI/APM/PMU support was not installed, yet ACPID is installed
3. Last complains it can't find either update-rc.d or chkconfig. I know i use rc-update not update-rc.d

Any ideas where to look from here?

rickysarraf, 2015-11-05:
On Thu, 2015-11-05 at 05:54 -0800, rob wrote:
> Just tried installing and get a few errors, presumably from using
> with openrc.

It will help a lot more, if you can copy/paste the entire screen log.
That said, you might be the 1st person testing it with openrc.

When maintaining openrc in Debian, I recollect that openrc had some
issues with sysvinit compatibility too. So, your first bet would be to
boot with sysvinit and see if the problem is reproducible.

> 1. Say's the service laptop mode does not exist

Yes. Because what we provide from upstream today, is sysvinit and
systemd support. If you are well versed with openrc and are willing to
integrate it, I'll be okay to accept a patch.

> 2. ACPI/APM/PMU support was not installed, yet ACPID is installed

Again. If the console log was there, it'd have been easier to
determine.

> 3. Last complains it can't find either update-rc.d or chkconfig. I
> know i use rc-update not update-rc.d

So this is understood, with what I mentioned above.

> Any ideas where to look from here?

The first best start is to boot into sysv init and try.

Also, LMT is event aware. So in reality, it starts the moment the linux
kernel boot and udev comes into effect. It is just that you don't see a
nice system service message.

To verify if it is running, you can check for lmt under /var/run/

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

rickysarraf, 2015-11-05:
On Thu, 2015-11-05 at 05:46 -0800, rob wrote:
> I'm running Manjaro and LMT v1.66-1.
> I'll start with testing from the latest source to see if that helps.

And I'd suggest you stick with the RPM package that we provide. I'm
assuming you are using that, and not the tarball.

You mentioned initially that LMT was working fine while of AC, and only
inactive when booting on battery.

If so, can you please provide your boot logs when you've booted on
battery ?

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

robjloranger, 2015-11-05:
OK, I'm just on my way to work now but this evening I'll reinstall the
package manjaro provides and post the relevant logs. Thanks.

rickysarraf, 2015-11-05:
On Thu, 2015-11-05 at 06:38 -0800, rob wrote:
> OK, I'm just on my way to work now but this evening I'll reinstall
> the
> package manjaro provides and post the relevant logs. Thanks.

Or you may want to ask your distribution to upgrade.

http://www.researchut.com/blog/laptop-mode-tools-1_68_1

I hadn't realized (as the upstream developer for it) that the latest
version is 1.68.1

:-)

You can also try latest RPM packages, but they are only built for
opensuse and Fedora.
http://samwel.tk/laptop_mode/packages

If you are familiar with RPM package building, then you can also build 
it yourself. We ship an up-to-date .spec file in the
tarball.

Ritesh

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

robjloranger, 2015-11-06:
Ok so everything is back as it was, installed via manjaro repo. Which log files are you looking for? I checked /proc/sys/vm/laptop-mode and it was 0 after boot, so not starting up for some reason.

rickysarraf, 2015-11-06:
On Thu, 2015-11-05 at 17:48 -0800, rob wrote:
> Ok so everything is back as it was, installed via manjaro repo. Which
> log files are you looking for? I checked /proc/sys/vm/laptop-mode and
> it was 0 after boot, so not starting up for some reason.

The entire system log. Right from boot all the way down.
-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

robjloranger, 2015-11-07:
sorry i'm not sure which log to provide. there is no system log in /var/log, many others but no system or boot.

rickysarraf, 2015-11-07:
On Fri, 2015-11-06 at 17:32 -0800, rob wrote:
> sorry i'm not sure which log to provide. there is no system log in
> /var/log, many others but no system or boot.

What is the output of command `ls -l /var/log/`

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

robjloranger, 2015-11-07:
drwxr-xr-x 2 clamav clamav     4096 03.11.2015 16:53 clamav/
drwxr-xr-x 2 root   root       4096 17.10.2015 16:45 ConsoleKit/
drwxr-xr-x 2 root   root       4096 17.10.2015 16:45 cups/
drwxr-xr-x 2 root   root       4096 31.03.2015 12:04 gssproxy/
drwx--x--x 2 root   lightdm    4096 07.11.2015 08:55 lightdm/
drwxr-xr-x 2 root   root       4096 30.09.2015 13:43 old/
drwxr-xr-x 2 root   root       4096 19.09.2015 14:10 speech-dispatcher/
-rw-r----- 1 root   log       32803 07.11.2015 08:55 acpid.log
-rw-r----- 1 root   log      624985 07.11.2015 08:55 auth.log
-rw-r----- 1 root   log       46894 07.11.2015 08:55 crond.log
-rw-r----- 1 root   log     1741117 07.11.2015 08:56 daemon.log
-rw-r----- 1 root   root      66733 07.11.2015 08:55 dmesg
-rw-r----- 1 root   log      365116 07.11.2015 08:55 errors.log
-rw-r----- 1 root   log     4419746 07.11.2015 08:56 everything.log
-rw------- 1 root   root      32032 07.11.2015 08:55 faillog
-rw-r--r-- 1 root   root         85 18.10.2015 17:49 gufw.log
-rw-r----- 1 root   log        9359 18.10.2015 17:56 iptables.log
-rw-r----- 1 root   log     1592708 07.11.2015 08:55 kernel.log
-rw-r--r-- 1 root   root      18980 03.11.2015 16:48 lastlog
-rw-r----- 1 root   log     3800558 07.11.2015 08:56 messages.log
-rw-r----- 1 root   log      126202 07.11.2015 08:55 news.log
-rw-r--r-- 1 root   root     175886 05.11.2015 20:22 pacman.log
-rw-r--r-- 1 root   root      95242 05.11.2015 17:26 pamac.log
-rw-r--r-- 1 root   root      91426 05.11.2015 18:05 pm-powersave.log
-rw-r--r-- 1 root   root     246594 05.11.2015 18:05 pm-suspend.log
-rw-r--r-- 1 root   root     287975 07.11.2015 08:55 rc.log
-rw-r----- 1 root   log       12449 07.11.2015 08:55 syslog.log
-rw-r----- 1 root   log      890797 06.11.2015 17:32 user.log
-rw-rw-r-- 1 root   utmp    1032576 07.11.2015 08:55 wtmp
-rw-r--r-- 1 xmms2  xmms2         0 17.10.2015 16:45 xmms2d.log
-rw-r--r-- 1 root   root      28632 07.11.2015 08:55 Xorg.0.log
-rw-r--r-- 1 root   root      29342 06.11.2015 17:32 Xorg.0.log.old

rickysarraf, 2015-11-08:
On Sat, 2015-11-07 at 08:57 -0800, rob wrote:
> -rw-r--r-- 1 root root 287975 07.11.2015 08:55 rc.log
> -rw-r----- 1 root log 12449 07.11.2015 08:55 syslog.log

It should be these 2 files.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

robjloranger, 2015-11-08:
***rc***
```
rc boot logging started at Sat Nov  7 08:55:44 2015

 * Setting system clock using the hardware clock [UTC] ...
 [ ok ]
 * Autoloaded 0 module(s)
 * Checking local filesystems  ...
/dev/sda2: clean, 367/65280 files, 56994/261120 blocks
 [ ok ]
 * Remounting filesystems ...
 [ ok ]
 * Updating /etc/mtab ...
 * Creating mtab symbolic link
 [ ok ]
 * Activating swap devices ...
 [ ok ]
 * Mounting local filesystems ...
 [ ok ]
 * Mounting misc binary format filesystem ...
 [ ok ]
 * Loading custom binary format handlers ...
 [ ok ]
 * Configuring kernel parameters ...
 [ ok ]
 * Creating user login records ...
 [ ok ]
 * Setting hostname to openheart ...
 [ ok ]
 * Setting terminal encoding [UTF-8] ...
 [ ok ]
 * Setting keyboard mode [UTF-8] ...
 [ ok ]
 * Loading key mappings [us] ...
 [ ok ]
 * Bringing up network interface lo ...
 [ ok ]
 * Bringing up interface lo
 *   127.0.0.1/8 ...
 [ ok ]
 *   Adding routes
 *     127.0.0.0/8 via 127.0.0.1 ...
 [ ok ]
 * Activating additional swap space ...
 [ ok ]
 * Setting up tmpfiles.d entries ...
 [ ok ]
 * Initializing random number generator ...
 [ ok ]

rc boot logging stopped at Sat Nov  7 08:55:46 2015

rc default logging started at Sat Nov  7 08:55:46 2015

 * Starting D-BUS system messagebus ...
 [ ok ]
 * Starting NetworkManager ...
 [ ok ]
Connecting.............      0sConnecting..............     0sConnecting...............    0sConnecting...............    0s
 * Marking NetworkManager as inactive. It will automatically be marked
 * as started after a network connection has been established.
 * WARNING: NetworkManager has started, but is inactive
 * Checking your configfile (/etc/syslog-ng/syslog-ng.conf) ...
 [ ok ]
 * Starting syslog-ng ...
 [ ok ]
 * Starting acpid ...
 [ ok ]
 * Starting bluetooth ...
 [ ok ]
 * Starting consolekit ...
 [ ok ]
 * Starting cronie ...
 [ ok ]
 * Starting avahi-daemon ...
 [ ok ]
 * Starting cupsd ...
 [ ok ]
 * WARNING: netmount will start when NetworkManager has started
 * Starting thermald ...
 [ ok ]
 * Setting up lightdm ...
 [ ok ]
 * Starting local ...
 [ ok ]

rc default logging stopped at Sat Nov  7 08:55:48 2015
```

***syslog***
```
Nov  7 08:55:48 openheart syslog-ng[1973]: syslog-ng starting up; version='3.6.3'
```

I think syslog is just the log for syslog-ng, which is why i skipped it previously. Hopefully rc.log is helpful.

robjloranger, 2015-11-08:
Here you can see laptop-mode is in the default runlevel.

```
rc-update show
   NetworkManager |      default                 
                     acpid |      default                 
                    binfmt | boot                         
               bluetooth |      default                 
                bootmisc | boot                         
              consolekit |      default                 
                    cronie |      default                 
                    cupsd |      default                 
                      dbus |      default                 
                     devfs |                       sysinit
                   dmesg |                       sysinit
                       fsck | boot                         
              hostname | boot                         
                 hwclock | boot                         
               keymaps | boot                         
                killprocs |              shutdown        
kmod-static-nodes |                       sysinit
         laptop-mode |      default                 
                     local |      default                 
           localmount | boot                         
              loopback | boot                         
               modules | boot                         
              mount-ro |              shutdown        
                    mtab | boot                         
                   net.lo | boot                         
             netmount |      default                 
                  procfs | boot                         
                     root | boot                         
          savecache |              shutdown        
                   swap | boot                         
             swapfiles | boot                         
                  sysctl | boot                         
                   sysfs |                       sysinit
            syslog-ng |      default                 
      termencoding | boot                         
              thermald |      default                 
         tmpfiles.dev |                       sysinit
      tmpfiles.setup | boot                         
                    udev |                       sysinit
         udev-trigger |                       sysinit
              urandom | boot                         
                      xdm |      default  
```

rickysarraf, 2015-11-09:
WHat does `/etc/init.d/laptop-mode status`  say ?

robjloranger, 2015-11-09:
Laptop mode status:

Mounts:
   /dev/sda4 on / type ext4 (rw,relatime,data=ordered)
   /dev/sda2 on /boot type ext2 (rw,noatime)

Drive power status:

   /dev/sda:
    drive state is:  active/idle

(NOTE: drive settings affected by Laptop Mode cannot be retrieved.)

Readahead states:
   /dev/sda4: 128 kB
   /dev/sda2: 128 kB

Laptop Mode Tools is allowed to run: /var/run/laptop-mode-tools/enabled
exists.

/proc/sys/vm/laptop_mode:
   0

/proc/sys/vm/dirty_ratio:
   20

/proc/sys/vm/dirty_background_ratio:
   10

/proc/sys/vm/dirty_expire_centisecs:
   3000

/proc/sys/vm/dirty_writeback_centisecs:
   500

/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq:
   1600000

/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq:
   2001000

/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq:
   800000

/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_cur_freq:
   1600000

/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_max_freq:
   2001000

/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_min_freq:
   800000

/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor:
   ondemand

/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor:
   ondemand

/proc/acpi/button/lid/LID/state:
   state:      open

/sys/class/power_supply/AC/online:
   0

Mounts:
   /dev/sda4 on / type ext4 (rw,relatime,data=ordered)
   /dev/sda2 on /boot type ext2 (rw,noatime)

Drive power status:

   /dev/sda:
    drive state is:  active/idle

(NOTE: drive settings affected by Laptop Mode cannot be retrieved.)

Readahead states:
   /dev/sda4: 128 kB
   /dev/sda2: 128 kB

Laptop Mode Tools is allowed to run: /var/run/laptop-mode-tools/enabled
exists.

/proc/sys/vm/laptop_mode:
   0

/proc/sys/vm/dirty_ratio:
   20

/proc/sys/vm/dirty_background_ratio:
   10

/proc/sys/vm/dirty_expire_centisecs:
   3000

/proc/sys/vm/dirty_writeback_centisecs:
   500

/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq:
   1600000

/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq:
   2001000

/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq:
   800000

/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_cur_freq:
   1600000

/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_max_freq:
   2001000

/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_min_freq:
   800000

/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor:
   ondemand

/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor:
   ondemand

/proc/acpi/button/lid/LID/state:
   state:      open

/sys/class/power_supply/AC/online:
   0

Laptop mode
enabled, active [unchanged]

*Rob Loranger*

Saanich, B.C.
*www.robloranger.ca <http://www.robloranger.ca>*

rickysarraf, 2015-11-10:
On Mon, 2015-11-09 at 05:34 -0800, rob wrote:
> Laptop mode
> enabled, active [unchanged]

Thanks Rob.

From what you've shared so far, LMT is enabled, and active.

But /proc/sys/vm/laptop_mode: 0
which implies that something overrode it.

Do you have any other power saving tool installed ?
At this boot, if you restart LMT, you should see the settings become
effective. THe best way to debug this would be to restart LMT now, keep
doing your usual work, and at intervals watch the power saving status,
to ascertain when does it get overridden.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

robjloranger, 2015-11-10:
I have pm-utils for suspend/hibernate functionality.
That is what happens, when i restart or start/stop LMT the appropriate settings take effect.

rickysarraf, 2015-11-10:
Ah. That's why. Please use either one. Pm-utils misbehaves with lmt.

s3nt fr0m a $martph0ne, excuse typ0s

rickysarraf, 2015-11-12:
I'm going to close this bug because from what you've mentioned, it must be pm-utils overriding your config.

closed 2015-11-12 by rickysarraf

0rkaM, 2016-06-04:
I know that this may seem a past issue, but it is still very relevant, since it effects many distribution and production installation, which for security and stability policy cannot install "cutting edge" or "latest version"... ex.: Ubuntu 14.04 LTS desktop and server, or equivalent of RHEL and CentOS.
Also on the above production systems PM-Utils are installed as the supported Power Management System and various other essential packages orbit/interact/depends with/on it... like uPower for Ubuntu and even while the blueprint map is to move to Systemd, PM-Utils are already integrated to both... so it cannot really be removed, without compromise stability and functionality (gnome/unity power settings interface with it).
Anyway I confirm the same problem when the system is cold booted on battery... also calling "laptop_mode status", or "service laptop-mode status" does not work properly and always reports enabled even if you "stop" it right before calling "status", but laptop_mode is not working, ignoring the config files. The only way to make it actually really be enable and work as it should, is to call it with "force", or plug the power and then un-plug it.

rickysarraf, 2016-06-04:
Integration with individual distributions is  up to them. On Debian, it works fine. And I think it'll work fine on any distribution using systemd. The original reported was using it on OpenRC, which is something I haven't tested it on. Someone needs to do that and report bugs. But my guess is it must be working fine there too because Gentoo does package Laptop Mode Tools, and openrc originated as a Gentoo Project.

For Ubuntu, I can't really confirm. Usually, Ubuntu will sync with Debian to get the updated packages. But that's **usually**. Lately, it turns out not all packages are auto synced. So unless someone picks and merges Laptop Mode Tools into Ubuntu release cycle, it'd be behind.

```
rrs@learner:~$ sudo service laptop-mode status
[sudo] password for rrs: 
● laptop-mode.service - Laptop Mode Tools
   Loaded: loaded (/lib/systemd/system/laptop-mode.service; enabled; vendor preset: enabled)
   Active: active (exited) since Thu 2016-06-02 20:32:20 IST; 1 day 19h ago
     Docs: man:laptop_mode(8)
           man:laptop-mode.conf(8)
           http://github.com/rickysarraf/laptop-mode-tools
  Process: 6891 ExecReload=/usr/sbin/laptop_mode auto (code=exited, status=0/SUCCESS)
 Main PID: 1175 (code=exited, status=0/SUCCESS)

Jun 04 15:37:12 learner laptop_mode[6197]: enabled, active [unchanged]
Jun 04 15:37:12 learner systemd[1]: Reloaded Laptop Mode Tools.
Jun 04 15:39:59 learner systemd[1]: Reloading Laptop Mode Tools.
Jun 04 15:39:59 learner laptop_mode[6703]: Laptop mode
Jun 04 15:39:59 learner laptop_mode[6703]: enabled, active [unchanged]
Jun 04 15:39:59 learner systemd[1]: Reloaded Laptop Mode Tools.
Jun 04 15:42:29 learner systemd[1]: Reloading Laptop Mode Tools.
Jun 04 15:42:29 learner laptop_mode[6891]: Laptop mode
Jun 04 15:42:29 learner laptop_mode[6891]: enabled, active [unchanged]
Jun 04 15:42:29 learner systemd[1]: Reloaded Laptop Mode Tools.
2016-06-04 / 15:43:13 ♒♒♒  ☺  
```

0rkaM, 2016-06-05:
About "status" I am going to file a bug report for it so we can address it in its own thread.

rickysarraf, 2016-06-06:
What a downstream picks is up to them. I cannot persuade them. There are many things Ubuntu does not pick from Debian, as supported. For what I know, items in Universe and Multiverse are not supported by Ubuntu. Anyways, integration is for the distribution owner to decide. For example, Ubuntu does not support GNOME or KDE or XFCE. That's doesn't mean all those developers need to expect Ubuntu to ship/support all of them.

If you can provide me with Steps To Reproduce, I may look into it. I develop and test it on a Debian box and everything works fine. The same views resonate by other LMT users on Debian.

If you want another distro, Arch and Gentoo do a fairly good work packaging LMT.

That said, back to your issue. The `laptop_mode status` output is legacy. The status command is for initscripts to provide with the status of the running daemon. LMT is NOT a daemon, it never was.
I think Bart (the previous maintainer) must have added it because, back then, there weren't really any useful Power Status tools. I personally would like to remove the 'status' command from LMT but we still are supporting sysvinit and there it complements the output.

0rkaM, 2016-06-06:
That is exactly my point... as Ubuntu does not support GNOME, KDE, XFCE and support is from the actual developers of those (quite large) application and their relative communities, likewise LMT is not supported by Ubuntu but by the respective developer and community.
It would be nice to support the software running on some of the largest Linux installation bases a part from straight Debian, like Ubuntu, RedHat and Slackware... than all the bases are belong to us.
Also Ubuntu does offer LMT on their official package repo... we just need to contact the maintainer and see if they would includes the new code and repackage.
On your second note, I really cannot switch to any other distro on this laptop: for policy reasons it need to stay on a Ubuntu LTS Linux... see Dell only offers Ubuntu LTS on theirs XPS laptops... the other machines I work on are on various Linux distros and perhaps they would not have problems, but they are severs and workstation... so it really doesn't matter much if LMT works on them.
I personally don't mind either way about "status", but I see many cases where it can be really useful... I just would like it to just be a status and not re-run the all script and look like it re-enable LMT.
And about steps: the steps to reproduce the "on Battery boot bug", are the same as above, but I can go into more details if needed; the "status" bug steps are already in its own issue thread #65.

rickysarraf, 2016-06-06:
On Mon, 2016-06-06 at 03:34 -0700, 0rkaM wrote:
> That is exactly my point... as Ubuntu does not support GNOME, KDE, XFCE and
> support is from the actual developers of those (quite large) application and
> their relative communities, likewise LMT is not supported by Ubuntu but by the
> respective developer and community.

It is not the actual developers, but interested people teaming up together to
pick a project and make it work with Ubuntu.

But that aside, I seriously don't have the resource to tackle multiple
distributions. You can be that person for Ubuntu, because you are using Ubuntu.
Just let me know if you see specific bugs in LMT that need to be fixed, to make
it work on Ubuntu. Let's work it as a team.

> It would be nice to support the software running on some of the largest Linux
> installation bases a part from straight Debian, like Ubuntu, RedHat and
> Slackware... than all the bases are belong to us.

Same goes for Red Hat and Fedora. Certain projects prefer one tool over another.
I really can't do much more than raise a request, which I have done in the past.
Apart from that, LMT releases now include RPM packages. Please see the release
page on github.

> Also Ubuntu does offer LMT on their official package repo... we just need to
> contact the maintainer and see if they would includes the new code and
> repackage.

To the best of my knowledge, there aren't really individual maintainers for
Ubuntu Universe/Multiverse packages. It may be a team, but no single maintainer
(like in Debian). 

Why don't you help LMT out? You can be the interface for LMT to interact with
these distros ? Or you can create a semi-official repository for these distros ?

The reason I'm asking this is because it should be a team effort. And I could
really use some help.

> On your second note, I really cannot switch to any other distro on this
> laptop: for policy reasons it need to stay on a Ubuntu LTS Linux... see Dell
> only offers Ubuntu LTS on theirs XPS laptops... [...]

I understand. But maybe you can raise bug reports against your distribution to
see what they have to say ?

> I personally don't mind either way about "status", but I see many cases where
> it can be really useful... I just would like it to just be a status and not
> re-run the all script and look like it re-enable LMT.

It doesn't do any real script execution for that command. It just reads values
and tries to report them in an overly descriptive way.

Like I mentioned before, it is there for legacy reasons. And now with systemd
integration, one wouldn't really see it by default. So eventually I may just
drop it out.

> And about steps: the steps to reproduce the "on Battery boot bug", are the
> same as above, but I can go into more details if needed; the "status" bug
> steps are already in its own issue thread #65.

Regarding the boot bug, I think you may be talking about this:

https://github.com/rickysarraf/laptop-mode-tools/commit/c59e6377cab4938ba30da1d9
1f025b3be7f7a537

or 

https://github.com/rickysarraf/laptop-mode-tools/commit/dafc4c922c28ea1ff7ea1911
a08a7e0c523167e9

I think it is time now for a new release. :-)

But if you are interested to test and provide feedback, you can pick these fixes
from the repository.

PS: I'd be happy if someone could maintain a repo (PPA, OBS, etc) for the famous
distributions (Ubuntu, Elementary, Fedora etc). I really would appreciate the
help.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."