rickysarraf/laptop-mode-tools #50: Running exec-commands as current user

Hori873, 2015-12-03:
How can I run commands insider the exec-commands module as the current user, instead of root?
I want to be able to use the gsettings command there so I can disable gnome animations and disable certain gnome-shell extensions. I also want to stop certain apps and then restart them when my charger is plugged (I can stop them, but not restart them)

rickysarraf, 2015-12-04:
On Thu, 2015-12-03 at 13:27 -0800, Horatiu wrote:
> How can I run commands insider the exec-commands module as the
> current user, instead of root?
> I want to be able to use the gsettings command there so I can disable
> gnome animations and disable [...]

Both, su and sudo, can help you do that.

```
root@learner:~# su -c "ls /root" rrs
ls: cannot open directory /root: Permission denied
```

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

Hori873, 2015-12-04:
I've tried su user -c 'whatever' but it dosen't work... it does if I do it
in terminal while being logged in as root, but not with laptop-mode-tools :(
Specifically i want to set gnome animations off using gsettings

rickysarraf, 2015-12-04:
On Fri, 2015-12-04 at 05:07 -0800, Horatiu wrote:
> I've tried su user -c 'whatever' but it dosen't work... it does if I
> do it
> in terminal while being logged in as root, but not with laptop-mode-
> tools :(
> Specifically i want to set gnome animations off using gsettings

THat's because the forked shell does not inherit your Desktop Session.
If you know the exact environment variables, you can work around it.

Otherwise, you'll have to write a client/server channel over dbus.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

rickysarraf, 2015-12-08:
I'm going to close this because it is not in the scope of LMT. LMT is still a system service.

closed 2015-12-08 by rickysarraf