rickysarraf/laptop-mode-tools #51: error in install.sh

latot, 2015-12-23:
Hi, well i try to install it in gentoo but i get this error D:

```
>>> Install laptop-mode-tools-9999 into /var/tmp/portage/app-laptop/laptop-mode-tools-9999/image/ category app-laptop
./install.sh: line 320: syntax error near unexpected token `('
./install.sh: line 320: `	echo "Installed PMU (pmud/pbbuttonsd) support."'
 [31;01m*[0m ERROR: app-laptop/laptop-mode-tools-9999::Local-X failed (install phase):
 [31;01m*[0m   (no error message)
 [31;01m*[0m 
 [31;01m*[0m Call stack:
 [31;01m*[0m     ebuild.sh, line  90:  Called src_install
 [31;01m*[0m   environment, line 2535:  Called die
 [31;01m*[0m The specific snippet of code:
 [31;01m*[0m       DESTDIR="${D}" INIT_D="none" MAN_D="/usr/share/man" LIB_D="/usr/$(get_libdir)" UDEV_D="$(get_udevdir)" SYSTEMD_UNIT_D="$(systemd_get_unitdir)" TMPFILES_D="/usr/lib/tmpfiles.d" ACPI="$(use acpi && echo force || echo disabled)" PMU="$(false && echo force || echo disabled)" APM="$(use apm && echo force || echo disabled)" SYSTEMD=yes sh ./install.sh || die;
 [31;01m*[0m 
 [31;01m*[0m If you need support, post the output of `emerge --info '=app-laptop/laptop-mode-tools-9999::Local-X'`,
 [31;01m*[0m the complete build log and the output of `emerge -pqv '=app-laptop/laptop-mode-tools-9999::Local-X'`.
 [31;01m*[0m The complete build log is located at '/var/tmp/portage/app-laptop/laptop-mode-tools-9999/temp/build.log'.
 [31;01m*[0m The ebuild environment file is located at '/var/tmp/portage/app-laptop/laptop-mode-tools-9999/temp/environment'.
 [31;01m*[0m Working directory: '/var/tmp/portage/app-laptop/laptop-mode-tools-9999/work/laptop-mode-tools-9999'
 [31;01m*[0m S: '/var/tmp/portage/app-laptop/laptop-mode-tools-9999/work/laptop-mode-tools-9999'
```

so, checking the script:

```
bash -n install.sh
install.sh: línea 320: error sintáctico cerca del elemento inesperado `('
install.sh: línea 320: `	echo "Installed PMU (pmud/pbbuttonsd) support."'
```

Thx.

rickysarraf, 2015-12-24:
This is something Gentoo specific. Have you checked with the Gentoo Maintainer ?

latot, 2015-12-24:
the first part is the log of gentoo, but the second its the bash check of the script, and say it have a syntax error.

latot, 2015-12-24:
Hi, error in line 162:

```
"$DESTDIR/usr/share/polkit-1/actions
```

rickysarraf, 2015-12-24:
@latot Yes. But I can't determine what syntax error. The best I can do is to setup a test env first.

rickysarraf, 2015-12-24:
Woops!! Thanks. Fixed. 811228d

closed 2015-12-24 by rickysarraf