rickysarraf/laptop-mode-tools #52: ATA1 Freeze

latot, 2015-12-30:
Hi, well here probably a weird bug, when i have installed laptop-mode-tools i get this error some times, and actually always in boot:

```
dic 30 10:58:52 drn kernel: ata1.01: exception Emask 0x0 SAct 0x0 SErr 0x0 action 0x0 frozen
dic 30 10:58:52 drn kernel: ata1.01: cmd a0/00:00:00:08:00/00:00:00:00:00/b0 tag 0 pio 16392 in
                                     opcode=0x4a 4a 01 00 00 10 00 00 00 08 00res 00/00:00:00:08:00/00:00:00:00:00/b0 Emask 0x2 (HSM violation)
dic 30 10:58:54 drn kernel: r8169 0000:04:00.2 enp4s0f2: rtl_counters_cond == 1 (loop: 1000, delay: 10).
```

I check the disk and is good, the only solution i found was if i unmerge the app(disable the service don't works) there the message is gone and i have this in all kernels i tested (from 4.0, to 4.33-r2)

Gentoo 64

Cya.

rickysarraf, 2015-12-30:
On Wed, 2015-12-30 at 06:12 -0800, latot wrote:
> Hi, well here probably a weird bug, when i have installed laptop-
> mode-tools i get this error some times, and actually always in boot:
> dic 30 10:58:52 drn kernel: ata1.01: exception Emask 0x0 SAct [...]

https://access.redhat.com/documentation/en-US/Red_Hat_Enterprise_Linux/
6/html/Power_Management_Guide/ALPM.html

Ofcourse, the message will be gone. But then you will consume more
power.

LPM often triggers kernel driver bugs. If you are not okay reporting
them, then yes, you could either blacklist such faulty devices in
Laptop Mode Tools, or else you could simply uninstall Laptop Mode
Tools.

But neither would change the fact that your machine is consuming more
power, and that there is a driver bug somewhere, triggered when LPM is
activated.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

latot, 2015-12-30:
Oki doki, i probably will report it.

Thx.

rickysarraf, 2015-12-31:
Thanks @latot  I'm going to close this bug report.

closed 2015-12-31 by rickysarraf