rickysarraf/laptop-mode-tools #54: LMT can't determine powerstate correctly

dmig, 2016-02-04:
On my CUBE i7 Stylus tablet I noticed that laptop mode was never enabled. Examining logs shown this:
```
Feb  4 11:59:56 dmig-tablet laptop-mode: No /sys/class/power_supply, ACPI, APM or PMU power management information found -- assuming AC power is present.
Feb  4 11:59:56 dmig-tablet laptop-mode: On AC power: Deactivating, because ENABLE_LAPTOP_MODE_ON_AC is not set.
```
/sys/class/power_supply on my device has only BAT0 device and no other powersources. Also, other tests in /usr/sbin/laptop_mode fail to detect AC.
I checked /sys/class/power_supply/BAT0 contents and made this patch.

```
--- laptop_mode.original	2016-02-04 12:52:36.842631936 +0700
+++ laptop_mode	2016-02-04 12:51:17.503055521 +0700
@@ -452,6 +452,7 @@
 
 	# First try /sys/class/power_supply/*
 	FOUND_SYS_CLASS_POWER_SUPPLY_AC=0
+	HAS_BATTERY=0
 	export ON_AC=0
 	for POWER_SUPPLY in /sys/class/power_supply/* ; do
 		if [ -f $POWER_SUPPLY/type ] ; then
@@ -461,6 +462,8 @@
 				if [ "$(cat $POWER_SUPPLY/online)" = 1 ] ; then
 					ON_AC=1
 				fi
+			else
+				HAS_BATTERY=1
 			fi
 		fi
 	done
@@ -503,6 +506,17 @@
 		else
 			ON_AC=1
 		fi
+	elif [ $HAS_BATTERY = 1 ] ; then
+		for POWER_SUPPLY in /sys/class/power_supply/* ; do
+			if [ -f $POWER_SUPPLY/type ] ; then
+				if [ "$(cat $POWER_SUPPLY/type)" = "Battery" ] ;then
+					log "VERBOSE" "Determining power state from $POWER_SUPPLY/status."
+					if [ "$(cat $POWER_SUPPLY/status)" = "Charging" ] ; then
+						ON_AC=1
+					fi
+				fi
+			fi
+		done
 	else
 		log "VERBOSE" "No /sys/class/power_supply, ACPI, APM or PMU power management information found -- assuming AC power is present."
 		ON_AC=1
@@ -555,7 +569,7 @@
 		fi
 
 		log "STATUS" " "
-		STATFILES="/proc/sys/vm/laptop_mode /proc/apm /proc/pmu/info /proc/sys/vm/bdflush /proc/sys/vm/dirty_ratio /proc/sys/fs/xfs/age_buffer /proc/sys/fs/xfs/sync_interval /proc/sys/fs/xfs/lm_age_buffer /proc/sys/fs/xfs/lm_sync_interval /proc/sys/vm/pagebuf/lm_flush_age /proc/sys/fs/xfs/xfsbufd_centisecs /proc/sys/fs/xfs/xfssyncd_centisecs /proc/sys/vm/dirty_background_ratio /proc/sys/vm/dirty_expire_centisecs /proc/sys/fs/xfs/age_buffer/centisecs /proc/sys/vm/dirty_writeback_centisecs /sys/devices/system/cpu/*/cpufreq/cpuinfo_*_freq /sys/devices/system/cpu/*/cpufreq/scaling_governor /proc/acpi/button/lid/*/state /proc/acpi/ac_adapter/*/state /proc/acpi/battery/*/state /sys/class/power_supply/*/online /sys/class/power_supply/*/state"
+		STATFILES="/proc/sys/vm/laptop_mode /proc/apm /proc/pmu/info /proc/sys/vm/bdflush /proc/sys/vm/dirty_ratio /proc/sys/fs/xfs/age_buffer /proc/sys/fs/xfs/sync_interval /proc/sys/fs/xfs/lm_age_buffer /proc/sys/fs/xfs/lm_sync_interval /proc/sys/vm/pagebuf/lm_flush_age /proc/sys/fs/xfs/xfsbufd_centisecs /proc/sys/fs/xfs/xfssyncd_centisecs /proc/sys/vm/dirty_background_ratio /proc/sys/vm/dirty_expire_centisecs /proc/sys/fs/xfs/age_buffer/centisecs /proc/sys/vm/dirty_writeback_centisecs /sys/devices/system/cpu/*/cpufreq/cpuinfo_*_freq /sys/devices/system/cpu/*/cpufreq/scaling_governor /proc/acpi/button/lid/*/state /proc/acpi/ac_adapter/*/state /proc/acpi/battery/*/state /sys/class/power_supply/*/online /sys/class/power_supply/*/state /sys/class/power_supply/*/status"
 
 		for THISFILE in $STATFILES ; do
 			if [ -e "$THISFILE" ] ; then
```

dmig, 2016-02-04:
One more problem is first line of `99-laptop-mode.rules`:
```
ACTION=="change", SUBSYSTEM=="power_supply", ENV{POWER_SUPPLY_NAME}=="|AC|ACAD", RUN+="lmt-udev auto"
```
On my system event just changes `POWER_SUPPLY_STATUS` field, `POWER_SUPPLY_NAME` is always `BAT0`

dmig, 2016-02-04:
I'd suggest changing that line to:
```
ACTION=="change", SUBSYSTEM=="power_supply", ENV{POWER_SUPPLY_NAME}=="|BAT*|AC|ACAD", RUN+="lmt-udev auto"
```
Or completely remove `ENV{POWER_SUPPLY_NAME}` filter

rickysarraf, 2016-02-04:
On Wed, 2016-02-03 at 22:49 -0800, dmig wrote:
> On my CUBE i7 Stylus tablet I noticed that laptop mode was never
> enabled Examining logs shown this:
> Feb  4 11:59:56 dmig-tablet laptop-mode: No /sys/class/power_supply,
> ACPI, APM or PMU power [...]

Ummm.. There are a bunch of things here then.

1. Please file a bug with the kernel reporting that the AC power supply
subsystem does not show up.

2. You don't need the code changes that you've proposed. Since Laptop
MOde Tools, when it cannot detect a power supply, makes an assumption
that the device is ON_AC, all you need is to tell Laptop Mode Tools to
run even while on AC. This feature is something we have designed and
tested well with our users.

You should go that route, instead of deliberating it to other
(untested) means.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

rickysarraf, 2016-02-04:
On Thu, 2016-02-04 at 13:26 +0530, Ritesh Raj Sarraf wrote:
> 2. You don't need the code changes that you've proposed. Since Laptop
> MOde Tools, when it cannot detect a power supply, makes an assumption
> that the device is ON_AC, all you need is to tell Laptop [...]

The file to change is /etc/laptop-mode/laptop-mode.conf

And the setting to change is:

#
# Enable laptop mode when on AC power.
#
ENABLE_LAPTOP_MODE_ON_AC=0

Change that to 1 and it should be in effect.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

rickysarraf, 2016-02-04:
On Wed, 2016-02-03 at 23:23 -0800, dmig wrote:
> One more problem is first line of 99-laptop-mode.rules:
> ACTION=="change", SUBSYSTEM=="power_supply",
> ENV{POWER_SUPPLY_NAME}=="|AC|ACAD", RUN+="lmt-udev auto"
> On my system event just changes POWER_SUPPLY_STATUS field,
> POWER_SUPPLY_NAME is always BAT0

Based on my previous comment, this shouldn't be needed any more. But
otherwise too, remember that battery subsystems are more prone to ill
reporting, which would flood Laptop Mode Tools invocations.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

rickysarraf, 2016-02-04:
On Wed, 2016-02-03 at 23:33 -0800, dmig wrote:
> I'd suggest changing that line to:
> ACTION=="change", SUBSYSTEM=="power_supply",
> ENV{POWER_SUPPLY_NAME}=="|BAT*|AC|ACAD", RUN+="lmt-udev auto"
> Or completely remove ENV{POWER_SUPPLY_NAME} filter

No. We need to narrow it down. Look at that file's git history. It'll
give you a better idea.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

rickysarraf, 2016-02-15:
Please re-open if you have more info. Thanks.

closed 2016-02-15 by rickysarraf