rickysarraf/laptop-mode-tools #55: defer udev events processing

dmig, 2016-02-04:
On my device kernel emits 3 power_supply events in a row. I think, this should be filtered.

I didn't think yet how to implement that using bash, but I have one simple idea for `lmt-udev`:
- make `lmt-udev` singleton, exit immediately if already running and no `force` param 
- sleep for _dejitter timeout_ before kernel event processing

rickysarraf, 2016-02-04:
On Wed, 2016-02-03 at 23:20 -0800, dmig wrote:
> On my device kernel emits 3 power_supply events in a row I think,
> this should be filtered
> I didn't think yet how to implement that using bash, but I have one
> simple idea for lmt-udev:
> make lmt-udev singleton, exit immediately if already running and no
> force param

Sorry to ask, but what is a singleton? I never went to college, so I
don't know these terms.

As for running a single invocation at any given time, we acquire a lock
and ensure entire code execution is lock safe.

> sleep for dejitter timeout before kernel event processing

We're doing better than that actually. At any given period we honor 2
events. One executes acquiring with a lock, while the other waits.
Others are ignored.

Have a look at /usr/sbin/laptop_mode. Is has all the locking
implementation details.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

rickysarraf, 2016-02-15:
I did not hear back from you again. So I'm going to close this issue.

closed 2016-02-15 by rickysarraf