rickysarraf/laptop-mode-tools #56: Set CPU_GOVERNOR per CPU

jmaurice, 2016-02-20:
Of course, when you're on AC power, you want my laptop to be in "performance" mode. But when you're on battery, why set *all* CPUs to "ondemand" ? Ideally I would like to have 1 CPU core set to "performance" and the other 7 CPUs set to "ondemand" - is this currently possible somehow or should I look at sending a PR ? Thanks

rickysarraf, 2016-02-20:
Battery or AC power, ondemand is the most optimal governor is all use cases. There's good reason why it is set as the default in the upstream kernel.

But if you want, you can make the changes. But even before raising the PR, I'd suggest you test it in your environment first.

rickysarraf, 2016-03-07:
I never heard back from you. Is this still a problem ?

If you are on Intel hardware (most common), this does not even apply. So can you share more details?

rickysarraf, 2016-05-29:
Closing this bug. Please re-open if you still have the issue.

closed 2016-05-29 by rickysarraf