rickysarraf/laptop-mode-tools #57: Failed service on boot

mirkojovanovic, 2016-02-20:
I have 10-20 failed services while booting. I searched for string `error` in logs and every single one of them is in line for laptop mode tools. I'm using archlinux. Here's the output of `journalctl -xb`
http://vpaste.net/lmqE5

rickysarraf, 2016-02-20:
You have multiple problems here.

1st line: It says  "operation not permited". That looks more like a permission error. What user is LMT running as, under Arch ?

2nd error: Wifi. That one is fine. Your wireless power saving module seem to be enabled. And the wireless device driver does not seem to support power saving. So that module is failing. It is not a fatal error overall.

Then the hdparm error. Your Arch package manager needs to improve the packaging and add a dependency on hdparm package.

The last, eth0, has more to do with the default ethernet device. Whatever it is named on your box, please use the same in the config file. 'eth0' usually used to be the default name.

mirkojovanovic, 2016-02-20:
The first error gets fixed when I comment lines in intel_pstate file. It probabbly tries to disable turbo, but my processor doesn't have it. I recommend making check if it has turbo or not.

Second, skiping for now.

Third, fixed.

The last, I don't know where to change it in config file there is no 'eth0' at all.

mirkojovanovic, 2016-02-20:
I got lot of new errors after commenting turbo lines(bumblebee, colord). http://vpaste.net/1YeT5

rickysarraf, 2016-02-20:
ethernet changes are in /etc/laptop-mode/conf.d/ethernet.conf.

For the Intel issue, we already are testing for  the file's presence. So you need to check and confirm on your box. It could be a bug with the cpu type you have.

mirkojovanovic, 2016-02-20:
I have i3-3110M 2.4Ghz, ok no problem just tell me what you need.

rickysarraf, 2016-02-20:
Run the intel_pstate module in DEBUG mode. Collect all the logs from syslog/console. And then share it in this bug report.

You can enable debug mode for individual modules in their individual config file.

mirkojovanovic, 2016-02-20:
modprobe can't find intel_pstate module.

rickysarraf, 2016-02-20:
Please check the settings in LMT's individual config file.

mirkojovanovic, 2016-02-20:
You mean in /etc/laptop-mode/laptop-mode.conf? What to check in there?

mirkojovanovic, 2016-02-20:
Ok, I found it in /etc/laptop-mode/conf.d/intel_pstate.config. What to do now?

mirkojovanovic, 2016-02-20:
Just search for 'error', there are first two(permission problem and wireless card not supporting power saving). And now I have bumblebee and colord errors.

mirkojovanovic, 2016-02-20:
Module intel_pstate cannot be loaded (on irc channel they told me that it is not a module it's integrated in kernel), debugging is on. What is next step?

rickysarraf, 2016-02-21:
On Sat, 2016-02-20 at 05:54 -0800, Mirko wrote:
> Ok, I found it in /etc/laptop-mode/conf.d/intel_pstate.config. What
> to do now?

Enable DEBUG=1 and then restart lmt. Based on your setup, the debug
logs will be in syslog on the console screen.

I think the issues you've reported so far are all disto specific. You
should have filed it with your distro developer.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

mirkojovanovic, 2016-02-21:
Arch linux doesn't use syslogs. Logs are in journal, the logs I sent you are output of `journalctl -xb'.

rickysarraf, 2016-02-21:
On Sat, 2016-02-20 at 05:05 -0800, Mirko wrote:
> I got lot of new errors after commenting turbo lines(bumblebee,
> colord). http://vpaste.net/1YeT5

I see no errors in your logs.

Feb 20 13:58:56 mirko laptop-mode[552]: Laptop mode
Feb 20 13:58:56 mirko laptop_mode[491]: Laptop mode
Feb 20 13:58:56 mirko laptop-mode[556]: enabled, active
Feb 20 13:58:56 mirko laptop_mode[491]: enabled, active

These are normal messages, that lmt is working fine.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

mirkojovanovic, 2016-02-21:
`Feb 21 19:41:48 mirko laptop_mode[470]: /usr/share/laptop-mode-tools/modules/intel_pstate: line 14: echo: write error: Invalid argument
Feb 21 19:41:51 mirko colord[1677]: /usr/lib/colord/colord-sane: error while loading shared libraries: libsane.so.1: cannot open shared object file: No such file or directory`

I have these two when I disable turbo lines, which resolves permission error from first message but I get these, plus one that you said it's not fatal.

I don't care much for these messages but I can't stand those fail messages on startup.

rickysarraf, 2016-02-22:
The intel_pstate error is valid but non-fatal.  Your other log had the details that your processor was falsely advertising it. Here it is...

```
Feb 20 18:46:45 mirko kernel: NMI watchdog: enabled on all CPUs, permanently consumes one hw-PMU counter.
Feb 20 18:46:45 mirko laptop_mode[485]: /usr/share/laptop-mode-tools/modules/intel_pstate: line 14: echo: write error: Operation not permitted
Feb 20 18:46:45 mirko laptop_mode[485]: SETTING OF KERNEL PARAMETER FAILED: echo 0 \> /sys/devices/system/cpu/intel_pstate/no_turbo
Feb 20 18:46:45 mirko kernel: intel_pstate: Turbo disabled by BIOS or unavailable on processor
Feb 20 18:46:45 mirko systemd[1]: Started Laptop Mode Tools.
-- Subject: Unit laptop-mode.service has finished start-up
```

And the other error, about colord, has nothing to do with LMT. That error simply is about a missing shared library.

So I think you should close this ticket, if you are convinced.

mirkojovanovic, 2016-02-22:
Ok, thanks.

closed 2016-02-22 by mirkojovanovic