rickysarraf/laptop-mode-tools #60: Standard user cannot launch application from launcher

fastholds, 2016-04-13:
If a standard user tries to launch the application "Laptop Mode Tools Configuration" from the launcher they receive the error that they are not root. Can it instead prompt for root password as may applications do, such as gparted?

rickysarraf, 2016-04-13:
The error you must be seeing is this:

[image]

I deliberately left it to individual distributions to decide what tool they'd want to use. For example, Kubuntu may want to use kdesudo. On GNOME, policykit is preferred.

And that's the same reason why the error message in the screenshot speaks of tool names.

And if that wasnt engouh, python-qt4 these days has annoying issues with drawing a GUI. See commit fc6bbf0d0a31dabc553486ddacdbb017bea3d50b, for example. The annoyance was so much that I decided to leave it at what it is there today, and push the wrapper tool decision to the individual distributions.

I could have extended that wrapper script to deal with the running Desktop Environment. But like I said, the MIT SHM bug was way too annoying.

But if you have suggestions, I'm open to it ?

fastholds, 2016-04-14:
Very good reason. My only suggestion would be to add to that message to "Run gksu lmt-config-gui" or something to that effect. I had my girlfriend install this on her laptop to help with poor battery life and it took some digging in the .desktop files of /usr/share/applications to figure out what the command was to launch the gui config (lmt-config-gui). I did not know the command off hand.

rickysarraf, 2016-04-14:
I have added it in the wiki here. https://github.com/rickysarraf/laptop-mode-tools/wiki/Laptop-Mode-Tools-Graphical-Configuration-Tool

I don't want to change the GUI text reactively. I will keep looking for a solution that can work cross desktop.

closed 2016-04-14 by rickysarraf