rickysarraf/laptop-mode-tools #61: Uninstalling laptop-mode-tools

rootd, 2016-04-21:
How to uninstall laptop-mode-tools? Something like an uninstall.sh script would come in handy.

rickysarraf, 2016-04-22:
This usually was delegated to individual GNU/Linux distributions. While it'd be nice to have something like this, I doubt there are many users. So having an uninstall script is something I don't intend to work any time soon.

What distribution are you on?

rootd, 2016-04-23:
I am running Ubuntu 16.04 right now

rickysarraf, 2016-04-23:
I'm not sure what Ubuntu's user interface to install/remove applications is, but you may want to try the following.

```
apt-get remove --purge laptop-mode-tools
```

Since this issue has not much to do with Laptop Mode Tools, and we expect users to use Laptop Mode Tools through a package, I'm closing this bug report.

closed 2016-04-23 by rickysarraf