rickysarraf/laptop-mode-tools #62: add start stop programs for terminal blanking

gavsiu, 2016-04-28:
Is this possible? I have a keyboard with backlight and would like to turn it off when the screen turns off and turn it back on when the screen turns on.

I could use `xset -q`, but there would be a delay as I wouldn't want it querying continuously.

rickysarraf, 2016-04-29:
On Thu, 2016-04-28 at 09:54 -0700, gavsiu wrote:
> Is this possible? I have a keyboard with backlight and would like to turn it
> off when the screen turns off and turn it back on when the screen turns on.
> I could use xset -q, but there would be a delay as I wouldn't want it querying
> continuously.

What do you mean by screen off ? Are you talking about regular desktop screen
off ? Or are you talking about console terminal blanking ?

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

gavsiu, 2016-04-30:
When the screen goes off during idle. Sorry, I thought it was the same.

rickysarraf, 2016-05-01:
On Sat, 2016-04-30 at 12:56 -0700, gavsiu wrote:
> When the screen goes off during idle. Sorry, I thought it was the same.

Desktop screen idle is controlled by the display server, which typically is
Xorg. That further is controlled by your Desktop Manager, like GNOME, KDE etc.

There's not much Laptop Mode Tools can do here.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

closed 2016-05-01 by gavsiu