rickysarraf/laptop-mode-tools #63: Configurable cpuhotplug

cjones24, 2016-05-18:
The cpuhotplug module saves me a fair bit of power, but would be nice to be able to choose how many cores to turn off, as going from four to one has a dramatic effect on performance

rickysarraf, 2016-05-18:
I am glad that it is helping save more power. When I added it, I doubted its impression on x86. Are you on x86 hardware?

Good point of the configurability of cores.  What do you think would be the sensible approach for configuration? Something like:

```
# Possible values are:
# half -> disable half of available logical CPUs
# quarter -> disable 1/4 of available logical CPUs
# 3quarter -> disable 3/4 of available logical CPUs
DISABLE_AVAILABLE_CPU="half"
```

Or do you have any different suggestion?

cjones24, 2016-05-18:
i5-5200U, i'll run some tests with how much difference it makes by itself in a minute.
Also I think that config might cause people with odd-cored hardware some problems. 
Maybe:

```
# Number of cores enabled, 0 indicates all cores enabled
LM_AC_CPU_HOTPLUG=0
NOLM_AC_CPU_HOTPLUG=0
BATT_CPU_HOTPLUG=2
```
Im not sure how the backend of it all clicks together so this might not be viable since you've got a pretty good 0=no 1=yes thing going on everywhere else.

rickysarraf, 2016-05-18:
No. The fields you've mentioned are for a different purpose. There are users who want LMT enabled even while plugged in on AC (Media Jukebox, servers).

As for the number, I left it off because that makes an assumption that one known how many cpu cores their machine has. Where as what I proposed will determine that at runtime.

rickysarraf, 2016-05-18:
And I'm glad you are on x86, on an Intel Broadwell family. I have a Haswell box, and I'm going to try the same here. :-)
Thanks.

cjones24, 2016-05-18:
Have some power draw figures:
* Hotplug enabled:
  * Normal usage:    8647000μW
  * Under stress:     12885000μW
* Hotplug disabled:
  * Normal usage:   12682000μW
  * Under Stress:     17460000μW

Conditions were kept the same for each test and are actually medium/high usage (lot of tabs, speakers on, backlight maxed), the only change was enabling/disabling cpuhotplug and reloading the systemd module (kept an eye on /proc/cpuinfo to check it was working).  Stress test was just stress -c 4. Waited for the power draw to normalise each time then took the most average looking value.

I think a difference of over 4W under normal usage and nearly 5W under stress is pretty significant on a cpu with only 15W tdp.  I have a desktop ivy bridge platform I can test later on if you want.

As for the config you'd have to work around people with old tri-core amds and hexa-core intels if you're working in halfs and quarters.

rickysarraf, 2016-05-18:
On Wed, 2016-05-18 at 04:14 -0700, cjones24 wrote:
> Have some power draw figures:
> Hotplug enabled:
> Normal usage: 8647000μW
> Under stress: 12885000μW
> Hotplug disabled:
> Normal usage: 12682000μW
> Under Stress: 17460000μW
> Conditions were kept the same for [...]

Thank you very much for these number. They are exciting. :-)

> As for the config you'd have to work around people with old tri-core amds and
> hexa-core intels if you're working in halfs and quarters

Hmmm... good point. Let me cook something up. I'll ping you back.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

rickysarraf, 2016-05-18:
@cjones24 This should work as you requested. Can you test it?
https://github.com/rickysarraf/laptop-mode-tools/commit/d8827ef047e6496daa317cedbc818c5952712855

I'm re-opening this bug report. After you verify the patch, you can update this bug report.

cjones24, 2016-05-18:
all the modes work well on the laptop, so much better with 2 cores.  I'll do some more power tests later on but quarter, half and 3quarter all work as intended, awesome. Great turnaround

closed 2016-05-18 by cjones24

aartamonau, 2016-09-28:
Looking at the patch, it's currently unaware of cpu topology. I'm not able to confirm that with numbers, but intuitively it seems that on a machine with hyperthreading enabled (pretty common), it will be more beneficial to disable as many physical cores as possible. E.g., if I have two physical cores with two threads each, and I want to disable two logical cpus, then it should disable two logical cpus sharing the same physical core.

I can come up with a patch, if this makes sense to you, @rickysarraf.

rickysarraf, 2016-09-28:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

On Wed, 2016-09-28 at 06:08 -0700, Aliaksey Artamonau wrote:
> Looking at the patch, it's currently unaware of cpu topology. I'm not able to
> confirm that with numbers, but intuitively it seems that on a machine with
> hyperthreading enabled (pretty common), it [...]

Do you see this related to #71 ?
https://github.com/rickysarraf/laptop-mode-tools/issues/71

What you explained makes sense to me, but it'd be nicer if we can confirm the
same in the Linux sources or x86 CPU vendor documentation.

> I can come up with a patch, if this makes sense to you, @rickysarraf.
> 

Yes. Please.

- -- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."
-----BEGIN PGP SIGNATURE-----

iQIcBAEBCgAGBQJX6+abAAoJEKY6WKPy4XVpc7YQAJjvHM2hTmupktG2P5g0dE+Y
5P7ImXy8oj8hh0z/MPMJxtsWGkdVcjIhJ4gBWpHNKeLeKHjgN+i/PWsedKGhOtp3
rUZ8GC0w3VQik3Jt4SOwUQ0THYSp3yq3Hxh3P39+qZ2Y/uoM3PjWY5+pZWl9J8EQ
ux8BU43NKruuOxkp9z0F1jaN+tImT7K9CMjKdDC4t856BNZ/Y7GAXvjbYQVXA3Da
C1RQFGb/eE83aR6E/c1mlUEJs3Xc3mZE3UZ5K4vWpKSLk+WKuSyx4ofLtVjkMq63
dodCKHAp5bi/e2qjlxPAI92NWAMMgSURg6JOpd7AXW9FcBSaaWI019WM+hz6P14H
uAHAhmd/e0u1FOkfWCU6r5gwXthm5iK6ID+S87giB1neM7RcwpN+BCmM08kYTd+v
Sf7yW27o2NbGhNjZxyYtrhIMWcmIi9RlV0vG+8CviJAY9+I/xeoYEDaybOZKml//
mWEzuczdnR41XqNmPaB1u1a5nLHJO9h/PnmN5Jg7ZfAdsRwYXao2OehvkSCm1K4p
8JcIfn6BSrG2elAjWMsUvMNPJWOGD6ApW4oikYPM8Ehhlf3FBXFJX4nFlR8Os2mY
M6XXkHP/WmJo4L4stXy6vh6br603soirU+ComUBj4mRHCfl8Mqnm5UG2+KJmnTgm
l2qaN3uz0oF1VUgFKitU
=3aBT
-----END PGP SIGNATURE-----

aartamonau, 2016-09-29:
Yes, I've seen that issue, but I wasn't sure what the reporter meant by "logical order". On the second thought, would it make sense to let user configure which CPUs they want to disable specifically (some kind of list)? That way, the hotplug module doesn't get much more complicated and there's no extra burden of proving that the chosen behavior is actually the best one in terms of power saving.

rickysarraf, 2016-09-29:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

On Wed, 2016-09-28 at 23:00 -0700, Aliaksey Artamonau wrote:
> Yes, I've seen that issue, but I wasn't sure what the reporter meant by
> "logical order". On the second thought, would it make sense to let user
> configure which CPUs they want to disable specifically [...]

Yes. I think that will be a much better option than we determining attributes
(logical order) from a wide range of available processors.

We already have simple english based config options. What you've proposed above
will complement it, for almost all use cases.

- -- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."
-----BEGIN PGP SIGNATURE-----

iQIcBAEBCgAGBQJX7MUvAAoJEKY6WKPy4XVpcKgQAJFBkk7R+zwbK6RVozMvaI3C
rd0J+mXwfv+DZdDZU1dlOOOaxOgGPH/CkbEAneDiUYxhxL73vNpkP96IeoJ+anhR
Cdrtl/ecmK7V2jP+Su/oqC1c3zYrZXHIroAigSgkZhg5bMsuBm+MKTRABIPc+kWV
/4ntm1LxsFCosV8DDg41mKpy/Ohn/ysD+Aom2vYxt33qwd5XueJGYTFtH2eMmtQH
SJFqP9NmrmV0SSg8nJDlrNNnjOwm2N4zSEOhsJ/yYUxgY82LyYxW+QHxaSkfWwtA
vBrPxnkOKzgmHoDJoXItS02HMHHrlZBBUBZMyAEj4VvMS8+HMZyzW6kIeK9r2y3s
geQecPdmcXAN6j534N2c3EMR4m4Wpy6iH0Nk+v0Jn+1+InPZcU4OAmOTGn72y5pW
byePs9fwrIt8Imd2iEx9VTI77hcgZjvW0qQjDP4+akrVD7zUghRZp399EURSK/mJ
IasWnyk9zzYyuCyPthrnKqgd/u4Cqo49OF7vpKz5DxAZuaB3pGTHNxCEcLTjlek6
kKR1X/TGUAOpfHYx70dtMwrsBDOzN1sKpqnPtm8I4EPMcNXelG5z7TJpTitvdg5J
iItmkYYyTBjJ3+egGyG+3ltD/itBM1O4BqAabTBhi36qDKrtChgha5xXAhnn28Hx
iqphmJ/IG80ccXBeFWkB
=e7wY
-----END PGP SIGNATURE-----