rickysarraf/laptop-mode-tools #64: Can't detetct if laptop is on battery

snugglej, 2016-06-05:
I'm running ubuntu-mate with laptop-mode-tools on an xps 13. 

There currently is no way for the laptop-mode-tools script to detect if it is on power or battery because neither /sys/class/power_supply/BAT0/online,  /proc/acpi/ac_adatper/, or /proc/pmu/info. So it will always fail to thinking that the system is on AC power.

Would you be willing to add some logic to detect if /sys/class/power_supply/$POWER_SUPPLY has online or status and if it has status then checking to see if it has AC power.

Example Request:
 if [ "$(cat $POWER_SUPPLY/type)" != "Battery" ] ;then
                 log "VERBOSE" "Determining power state from $POWER_SUPPLY/      online."
                 FOUND_SYS_CLASS_POWER_SUPPLY_AC=1
                 if [ "$(cat $POWER_SUPPLY/online)" = 1 ] ; then
                     ON_AC=1
                 elif [ "$(cat $POWER_SUPPLY/status)" != "Discharging" ] ; then
                     ON_AC = 1
                 fi
             fi

Note: (Something I found online)
A friendly reminder: since Linux kernel 2.6.24 using /proc to store ACPI info has been discouraged and deprecated.

Now we are encouraged to use -> /sys/class/power_supply/BAT0.

UPDATE: Linux 3.19 and onwards, we should look at the following directory -> /sys/class/power_supply/BAT1/

For example capacity -> cat /sys/class/power_supply/BAT1/capacity status -> cat /sys/class/power_supply/BAT1/status

rickysarraf, 2016-06-05:
I'm not sure if I understand the bug report correct.

But currently we are determining the machine state. Look at the follwoing:

https://github.com/rickysarraf/laptop-mode-tools/blob/lmt-upstream/usr/sbin/laptop_mode#L453

0rkaM, 2016-06-06:
I think the link you are referencing to is exactly what he is referring to. The part to "determine the power state" maybe it is not working as planned on his installation; I wonder if it could be related to the "boot on battery" problem.

I checked it and the code is the same on my Dell XPS 12 Ubuntu 14.04 LTS with LMT version 1.64 (the only "legally" available on LTS without breaking policy) and I am also experiencing issues #65 and #49, of which I am trying to track the cause down.

I think he basically said the way it is checking:

> /sys/class/power_supply/*/online
> /proc/acpi/ac_adapter/*/state
> /proc/pmu/info

it does not work for him.

I can see in some laptop some of those do not exist, because of hardware and depending on kernel, also in modern laptops and installations, some are just deprecated and fading away, or just not the right place... but you know all that.

Also the same is valid for **/proc/apm**

So the actual fix he was showing may actually work him, but I fail to see why would checking `if [ "$(cat $POWER_SUPPLY/status)" != "Discharging" ]` would be better than `if [ "$(cat $POWER_SUPPLY/online)" = 1 ]`, unless his system `$POWER_SUPPLY/*` does not also resolve to `/sys/class/power_supply/ACAD/` but only to `/sys/class/power_supply/BAT1/`, therefore his system only has `/sys/class/power_supply/BAT1/status`, but not `/sys/class/power_supply/ACAD/online`.

rickysarraf, 2016-06-06:
This bug was reported as a bug in the way LMT detect the battery. That code
hasn't changed in a while because the interface is intact. I would be surprised
if Linux changed that without good reasons. The order is for backwards
compatibility.

If there is hardware which is not covered, please provide details about it. If
it is a new standard interface, I'll add it. For broken, buggy, non-standard
laptops, there's not much to do.

The other power management code is for other (non-acpi) architectures

On Mon, 2016-06-06 at 03:51 -0700, 0rkaM wrote:
> I think the link you are referencing to is exactly what he is referring to.
> The part to "determine the power state" maybe it is not working as planned on
> his installation; I wonder if it could be [...]
-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

snugglej, 2016-06-06:
@rickysarraf You are correct, I'm running Ubuntu-Mate 16.04 on an XPS 13 9350. I have acipd installed and it doesn't resolve the correct information, I can't install the other option and my $POWER_SUPPLY only has BAT0 which doesn't have online. So I only have the option of checking status or forcing laptop-mode-tools to turn on.

snugglej, 2016-06-06:
Mounts:
   /dev/mapper/ubuntu--mate--vg-root on / type ext4 (rw,relatime,errors=remount-ro,commit=600,data=ordered)
   /dev/nvme0n1p2 on /boot type ext2 (rw,relatime,block_validity,barrier,user_xattr,acl)
   /dev/nvme0n1p1 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
 
Drive power status:
   Cannot read /dev/[hs]d[abcdefgh], permission denied - /usr/sbin/laptop_mode needs to be run as root
 
(NOTE: drive settings affected by Laptop Mode cannot be retrieved.)
 
Readahead states:
   /dev/mapper/ubuntu--mate--vg-root: 3072 kB
   /dev/nvme0n1p2: 3072 kB
   /dev/nvme0n1p1: 3072 kB
 
Laptop Mode Tools is allowed to run: /var/run/laptop-mode-tools/enabled exists.
 
/proc/sys/vm/laptop_mode:
   5
 
/proc/sys/vm/dirty_ratio:
   60
 
/proc/sys/vm/dirty_background_ratio:
   40
 
/proc/sys/vm/dirty_expire_centisecs:
   60000
 
/proc/sys/vm/dirty_writeback_centisecs:
   60000
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq:
   1285984
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq:
   3100000
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_cur_freq:
   1511046
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_max_freq:
   3100000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_cur_freq:
   1646835
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_max_freq:
   3100000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_cur_freq:
   1652015
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_max_freq:
   3100000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor:
   powersave
 
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor:
   powersave
 
/sys/devices/system/cpu/cpu2/cpufreq/scaling_governor:
   powersave
 
/sys/devices/system/cpu/cpu3/cpufreq/scaling_governor:
   powersave
 
/proc/acpi/button/lid/LID0/state:
   state:      open
 
Mounts:
   /dev/mapper/ubuntu--mate--vg-root on / type ext4 (rw,relatime,errors=remount-ro,commit=600,data=ordered)
   /dev/nvme0n1p2 on /boot type ext2 (rw,relatime,block_validity,barrier,user_xattr,acl)
   /dev/nvme0n1p1 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
 
Drive power status:
   Cannot read /dev/[hs]d[abcdefgh], permission denied - /usr/sbin/laptop_mode needs to be run as root
 
(NOTE: drive settings affected by Laptop Mode cannot be retrieved.)
 
Readahead states:
   /dev/mapper/ubuntu--mate--vg-root: 3072 kB
   /dev/nvme0n1p2: 3072 kB
   /dev/nvme0n1p1: 3072 kB
 
Laptop Mode Tools is allowed to run: /var/run/laptop-mode-tools/enabled exists.
 
/proc/sys/vm/laptop_mode:
   5
 
/proc/sys/vm/dirty_ratio:
   60
 
/proc/sys/vm/dirty_background_ratio:
   40
 
/proc/sys/vm/dirty_expire_centisecs:
   60000
 
/proc/sys/vm/dirty_writeback_centisecs:
   60000
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq:
   1612710
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq:
   3100000
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_cur_freq:
   1726562
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_max_freq:
   3100000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_cur_freq:
   1832898
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_max_freq:
   3100000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_cur_freq:
   1945937
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_max_freq:
   3100000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor:
   powersave
 
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor:
   powersave
 
/sys/devices/system/cpu/cpu2/cpufreq/scaling_governor:
   powersave
 
/sys/devices/system/cpu/cpu3/cpufreq/scaling_governor:
   powersave
 
/proc/acpi/button/lid/LID0/state:
   state:      open
 
Laptop mode 
enabled, active [unchanged]

rickysarraf, 2016-06-07:
Thank you for your patience @0rkaM  and @snugglej .

I think this issue should now be fixed with commmit: 09b98315a9c752dee0cfbd1ee66a9a91fe10bb1f

WIll you be in a position to verify the fix and update on this bug report ?

snugglej, 2016-06-08:
@rickysarraf I'm in a position to verify this fix and update the bug report. I'll be pulling the file down and testing. 

Thanks!

snugglej, 2016-06-09:
@rickysarraf 

I was wondering if you can update the code to be:

if [ -f $POWER_SUPPLY/type ] ; then
....
elif [ "$(cat $POWER_SUPPLY/type)" == "Battery" ] ; then
...
fi
instead of the elif being nested inside of that if statement? Because the file $POWER_SUPPLY/type doesn't exist and so it would skip over that whole if segment it would never check the battery status.

If you want I will be happy to do a fork and make the change on my own but I don't want to step on other people's feet.

rickysarraf, 2016-06-09:
@snugglej If the type **_file_** identifier is not present, the 2nd case you propose cannot be evaluated.

snugglej, 2016-06-09:
@rickysarraf  Sorry you are correct, I was not paying attention last night when I was looking over the file. 

There seem to be some issues with code which has ";then" instead of "; then" , should I just correct those in my copy of the file

Incorrect
if [ "$(cat $POWER_SUPPLY/type)" == "Mains" ] ;then
Correct
if [ "$(cat $POWER_SUPPLY/type)" == "Mains" ] ; then

Incorrect
 elif [ "$(cat $POWER_SUPPLY/type)" == "Battery" ] ;then
Correct
 elif [ "$(cat $POWER_SUPPLY/type)" == "Battery" ] ; then

rickysarraf, 2016-06-10:
On Thu, 2016-06-09 at 13:21 -0700, snugglej wrote:
> @rickysarraf Sorry you are correct, I was not paying attention last night when
> I was looking over the file.
> There seem to be some issues with code which has ";then" instead of "; then" ,
> should I just correct those in my copy of the file

Sorry about that error. I've fixed and pushed it to the git repo.

-- 

Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

snugglej, 2016-06-10:
Thanks for fixing that! I'll update to the new file just to continue testing.

snugglej, 2016-06-14:
Continued testing has proven that the new script is working in detecting that I"m on battery since depending on the state I boot in I may be missing the AC folder. Each time I run lapto_mode start/stop/force it detects that I'm on battery. So my request has been addressed but I'm looking at the other bug with detection while booting off AC because I still experience that.

Thanks for all your work!

rickysarraf, 2016-06-14:
Thanks. I'll close this bug report then. Whatever the other issue is, once you have enough information, please file a new bug report.

closed 2016-06-14 by rickysarraf