rickysarraf/laptop-mode-tools #65: LTM status report show incorrect state.

0rkaM, 2016-06-05:
Calling "laptop_mode status", or "service laptop-mode status" does not work properly and always reports enabled even if you "stop" it right before calling "status":

```
makr0@espresso:~$ sudo service laptop-mode stop
[sudo] password for makr0: 
 * Laptop mode 
```
**`diabled, not active`**
```
makr0@espresso:~$ sudo service laptop-mode status
Laptop mode status:

Mounts:
   /dev/mapper/ubuntu--vg-root on / type ext4 (rw,errors=remount-ro,commit=0)
   /dev/sda2 on /boot type ext2 (rw)
   /dev/sda1 on /boot/efi type vfat (rw)
 
Drive power status:
   
   /dev/sda:
    drive state is:  active/idle
 
(NOTE: drive settings affected by Laptop Mode cannot be retrieved.)
 
Readahead states:
   /dev/mapper/ubuntu--vg-root: 128 kB
   /dev/sda2: 128 kB
   /dev/sda1: 128 kB
 
Laptop Mode Tools is allowed to run: /var/run/laptop-mode-tools/enabled exists.
 
/proc/sys/vm/laptop_mode:
   0
 
/proc/sys/vm/dirty_ratio:
   40
 
/proc/sys/vm/dirty_background_ratio:
   10
 
/proc/sys/vm/dirty_expire_centisecs:
   3000
 
/proc/sys/vm/dirty_writeback_centisecs:
   500
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq:
   1225687
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_cur_freq:
   1297054
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_cur_freq:
   1361039
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_cur_freq:
   1344656
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor:
   powersave
 
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor:
   powersave
 
/sys/devices/system/cpu/cpu2/cpufreq/scaling_governor:
   powersave
 
/sys/devices/system/cpu/cpu3/cpufreq/scaling_governor:
   powersave
 
/proc/acpi/button/lid/LID0/state:
   state:      open
 
/sys/class/power_supply/ACAD/online:
   0
 
Laptop mode 
```
**`enabled, active`**
```
makr0@espresso:~$ 
```

rickysarraf, 2016-06-06:
What init are you on ?

rickysarraf, 2016-06-06:
Irrespective of the init system, your bug is valid. Because if the user instructs it to stop, they'd like to see it stopped.

The correct way to disable LMT is to disable it in the configuration file. Quoting the manpage:

```
   /etc/laptop-mode/laptop-mode.conf
       This is the main configuration file. It contains the settings for enabling  and  dis‐
       abling, plus the core features: the Linux kernel laptop mode feature and related set‐
       tings.

       ENABLE_LAPTOP_MODE_TOOLS
                 This module determines if laptop-mode-tools should be run or  not.  Default
                 is 1 Set it to 0 if you would like to completely disable laptop-mode-tools

```

This rule applies irrespective of the init system.

rickysarraf, 2016-06-06:
I've updated the same in FAQs.

https://github.com/rickysarraf/laptop-mode-tools/wiki/FAQ#can-i-force-laptop-mode-onoff

rickysarraf, 2016-06-14:
I did not hear back from you again. Can we close this bug ?

0rkaM, 2016-06-17:
Sorry I am a little disabled by an accident at work and my year old sone is sick so it takes me longer to get things done.
Anyway, I would like to see the status functionality working as just reporting the status of LMT and not changing it from "disabled, not active" to "enabled, active" when asked to report on it.
But it is your call, if you feel that updating the FAQ will suffice than we can revisit it when someone else complain about it.

rickysarraf, 2016-08-15:
I'm going to close this one now. Because with modern init systems, the status isn't really propagated. For SysvInit, it was chosen to propagate, as back then it made sense. But not with newer init systems.

For example, on systemd, there's no concept of ExecStatus header. And its status output has more to do with command exit status and strings around.

But yeah, if there are ideas, I'm open to it and then please feel free to re-open this bug.

closed 2016-08-15 by rickysarraf