rickysarraf/laptop-mode-tools #67: Wrong docs comments in runtime_pm.conf

jprukner, 2016-07-13:
I think there is a wrong example in comments in file runtime_pm.conf
'''
# AUTOSUSPEND_DEVID_BLACKLIST="usbhid usb-storage"
'''
I tried uncomment mentioned line and after reboot mouse keeps sleeping.
My opinion is that usbhid nor usb-storage are not DEVIDs. I think they are DEVTYPEs.

jprukner, 2016-07-13:
Well, it seems there is not up to date version in Debian repositories. I installed it using apt-get...

closed 2016-07-14 by jprukner

rickysarraf, 2016-07-15:
I'm assuming the issue is resolved for you, that's why you closed the bug report ? If not so, please feel free to re-open it.

As for the updated version in Debian, it usually is very close to upstream in Debian Testing/Unstable. The version in Debian Unstable has the same settings.
https://packages.debian.org/sid/laptop-mode-tools

jprukner, 2016-07-15:
Yes, that is why I closed it. I didn't install laptop mode tools correctly. Sorry for unnecessary bug report. @rickysarraf