rickysarraf/laptop-mode-tools #71: cpuhotplug does not follow logical cpu order, and other minor issues

roobre, 2016-08-20:
Just looked at the cpuhotplug code, here are some issues I found:

1. Cpu offlining is not done in a logical order. This can greatly decrease potential power saving on hyperthreaded cpus.

2. Cpu 0 can never be offlined, however, it is counted as so if the loop passes through it.

3. LMT does not online the remaining CPUs after the desired number of cpus have been offlined. This can lead to undesirable scenarios and race conditions if the user changes the setting without rebooting the machine.

I'm working on a PR to fix this, stay tuned.

rickysarraf, 2016-08-21:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

Hi,

On Sat, 2016-08-20 at 09:42 -0700, Roberto Santalla wrote:
> Just looked at the cpuhotplug code, here are some issues I found:
> Cpu offlining is not done in a logical order. This can greatly decrease
> potential power saving on hyperthreaded cpus.

can you please point me to the source for this information? I'm interested to
first read on this.

> Cpu 0 can never be offlined, however, it is counted as so if the loop passes
> through it.

Yes. I'm aware of that. But is that really a problem? We increased the count in
max_available_cpus() for that reason only, IIRC.

> LMT does not online the remaining CPUs after the desired number of cpus have
> been offlined. This can lead to undesirable scenarios and race conditions if
> the user changes the setting without rebooting the machine.

Ummm. I've been using the module and haven't run into this problem. All logical
CPUs come online once we plug in the power.

I need some logs or steps to reproduce this locally.

> I'm working on a PR to fix this, stay tuned.

- -- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."
-----BEGIN PGP SIGNATURE-----

iQIcBAEBCgAGBQJXuYn8AAoJEKY6WKPy4XVpy8IP/2usYNmgF4coCd6oDFc2OU9P
cTaUCkMQEQJNwytA+1ViCsXJ0d8b7200N2ahR1A9Z6snb4xq+xhhw+MyeM4cRoVc
q8ayM3YhFgeguHKy50LSp1mDBuBJL4YjPQrgZCo0CsWM/GFspuzhA4oHOCxQqaG8
z0QzlUzixKoLNWWnKXEHDMfzCt39+NPEpyuNPPrzKYAEd6HxSl6ek4Vnm9Y/EtHk
qXwcrpF3GDTuB0aZ7KE++iX3HJX4fK1yQm/sdNggrSGjyGReNNXzgOUMGbM4F8/b
VGzy8mJsLJ8PEjaX6Je34JmIxX0DkiCZRBWogoj1K1m98ZzyGc4x8LOwWxV86Vcv
e1ri/u2gdgqNuBD1nMca2vH0XeVXVtSNijhPQhsB7h/9OOibgf6siuNCoAO6gGmv
b0wTr4KP2k1lptlzSQPKzH9fr+9k5wNqfR/LWtLlT4AQ187fneuvW9VMm7E8Qulj
MhOCKDfycNNcNy+Qjr7Shx+MqKjPQdCwCXVi+8FNtTCZaNXcx1IMAsnU6PPiWbBS
aCCMl2MXcSQLQZ41mMta8NHyUEsIxpb6oPi777cxxeEHk6Joqi5R7KZ4SQMYrDa7
isp+KunQf/yMDg3yFBoe3b9E+O726c7SKCVyNqf3rlzyGnuiH6rzABM0zzNn17Vy
SGI28EYQlX7s/KdLq9mx
=OYW0
-----END PGP SIGNATURE-----

rickysarraf, 2016-08-21:
BTW, my findings are from the latest git snapshot.

roobre, 2016-08-29:
> can you please point me to the source for this information? I'm interested to
first read on this.

There is no official documentation afaik. However, I've seen this behaviour on every /proc/cpuinfo I've looked at on a hypertheaded machine (look at `core id`).

> Yes. I'm aware of that. But is that really a problem? We increased the count in
max_available_cpus() for that reason only, IIRC.

You are partially right. The behaviour is the desired because you compare using `$counter -le $disable_cpus`. If `$disable_cpus` is set to, let's say, 4, it will go into the if five times. This is, however, quite obscure, and may not scale well if for whatever reason there is more than one non-offlineable cpu. The counter should only be increased if the actual disable is possible.

> I need some logs or steps to reproduce this locally.

Enable module, set variable to 4 (for example), restart, set variable to 2, restart.

closed 2016-10-16 by rickysarraf