rickysarraf/laptop-mode-tools #75: Auto-hibernate is always triggered twice upon disconnecting power

carbohydratesn, 2016-10-06:
Hi, I've been trying to get this working properly for days, with no success.  Using a ThinkPad x230.

Hibernate is triggered on low battery, most of the time, but it's a little too overzealous.  

I've tried making my suspend script check to make sure the battery is actually lower than the threshold, and it works when run manually.  It won't cooperate when invoked by laptop-mode.  Regardless of my battery level, it suspends _twice_ whenever I unplug the laptop.

I've tried enabling debug mode for auto-hibernate, but can't find the output - please let me know if there's anything useful I could include.

Laptop-mode is enabled on battery and disabled on AC - could it have something to do with the transition?

carbohydratesn, 2016-10-07:
On further investigation it looks like something strange is happening with battery measurements.

A value of '30' in auto-hibernate.conf causes the laptop to hibernate when it hits 80% battery.

At 100%, I can unplug it without triggering a hibernate.  Below 80%, it is triggered.  

Both acpi and the emacs package fancy-battery both pick up the correct level of charge.  laptop-mode does not.  This could have something to do with the age of the battery, perhaps laptop-mode is calculating based on the original capacity of the battery?  30% of a new battery could be equivalent to 80% of one that's a few years old.

'12' in the conf seems to correspond to a 5% real charge...very strange.

rickysarraf, 2016-10-07:
It could very well be a broken battery. But I could be wrong, because you mentioned that other tools are able to get a better battery reading....

First thing, could you please run it in verbose mode ? You can set that in /etc/laptop-mode/laptop-mode.conf
The logs will be logged into your journal (systemd) or syslog (sysvinit).

I think once you enable verbose setting, we'll get good enough information to determine the cause.

rickysarraf, 2016-10-07:
We still need to root cause this with evident information, but my guess is this is what must be happening.

Usually, when you are suspending, your battery level reported by the power_supply system is below the limit. Then you suspend. And charge the battery.

Upon resume, it picks the frozen values. It takes a couple of seconds before it reinitializes the new values (which caused the battery to charge while the machine was sleeping)

To experiment this theory, could you try the following:

```
rrs@learner:~/rrs-home/devel/Laptop-Mode-Tools/laptop-mode-tools (lmt-upstream)$ cat /lib/udev/rules.d/99-laptop-mode.rules 
ACTION=="change", SUBSYSTEM=="power_supply", ENV{POWER_SUPPLY_NAME}=="|AC|ACAD", RUN+="lmt-udev auto"
ACTION=="add|remove", SUBSYSTEM=="machinecheck", RUN+="lmt-udev auto"
ACTION=="add", SUBSYSTEM=="usb", RUN+="lmt-udev auto"
```

Disable the first 2 lines on your machine. But please do this after running once in verbose mode. I'm still interested to have the verbose logs

rickysarraf, 2016-10-07:
And please provide full bug details: LMT version, distro, etc.

carbohydratesn, 2016-10-07:
Well this is a fun journal;

http://pastebin.com/XV4J1Gnb

My /dev/fd does seem to be populated...but /proc/self/fd/* only has user rights.  I'll try seeing if I can get a proper verbose log on my end, but I'm new to this, advice would be much appreciated

`user@coniferous ~ $ ls -al /dev | grep fd
lrwxrwxrwx  1 root root        13 Oct  7 15:57 fd -> /proc/self/fd
lrwxrwxrwx  1 root root        15 Oct  7 15:57 stderr -> /proc/self/fd/2
lrwxrwxrwx  1 root root        15 Oct  7 15:57 stdin -> /proc/self/fd/0
lrwxrwxrwx  1 root root        15 Oct  7 15:57 stdout -> /proc/self/fd/1`

`user@coniferous ~ $ ls -al /dev/fd
lrwxrwxrwx 1 root root 13 Oct  7 15:57 /dev/fd -> /proc/self/fd`

`user@coniferous ~ $ ls -al /proc/self/fd
total 0
dr-x------ 2 user user  0 Oct  7 16:06 .
dr-xr-xr-x 9 user user  0 Oct  7 16:06 ..
lrwx------ 1 user user 64 Oct  7 16:06 0 -> /dev/pts/5
lrwx------ 1 user user 64 Oct  7 16:06 1 -> /dev/pts/5
lrwx------ 1 user user 64 Oct  7 16:06 2 -> /dev/pts/5
lr-x------ 1 user user 64 Oct  7 16:06 3 -> /proc/4855/fd`

More details - version;

`user@coniferous ~ $ sudo emerge -av laptop-mode-tools
[ebuild   R    ] app-laptop/**laptop-mode-tools-1.66**::gentoo  USE="acpi bluetooth -apm -scsi" 0 KiB`

(it's compiled with ACPI and bluetooth support, and without APM or SCSI)

Using Gentoo GNU/Linux, with Systemd.  Problem observed with kernels ranging from 4.4.6-4.8

Trying the modified 99-laptop-mode.rules now, and then I'll start poking at /dev/fd/ again

carbohydratesn, 2016-10-07:
I changed all references to /dev/fd to a normal file in my home dir, to get a proper log.

Log 1 - Laptop off, charged to 100%.  Powered on, unplugged laptop.  Battery ran down to 80% at which point it 'hibernated' (ran systemctl suspend).  Value currently set to "30" in auto-hibernate.conf.

http://pastebin.com/xMqmP0Pt

More logs incoming

rickysarraf, 2016-10-08:
I'll try to spare some time over this weekend to see if I can examine your logs.

But, meanwhile, are there good reasons why you are using such an old version ?

I checked the Gentoo page [1], and see Alon has kept maintaining LMT in the Gentoo repository, up-to-date. You should really upgrade to the latest version, 1.70, first to  verify if the newer release has fixed the problem.

[1] https://packages.gentoo.org/packages/app-laptop/laptop-mode-tools

rickysarraf, 2016-10-08:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

On Fri, 2016-10-07 at 15:23 -0700, carbohydratesn wrote:
> Well this is a fun journal;
> http://pastebin.com/XV4J1Gnb
> My /dev/fd does seem to be populated...but /proc/self/fd/* only has user
> rights. I'll try seeing if I can get a proper verbose log on my end, [...]

You definitely need to update your version of LMT to the latest. This issue was
known and fixed more than a year ago, in commit
id: d534b3a18e026cd00bd2ff51b3ebf0b6b725c58f, when we started working on
integrating LMT with systemd.

I will not recommend using LMT, at such old version, in a systemd environment.

- -- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."
-----BEGIN PGP SIGNATURE-----

iQIcBAEBCgAGBQJX+PLEAAoJEKY6WKPy4XVpvHYP/1LeVPMA28hR//nY6saXKLB9
ndzALrUhvMA7IMPPyeEjujWBsrsFwPJSu2IEiIqQQKaY+X4gJG7jQqtRoODVWcdF
+WeINZCAwTtiicDOr7ZJRGxo4GrOJwinGFLyV4ho1nppsi5wQCGorZyCiBmXc+Lk
XSBwVNjfcqtvt4Aj7ZhebweuGY3J34EZf9kQL1URQrPgeCPTukrF2clTbNVXOeCy
caQ+BTYAVtyrQ+vkgjxpjSHfYuV3fCOmm/ZgvbG85vFM3idArz5w7df15vJ4NwTY
JP5BXa9JXF3GNSn1t+TDAYk4qOdpq1XAkBDXope4+I+69y2waYsUwH9Ndcb+qM7Y
KB4V3FEhtUASnSW4rqjZLqqRo09BLJOiZOtXZFFUvowQivkr/C1i2xnL702GNujk
4JjXzmue0hjTqaH7ToVQZn+8h7r4r8hUTOvqeGYxWMUzYrDXbCZahrsVuD+Mz1+q
gSBAAGMGvlc04IcNpEbQIq2y8uyplkXU0lTQmTBK4wziW/cDfbjW0FpZFtA4KkUx
YxIzXw6RUZpdXzJhj4mQeI6woZieHdCFIstL40Bz0nxUPdzzC4eHOCcIq7q/Zywi
0oeikY2wWXnkMu9BqIVrMeOYPXvK/Bzt7wLTQBw7Cd5fh3bBXDDn/2Al93x33dUr
LfSzjqy0KIrUm7KIYSNl
=57GT
-----END PGP SIGNATURE-----

rickysarraf, 2016-10-08:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

On Fri, 2016-10-07 at 16:24 -0700, carbohydratesn wrote:
> I changed all references to /dev/fd to a normal file in my home dir, to get a
> proper log.
> Log 1 - Laptop off, charged to 100%. Powered on, unplugged laptop. Battery ran
> down to 80% at which point it [...]

I'll look at the code closely, but from the documentation, it is clear that it
calculates the trigger value as percentage of Full Capacity.

So, in your case, the math is:

(Remaining Charge / Full Capacity) * 100 = Trigger Percentage

(16110000 / 62160000) * 100 = 25.916988417 %

Which is below the below the threshold that you've specified for the trigger of
auto-suspend, i.e. 30%.

- -- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."
-----BEGIN PGP SIGNATURE-----

iQIcBAEBCgAGBQJX+PURAAoJEKY6WKPy4XVp/9gQAJMcKJLCZeWeFSpCtR5k8usA
HWLOuZxTuha3L3Vvmk7BH2TOktQuVpg+/zLL1pvN2Cb/TTFOIqudL2gWEadDAqAA
lSJuN3YkwCYIEKTG14B3a13fkAFniIUPnmyYMnKzKguapJqKk4WrK8aFZ3g8XExX
ioNMFS+rxeP23MyLIjg+4fiLdM/MWUy7RzxQFt724xdECgcgfG8A68OwR/cwFasR
laMz6ynrDzkw8C4+kD+fgBNE3TZoD2gfKQFDsyyr6iKNVyZC5bN7nsKPMPalDKvf
39vMb7mTS8EP4t8H5ISmK5WXQXsMOpUNUrcTTqVHGwlUU9eQ+8zEVZuDnao4rsWD
OPt/8RlpSnhjSXinbe81l1Cy3we1t2D3ltTyVd/K23IGCRpLqZ853eS5JM+BepYo
EKEEi7ijvS0qt4+QvLqWlpBg1UZQ/GqpaoMYXb+qt8NNYdchlYVSvop3FeDGNFCU
O2V2r/wSNyWtyGyevTipmb+gjwwUEXNQgNn4LhWVvO/a27p3xSLvaohIj8etuvSP
Usv13bLu/0yrO3XwBkqQGfndxc6VE32yYeLHl49Q5tRRxq74MfHcqEJKucU/RB/R
R7gNaNRePiD/sHRMc+3lHAqlv2NaFgyFtk9+h+4+q9RVNCSXIizgsF4SLEhTimaM
ft1qd8wMj6zZ3Dn8QM1v
=IEM9
-----END PGP SIGNATURE-----

rickysarraf, 2016-10-08:
Oh!! I just noticed that you and I have the same conclusion. I think we may now have a fix soon.

carbohydratesn, 2016-10-08:
Fantastic!  That sounds great - and I'm updating to 1.7, 1.66 is the latest available in Gentoo's stable repos.

Until that's fixed, I've set my threshold to "8", which results in me hibernating at 20-30%, which is perfect for now :)

rickysarraf, 2016-10-08:
Great. I still request you to provide me the verbose logs, after upgrading to 1.70.

rickysarraf, 2016-10-08:
Having thought of this bug, I have to say that there's no code/logic change needed, for LMT default values.

```
user@coniferous ~ $ sudo acpi -i
Battery 0: Discharging, 64%, 00:52:32 remaining
Battery 0: design capacity 5407 mAh, last full capacity 1743 mAh = 32%
```

Taking your example itself. We cannot rely on battery current charge because that keeps varying, depending on how you use your machine, how you charge the battery, battery quality and other factors.

And the results it'd provide would be plain wrong, practically. For example, 5% of 1743 mAh will be a terribly low value.

The configuration file does provide the flexibility to tweak the value according to your need. Such as your case, wherein if you have a degrading battery, you can always choose to tweak the percentage level.

I'm going to close this bug, but if you think otherwise, please do feel free to re-open it.

carbohydratesn, 2016-10-08:
Sorry, but I do think otherwise...if you're going to use a percentage of total capacity to trigger hibernation, it should be the **real** capacity of the battery, not a theoretical perfect, impossible capacity.  

I could understand your position if it was configured with a straight mAh value - "hibernate when there is 1000mAh left".  But it's not, it's configured with a percentage.  And that percentage is, currently, _wrong_.

> We cannot rely on battery current charge because that keeps varying, depending on how you use your machine, how you charge the battery, battery quality and other factors.

It does vary, that's true - but you _absolutely_ can rely on it, for _at least_ the duration of a session.  

And if you want an accurate 0-100 percentage, you **must** rely on it.  LMT **can't** be accurate, without taking that into account.  

Fully charged, my battery is 100% full - not 32% full.  

Currently, LMT does not agree.  

> And the results it'd provide would be plain wrong, practically.  For example, 5% of 1743 mAh will be a terribly low value.

5% of 1743mAh would indeed be tiny - but I don't want to hibernate at 5% of that charge.  I want to hibernate at 30%.  

The problem is that, right now, LMT is forcing me to _say_ I want 5%, in order to actually hibernate at 30%.  

THAT'S plain wrong.  It's just not correct.

>Such as your case, wherein **if you have a degrading battery,** you can always choose to tweak the percentage level.

That's not an "if", ALL laptop users have a degrading battery.  It's IMPOSSIBLE for a battery to maintain its full original capacity over time.

We shouldn't have to translate a percentage to get the correct behavior out of LMT.  

It's a percentage.  A 0-100 scale.  LMT should map that onto the real capacity of the battery.  **If you set a value of 30% in the config, it should hibernate at 30%, regardless of the age of the battery.**

Not somewhere between 30-100%, increasing rapidly as the battery ages, like it currently does.  That's nonsense!

carbohydratesn, 2016-10-08:
Hm...looks like I can't re-open this issue, should I open a new one focusing on the core of the problem?

The original issue was actually two issues - the "hibernates twice" behavior was coming from using 'systemctl hybrid-sleep' as my hibernate command.  I understand what's happening with both far better now.

rickysarraf, 2016-10-09:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

On Sat, 2016-10-08 at 15:59 -0700, carbohydratesn wrote:
> Hm...looks like I can't re-open this issue, should I open a new one focusing
> on the core of the problem?
> The original issue was actually two issues - the "hibernates twice" behavior
> was coming from [...]

Are you saying "hibernate twice" was caused by 'systemd hybrid-sleep' ?

- -- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."
-----BEGIN PGP SIGNATURE-----

iQIcBAEBCgAGBQJX+hSoAAoJEKY6WKPy4XVpIdgP+weP5dq5+4njI0HFpaksXi3q
cKNKmhiZx/6VUZF+ZeoE5mIS+poXBzAqlqIcS8KN/vIBfqUhGT29T5D1uP3MLgEg
APpp+73hVkNaeUcuC19BEsvX/SBsNCCEwL+Wn96T/0v0D/LgpGx4kh5Afc6lxgil
bWb+Xq1G003lwt3Qx/s2dNNWyBE4CGCzIj83YodmLBB/50MBeGHNOA1eqG2MQAQ5
+edQMyOYijuysOrC0d4kCWFv18y2XAAbksWarfYCXu6o1H+zuvoaQjjyHqIsT+Ys
IccW6mSKB/LXv4KRePAc20X1RgI0UyIm2nvC1Jsqx3HlV9bqwyq+NfDDEYeeTvwz
DePcGJqgNwuPZ/YLfa26XsgYRHJ+tCe5mKUxyo+6HF4lYn2DOrQmc8M6quLn5r9g
GJRNXtYvQ5mR2oyAeorqvYXzZzqDRW2YyemGUNUory+9iYhZsH5tnj1zT23aFNNX
xkg927lLrj5RK+enaKsvhn//Hw4ktJfvGX5mec/Hbrs/rZZDYhZONMclOLkGVBrN
qtgwlHh337e2HhyV/W3q8DWskh68xF1bGRuvnjKqbkzj9aMN03oot76NNHAUOU/t
F/U1XHFtaFFC5fUuwKeEeejZpHJEV78eUQDcemtSLUf07RyVvAT2htqqwrWp0thv
cIXXsvv4u2IiMmiNxWV6
=eIGu
-----END PGP SIGNATURE-----

rickysarraf, 2016-10-09:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

On Sat, 2016-10-08 at 15:05 -0700, carbohydratesn wrote:
> That's not an "if", ALL laptop users have a degrading battery. It's IMPOSSIBLE
> for a battery to maintain its full original capacity over time.
> We shouldn't have to translate a percentage to get the [...]

So how do you propose to fix this ?

Having 30% as the default is not correct, and just a waste of resource.
The default of 5%, which is when it triggers hibernation, works in both cases;
whether you have a degrading battery, or a brand new one.

- -- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."
-----BEGIN PGP SIGNATURE-----

iQIcBAEBCgAGBQJX+hWXAAoJEKY6WKPy4XVpk2gP/iW5wg1XDHHeZg+f8jIq6XZJ
gfl/1QEM5RVyu10rRtIl/8R4GQT47Par1l3e5hfd7geXF9x1WmDxTK4k5K/fS4d7
nF3e7nNhpqKeZnEOkjbJrcs2LrI1XpQSGvqmTzcD1yNecAyw4CiduwRmHVYhvlea
Lt02g+L0l9GfGLuDgsiV5MKPvs4wdX01fIXouWsMcah0r62c7lnfjd3hM26Pqkx9
Jjgf2G88rF0PsTQbg6lfSNBlRpdv1U7O/EOirGZfpZa5ZNw/92aOmxX1OEp1qn1j
zOZQyFQbujIRimT4ybM7sOs3XKe3LuO0i80wzpa9fSrX99jzm5ZYlfwYhsCkvFlg
29xMKlS2xy4ET18PpcV65xSwUfgJXVaqQvt53dpQI5YCixVmbNQMhOyLUoxF3OCR
dLL8HuvuvrwhYwvSMsljAWtHqCnYMDXvUgTyDbK5Rg3ajRcLqJvm9Qwtm7TSam+V
c0Hy/64qDt4pdZDTFsoOdgnQ4trww8Ygkn8HCOqon90h9bz3iwn14+mZ3pk/w+Y6
bcg55bwoi+CGS/FxOUD32UmodCnbH3PR+pLvT2itmBuHdKe70NMsnpA0SSIsw2oe
xIj+Ywy4LATAsyS/zL7BGfYOfi2BReBs7GD88zYISl9DTWobHBjdFoo/m+/0RoEP
QVhBnCfIEBBdio3afT2J
=qulw
-----END PGP SIGNATURE-----

rickysarraf, 2016-10-09:
I've reopened the issue.

rickysarraf, 2016-10-09:
It is accordingly documented in the manpage. Any other basis for determination of an ideal value for battery charge, ideas welcome.

```
   /etc/laptop-mode/conf.d/auto-hibernate.conf
       The auto-hibernate module allows you to automatically hibernate  your  computer  when
       the battery goes critical or when the battery level goes below a certain threshold.

       ENABLE_AUTO_HIBERNATION
                 When  this option is enabled, laptop mode will automatically hibernate your
                 computer when the battery level reaches a certain  configurable  threshold.
                 This feature is only available when ACPI is enabled.

       HIBERNATE_COMMAND
                 This  option  specifies  the  command  that laptop mode should execute when
                 auto-hibernation is triggered. Normally, this  is  set  to  something  like
                 "/usr/sbin/hibernate".

       AUTO_HIBERNATION_BATTERY_CHARGE_PERCENT
                 The  battery level threshold for auto-hibernation, as a percentage of total
                 battery capacity.

       AUTO_HIBERNATION_ON_CRITICAL_BATTERY_LEVEL
                 When this option is enabled, auto-hibernation will kick in when the battery
                 reports its state as "critical".

```

carbohydratesn, 2016-10-09:
I beg your pardon?  The documentation is fine, it's the method of calculating that percentage that's the problem.

>So how do you propose to fix this ? Having 30% as the default is not correct,

I'm not talking about setting another default...it seems like you're REALLY misunderstanding me here.  

-

Right now, LMT is using the "design capacity" for its calculations.  This results in unreliable, confusing behavior.  If you set a value in the config, right now, it doesn't hibernate at that percentage.  It hibernates somewhere between that percentage and 100%.  

Right now, if I set 30, it triggers at 80.  I set 8, and it triggers at 30.  That's inconsistent and confusing.  And it's because of the max capacity value LMT Is using.  The "design capacity" is the wrong value to use.  It's not accurate or correct.

Using the "last full capacity" instead would give the correct behavior.  If you set a % value in the config, it would actually hibernate **at that value**.  If you set 30, it would trigger at 30.  If you set 8, it would trigger at 8.  Things would be consistent, and correct.

The documentation is fine, the default is fine.  I'm not even talking about those.  It's the numbers behind calculating the current percentage that are the problem.

Do you understand the difference between using the "design capacity" vs. 'last full capacity'?  And how they fit into LMT's calculations?

carbohydratesn, 2016-10-10:
After playing around with my /usr/sbin/laptop_mode for a bit, I have submitted two pull requests - one for the core of this issue, and another to correct 'jumping the gun' on assuming my battery is broken.

https://github.com/rickysarraf/laptop-mode-tools/pull/77

https://github.com/rickysarraf/laptop-mode-tools/pull/76

I've done several hours of testing on these changes myself, and everything seems to be working perfectly now - so I hope you will consider accepting them!  I would love to be able to keep using LMT, it's a great tool.

If you do some real-world testing with hibernation behavior, especially with a battery that has lost some of its capacity, I'm sure you will see that pull #77 makes LMT's hibernation behavior much more consistent. 

It hibernates _at the specified percentage_ now :grinning:

rickysarraf, 2016-10-16:
I'm closing this now, with the justification in the comments of the pull request.

closed 2016-10-16 by rickysarraf

wolterhv, 2021-05-03:
I know this bug is long closed, but I share @carbohydratesn 's point of view. Nobody keeps track of the instantaneous charge in their batteries. Users of battery-powered devices largely track only their battery's SOC (state-of-charge). This means, SOC becomes the reference for getting a notion of the energy left in the battery.

You mentioned earlier 

> I don't like the idea to base assumptions on a moving target.

If your reference is battery SOC, which is the case for most users, then battery design charge capacity _becomes_ a moving target. In this scenario, users are therefore forced to somehow measure the percentage of remaining charge w.r.t. battery design charge capacity, in order to tell whether their computer is likely to auto-hibernate soon or not. Power users can do this easily by checking `/sys/class/power_supply/...`, but non-power users, who rely on the stats their desktop-environment gives them, do not have a way of measuring this.

I understand that you are inclined to have this functionality, but what do you say about adding a new configuration option, perhaps something called `AUTO_HIBERNATION_BATTERY_CHARGE_REFERENCE` that can be set to `SOH` or to `design`, where `SOH` means "state-of-health" and `design` means "design charge capacity"? 

With this new setting set to `design`, everything remains unaffected, but with the setting set to `SOH`, auto-hibernation will trigger when SOC equals `AUTO_HIBERNATION_BATTERY_CHARGE_PERCENT`, which is what @carbohydratesn and I prefer.

rickysarraf, 2021-05-17:
> I know this bug is long closed, but I share @carbohydratesn 's point of view. Nobody keeps track of the instantaneous charge in their batteries. Users of battery-powered devices largely track only [...]

Please propose a patch and then we can look at it, again. This is already 5 years old and these days I don't have the same luxury in time. 

But as long as your proposed patch does not change the default behavior, I don't see any strong reason to not add it.