rickysarraf/laptop-mode-tools #79: GUI fails to start: "name 'os' is not defined"

sebastianst, 2016-10-25:
Upon starting the gui, I get the following error:
```
$ lmt-gui            
Using PyQt4
Traceback (most recent call last):
  File "/usr/bin/lmt-gui", line 31, in <module>
    os.putenv('QT_X11_NO_MITSHM', "1")
NameError: name 'os' is not defined
```
I am on Arch Linux and installed LMT via the [AUR package](https://aur.archlinux.org/packages/laptop-mode-tools/). The optional dependency `python2-pyside` is also installed.

rickysarraf, 2016-10-25:
It is fixed in my repo and slated for next release. You can check in the
git repository. It is a one line fix.

I'm sorry I'm unable to pinpoint the fix, because my laptop charger went
bad today and my laptop has drained out

s3nt fr0m a $martph0ne, excuse typ0s

rickysarraf, 2016-10-27:
This commit: https://github.com/rickysarraf/laptop-mode-tools/commit/976be20c0f5bc8078952c0a9a0661c8fb936f994 fixes the problem.

rickysarraf, 2017-01-12:
1.71 is released now.

closed 2017-01-12 by rickysarraf