rickysarraf/laptop-mode-tools #81: nouveau pstate

rshadow, 2016-11-30:
Please add power state for nouveau nvidia video cards.

This is my pstate:

```shell
# cat /sys/kernel/debug/dri/1/pstate
07: core 270-405 MHz memory 838 MHz AC DC *
0a: core 270-900 MHz memory 1560 MHz
0e: core 270-900 MHz memory 4000 MHz
0f: core 270-900 MHz memory 5016 MHz
AC: core 405 MHz memory 837 MHz
```
To set minimal power I exec:

```shell
echo 07 > /sys/kernel/debug/dri/1/pstate
```

Note:
 * /1/ is not fixed. It is a card number and may be any number
 * I think other cards have own minimal code, not a `07`
 *  Documentation available in https://nouveau.freedesktop.org/wiki/KernelModuleParameters/ `pstate` parameter
* for 4.5+ kernek debugfs need to be mounted like:

```fstab
 none /sys/kernel/debug debugfs defaults  0       0
```

rickysarraf, 2016-11-30:
Thanks. I'll look into it. But I do not have this hardware, so I'm going to rely on you for the initial testing.

Looking at the docs, what about runpm. Does that too have an interface in debugfs ?

rshadow, 2016-11-30:
Yes I make testing on my laptop.

> runpm

removed in 4.5

```shell
$ uname -a
Linux rubin-laptop 4.8.0-1-amd64 #1 SMP Debian 4.8.7-1 (2016-11-13) x86_64 GNU/Linux
```

```shell
# ls /sys/kernel/debug/dri/1/
clients  DP-1  DP-2  eDP-1  gem_names  HDMI-A-1  name  pstate  vbios.rom
```
I think it is not necessary to implement the deprecated parameters.

rshadow, 2016-11-30:
`runpm` still exists in kernel module:

```shell
 # modinfo nouveau
...
parm:           tv_norm:Default TV norm.
                Supported: PAL, PAL-M, PAL-N, PAL-Nc, NTSC-M, NTSC-J,
                        hd480i, hd480p, hd576i, hd576p, hd720p, hd1080i.
                Default: PAL
                *NOTE* Ignored for cards with external TV encoders. (charp)
parm:           vram_pushbuf:Create DMA push buffers in VRAM (int)
parm:           nofbaccel:Disable fbcon acceleration (int)
parm:           tv_disable:Disable TV-out detection (int)
parm:           ignorelid:Ignore ACPI lid status (int)
parm:           duallink:Allow dual-link TMDS (default: enabled) (int)
parm:           hdmimhz:Force a maximum HDMI pixel clock (in MHz) (int)
parm:           config:option string to pass to driver core (charp)
parm:           debug:debug string to pass to driver core (charp)
parm:           noaccel:disable kernel/abi16 acceleration (int)
parm:           modeset:enable driver (default: auto, 0 = disabled, 1 = enabled, 2 = headless) (int)
parm:           runpm:disable (0), force enable (1), optimus only default (-1) (int)
```

%)

rickysarraf, 2016-12-20:
The values that are presented, are they uniform for all class of devices of the nouveau family ?
The documentation on the fd.o page does not talk much about it. I can't find any documentation even for the 'AUTO' value that you've mentioned.

```
# cat /sys/kernel/debug/dri/1/pstate
07: core 270-405 MHz memory 838 MHz AC DC *
0a: core 270-900 MHz memory 1560 MHz
0e: core 270-900 MHz memory 4000 MHz
0f: core 270-900 MHz memory 5016 MHz
AC: core 405 MHz memory 837 MHz
```

rshadow, 2016-12-30:
>  I can't find any documentation even for the 'AUTO' value that you've mentioned.

I found it somewhere in internet =) I try it and it`s work.

rshadow, 2016-12-30:
> The values that are presented, are they uniform for all class of devices of the nouveau family ?

I don`t know. But I think lower mode in the top of list.

```shell
# cat /sys/kernel/debug/dri/1/pstate                                      1 
07: core 270-405 MHz memory 838 MHz AC DC *
0a: core 270-900 MHz memory 1560 MHz
0e: core 270-900 MHz memory 4000 MHz
0f: core 270-900 MHz memory 5016 MHz
AC: core 405 MHz memory 837 MHz
```

`AC` not valid:

```shell
# echo AC > /sys/kernel/debug/dri/1/pstate                            
echo: write error: invalid argument
```

`any string` not valid:

```shell
# echo SOME > /sys/kernel/debug/dri/1/pstate                          
echo: write error: invalid argument
```

`hex` is valid:

```shell
# echo 0e > /sys/kernel/debug/dri/1/pstate
# cat /sys/kernel/debug/dri/1/pstate                                        
07: core 270-405 MHz memory 838 MHz
0a: core 270-900 MHz memory 1560 MHz
0e: core 270-900 MHz memory 4000 MHz AC DC *
0f: core 270-900 MHz memory 5016 MHz
AC: core 405 MHz memory 5016 MHz
```

`AUTO` is valid - set maximum:
```shell
# echo AUTO > /sys/kernel/debug/dri/1/pstate
# cat /sys/kernel/debug/dri/1/pstate                                        
07: core 270-405 MHz memory 838 MHz
0a: core 270-900 MHz memory 1560 MHz
0e: core 270-900 MHz memory 4000 MHz
0f: core 270-900 MHz memory 5016 MHz *
AC: core 405 MHz memory 5016 MHz AC DC
```

rickysarraf, 2016-12-30:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

On Fri, 2016-12-30 at 07:01 -0800, Roman V. Nikolaev wrote:
> I don`t know. But I think lower mode in the top of list.
> # cat /sys/kernel/debug/dri/1/pstate                                      1 
> 07: core 270-405 MHz memory 838 MHz AC DC *
> 0a: core 270-900 MHz memory 1560 MHz
> 0e: core 270-900 MHz memory 4000 MHz
> 0f: core 270-900 MHz memory 5016 MHz
> AC: core 405 MHz memory 837 MHz
> AC not valid:

I was so hoping to use the 1st column. But I'm not confident of that now.
Let me investigate more on this.

- -- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."
-----BEGIN PGP SIGNATURE-----

iQIzBAEBCgAdFiEEQCVDstmIVAB/Yn02pjpYo/LhdWkFAlhmjIcACgkQpjpYo/Lh
dWkXTg/8COXHVvfcSrkrme5PAQvPKpYSeE3uvdEcgvQsG6dJ74goNp7Oo78dT+zt
V2+wjOxQXfbtuhclL8POcMN8AviyOBoXvmwlRfALF2wH6Tt9TFaCvPfo3Kvcj9h9
CirxrPcjHI7RNCiE56xdsushFTl+dwY763QyM4xxARShp927a+Qro30SXzEpy8vo
cFEGl0RkQoRUsap89S9yqXaU1lzWpKvW4WyXLpo7PPMtHIINbowVqOtGCS2MHY/F
DCJOZ+Qyjwyf92xVZ3d6MVuD4O7VXEoHn1EYVuDuM+smFxVWdaFbznG5fHrsVgFG
Zom+k0V2Qp5X1Wa9sgVT5nInBMZlUpuA31/ZlGvKIppCyuuINeKFtDqRlPGNT1Gj
2fNkry26LD9IPrS3B310JIeU4lbRP6wJwst8XgYEBemXvjs1nDypmG8vEltJL0ao
+fWxTlXe9iB5jnWcNBXkNTGBhA8Ia7QSZcT4W6PUxRndU7MHUBW891JBP2bD7A+H
ynZhTsYC5B3B2Z7IVgcgwYykxajw2wxWzukMOmhOxZXI8j3d/3Oer0tAv2YGXKG5
tPZHOxs6x4LyDMyQPOziowO57sxTunTkOvmzz6AmQwaXcz5embAkfaLzn6yKknev
FRLnMZxXen5l0XgJyDITupNpV2Fg8PqPolSrBkTQ3fDYHKN7zEY=
=SP6u
-----END PGP SIGNATURE-----

hno, 2018-10-10:
Stumbled across this in a related search. It's been some time since the issue were opened but if it still is valid then here is what I have found about pstate:

The first column is the pstate value, followed by clocking information, AC / DC selection flags and * if it's the current pstate. Except for the last line where the first column indicates the current power source (AC or DC) and the rest of the line indicates the current clocking.

Syntax for setting pstate is
[ac:][dc:]<hexvalue>
If neither ac: or dc: is used then the value is applied to both power sources.
If either ac: or dc: is used then the value only applies to that power source
The special values "auto" and "none" means
auto: Let the GPU automatically tune pstate. Seems to always set to highest performance.
none: No pstate configured for the power source. I.e. keep current settings.

brunoais, 2020-05-29:
@rickysarraf Any updates on the investigation?

rickysarraf, 2020-05-30:
@brunoais @hno @rshadow 

Can you guys review and test this code in your setup with nvidia/nouveau cards ?
https://github.com/rickysarraf/laptop-mode-tools/pull/157

brunoais, 2020-05-30:
I may find time next weekend. I have this weekend completely full