rickysarraf/laptop-mode-tools #82: vgaswitcheroo

rshadow, 2016-11-30:
Please add ability to disable secondary card in [Optimus](https://nouveau.freedesktop.org/wiki/Optimus/) systems.

Example of dowble video cards:

```shell
# cat /sys/kernel/debug/vgaswitcheroo/switch                                
0:IGD: :Pwr:0000:00:02.0
1:DIS:+:DynPwr:0000:01:00.0
2:DIS-Audio: :Pwr:0000:01:00.1
```

IGD - Integrated Intel videocard
DIS - Nouveau Nvidia discrete videocard

Disable secondary videocard:

```shell
echo OFF > /sys/kernel/debug/vgaswitcheroo/switch
```
```shell
# cat /sys/kernel/debug/vgaswitcheroo/switch                                
0:IGD: :Off:0000:00:02.0
1:DIS:+:DynPwr:0000:01:00.0
2:DIS-Audio: :Pwr:0000:01:00.1
```

Enable secondary videocard:

```shell
echo ON > /sys/kernel/debug/vgaswitcheroo/switch
```

Note:
 * A good idea not just disable secondary card, but switch on integrated. But it`s buggy and broken in new kernels. Simple power off secondary card is stable.

rickysarraf, 2016-11-30:
Is vgaswitcheroo feature specific to nvidia/nouveau ? If so, we could merge this into #81

rshadow, 2016-11-30:
> Is vgaswitcheroo feature specific to nvidia/nouveau

No. This is a way to switch between any of the graphics.

rshadow, 2016-11-30:
More docs: https://01.org/linuxgraphics/gfx-docs/drm/gpu/vga-switcheroo.html

rickysarraf, 2016-12-19:
Thanks @rshadow 

```
Manual switching and manual power control

In this mode of use, the file /sys/kernel/debug/vgaswitcheroo/switch can be read to retrieve the current vga_switcheroo state and commands can be written to it to change the state. The file appears as soon as two GPU drivers and one handler have registered with vga_switcheroo. The following commands are understood:

    OFF: Power off the device not in use.
    ON: Power on the device not in use.
```

So, let's simply just have on/off. As it says above, "device not in use" should suffice the sane default power saving requirement.

rickysarraf, 2016-12-19:
Also to keep in my is the fact that the interface will not be writable when runtime pm for the driver is active, which is the default case.

```
Driver power control

In this mode of use, the discrete GPU automatically powers up and down at the discretion of the driver’s runtime pm. On muxed machines, the user may still influence the muxer state by way of the debugfs interface, however the ON and OFF commands become a no-op for the discrete GPU.

This mode is the default on Nvidia HybridPower/Optimus and ATI PowerXpress. Specifying nouveau.runpm=0, radeon.runpm=0 or amdgpu.runpm=0 on the kernel command line disables it.
```

closed 2016-12-20 by rickysarraf