rickysarraf/laptop-mode-tools #83: iwconfig txpower

rshadow, 2016-11-30:
By default `txpower` is `off`. Set it to `auto`:

```shell
iwconfig wlan0 txpower auto
```

Disable power control:
```shell
iwconfig wlan0 txpower on
```

For more info see `man iwconfig`

rickysarraf, 2016-11-30:
Thanks. THis looks good.

Note to self: Also check the equivalent in iw.

rickysarraf, 2016-12-19:
auto and fixed are documented as txpower on/off. And so does the below output reflect.

```
rrs@learner:~/rrs-home/devel/Laptop-Mode-Tools/laptop-mode-tools (lmt-upstream)$ sudo iwconfig wlan0 
wlan0     IEEE 802.11  ESSID:"FTW"  
          Mode:Managed  Frequency:2.472 GHz  Access Point: C4:6E:1F:D0:67:26   
          Bit Rate=150 Mb/s   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr=2347 B   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=70/70  Signal level=-36 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:12   Missed beacon:0

2016-12-19 / 18:52:32 ♒♒♒  ☺  

rrs@learner:~/rrs-home/devel/Laptop-Mode-Tools/laptop-mode-tools (lmt-upstream)$ sudo iwconfig wlan0 txpower fixed
2016-12-19 / 18:52:51 ♒♒♒  ☺  

rrs@learner:~/rrs-home/devel/Laptop-Mode-Tools/laptop-mode-tools (lmt-upstream)$ sudo iwconfig wlan0 
wlan0     IEEE 802.11  ESSID:"FTW"  
          Mode:Managed  Frequency:2.472 GHz  Access Point: C4:6E:1F:D0:67:26   
          Bit Rate=150 Mb/s   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr=2347 B   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=66/70  Signal level=-44 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:12   Missed beacon:0

2016-12-19 / 18:52:54 ♒♒♒  ☺  
```

closed 2016-12-19 by rickysarraf