rickysarraf/laptop-mode-tools #84: more customizable cpuhotplug

rshadow, 2016-11-30:
Please make it possible to specify values for each state. Something like:

```conf
BATT_CPU_HOTPLUG="1"
LM_AC_CPU_HOTPLUG="3quarter"
NOLM_AC_CPU_HOTPLUG="all"
```

instead of using only one mode.

rickysarraf, 2016-11-30:
That's nice to have feature. But I'm marking it as wishlist for now. If you have a patch, I'll be glad to review.

rickysarraf, 2016-12-19:
Having thought a little more to implement, I can't see any value this would bring.
It makes no sense to have lower number of CPUs enabled, when running on AC. The only exception I see is people who run LMT on machines, as a Jukebox. In that case too, something like the following will work.

```
LM_AC_CPU_HOTPLUG=0
DISABLE_AVAILABLE_CPU="3quarter"
```

Which means that, when we are on AC power, and LMT is enabled on AC, set the number of disabled CPUs to 3/4 of the total.

I don't see any logical value in what you have asked, even though it is not incorrect. I'm marking this closed because I don't see any functionality missing, which can't be achieved with the module in its current form.

If you have a use case, which I have completely missed, please feel free to re-open.

closed 2016-12-19 by rickysarraf

rshadow, 2016-12-30:
> If you have a use case, which I have completely missed, please feel free to re-open.

I turn off some cores because 8 is too match for simple work/browsing. My main goal is less to notebook warmed and burnt knees ;-) Himself resolutely need is for laptop-mode-tools.

rickysarraf, 2016-12-30:
For the requirement you've mentioned (disabling CPU cores under conditions), the cpuhotplug module in its current form, does it well.

For example:

```
DISABLE_AVAILABLE_CPU="half"

# Conditions under which to enable CPU Hot Plugging, i.e. shutdown of the CPU 
BATT_CPU_HOTPLUG=1
LM_AC_CPU_HOTPLUG=1
NOLM_AC_CPU_HOTPLUG=0
```

Would mean to disable half of all cores, when on battery. And also on AC in LM is enabled.

rshadow, 2016-12-30:
Yes I known. Thanks.