rickysarraf/laptop-mode-tools #85: keyboard backlight

rshadow, 2016-11-30:
Modern laptops have keyboard brightness. Please add analogue (really duplicate) of `lcd-brightness` module.

MacBook Pro example:

```shell
ls /sys/class/leds/smc::kbd_backlight/
brightness  device  max_brightness  power  subsystem  trigger  uevent
```

```shell
cat /sys/class/leds/smc::kbd_backlight/max_brightness                     
255
```

```shell
cat /sys/class/leds/smc::kbd_backlight/brightness                         
0
```

rickysarraf, 2016-12-01:
Hmmmm. THis is interesting. On my Lenovo Yoga 2 13, I don't have the backlight module available. My hardware does have keyboard backlight.

This needs more investigation for a generic way, covering most of the hardware.

```
rrs@learner:~/rrs-home/devel/Laptop-Mode-Tools/laptop-mode-tools (lmt-upstream)$ ls /sys/class/leds/
input0::capslock@  input0::numlock@  input0::scrolllock@  mmc0::@
2016-12-02 / 00:01:51 ♒♒♒  ☺  
```

rickysarraf, 2016-12-01:
pfff. on lenovos, only through tpacpi, which would only cover the TP product line. Sigh.

http://askubuntu.com/questions/383501/enable-the-keyboard-backlights-on-supported-lenovo-e-g-carbon-x1-with-command

And, as is shown here in the logs, there is no nice way to determine the device's type other than depending on name scanning, which won't work well. TP already have a different naming convention.

Heh.. And the most funny is my (b0rken) MMC slot shown under this subsystem.

rshadow, 2016-12-01:
> MMC

Oh ...

```shell
$ ls /sys/class/leds/ 
input2::capslock  input2::kana     input2::scrolllock  smc::kbd_backlight
input2::compose   input2::numlock  mmc0::
```

rshadow, 2016-12-01:
> generic way
`lcd-brightness` does not have a generic way.

Some kernel docs: https://www.kernel.org/doc/Documentation/leds/leds-class.txt

rickysarraf, 2016-12-02:
Hmmm. Good point. So let's just make the user define the explicit path, just like lcd-brightness. You already mentioned that in the beginning. Thanks. :-)

rickysarraf, 2016-12-19:
@rshadow I don't have the support in my hardware. So, can you test and confirm if it works find ?

closed 2016-12-19 by rickysarraf