rickysarraf/laptop-mode-tools #86: i915

rshadow, 2016-12-14:
Intel chipset have own powersaving features.

Module parameters example:

```shell
# modinfo i915 | grep power                                                                                                                                  U[desktop]
parm:           enable_rc6:Enable power-saving render C-state 6. Different stages can be selected via bitmask values (0 = disable; 1 = enable rc6; 2 = enable deep rc6; 4 = enable deepest rc6). For example, 3 would enable rc6 and deep rc6, and 7 would enable everything. default: -1 (use per-chip default) (int)
parm:           enable_dc:Enable power-saving display C-states. (-1=auto [default]; 0=disable; 1=up to DC5; 2=up to DC6) (int)
parm:           enable_fbc:Enable frame buffer compression for power savings (default: -1 (use per-chip default)) (int)
parm:           disable_power_well:Disable display power wells when possible (-1=auto [default], 0=power wells always on, 1=power wells disabled when possible) (int)
parm:           edp_vswing:Ignore/Override vswing pre-emph table selection from VBT (0=use value from vbt [default], 1=low power swing(200mV),2=default swing(400mV)) (int)
```

Enable:
```shell
# echo 7 > /sys/module/i915/parameters/enable_rc6
```

Disable:
```shell
# echo -1 > /sys/module/i915/parameters/enable_rc6
```

Similarly, for all parameters.

rickysarraf, 2016-12-19:
The example, that you've quoted, is not writable. Not all attributes of that module allow a value to be written. Can you please verify ?

rshadow, 2016-12-30:
Yes. Only `enable_fbc` accesible. Perhaps this is not the parameters that can be changed.
I'm sorry for your attention.

rickysarraf, 2016-12-30:
Nothing to be sorry about.

And enable_fbc is not a safe option to play with. The upstream folks intend to enable it, by default, only from Linux 4.10+. And that too for Skylake and above, IIRC.

Generally, most i915 module parameters are such. But anyways, given that most parameters are static, there's not much we can do here. Hence closing.

closed 2016-12-30 by rickysarraf