rickysarraf/laptop-mode-tools #88: Add delay before checking battery status

qwence, 2016-12-31:
I noticed laptop-mode-tools was not activating when I pulled the AC adaptor, and needed to be invoked manually (but it worked when I plugged in the laptop). The problem seems to be that /usr/bin/laptop_mode is checking battery status immediately after the unplug event, but on my laptop (macbook pro, late 2013), /sys/class/power_supply/BAT0/status takes a half second or so to reflect this. Adding `sleep 0.5` after line 466 of /usr/bin/laptop_mode fixes this for me, but I'm not sure what the "right" way to handle this would be.

rickysarraf, 2016-12-31:
My impression so far is that the delay in updates in sysfs is only during boot time. You are the first reporter to report delays during AC plug/unplug. Are you confident of the delay in update ?

Maybe you should start with verbose logging in LMT. That may give some more information on what is happening. Also, you should capture event logs from udev when doing plug/unplug. Use something like, udevadm to monitor for all events.

I've never owned a Mac. Can you verify with other Mac users, to see if this issue is a one off case, or a generic issue with Macs ?

qwence, 2016-12-31:
My bad, I neglected to check the git version before reporting -- this seems to have been fixed after the latest release. Thanks / sorry for the noise.

(For the record, the battery status (after removing the AC adapter) was Full but in the new version it concludes "In Battery state, but battery does not report discharge".)

rickysarraf, 2016-12-31:
Great thanks for the update. One more reason to start prepare the 1.71 release soon.

closed 2016-12-31 by rickysarraf