rickysarraf/laptop-mode-tools #90: Suspend issue

mishkadoma, 2017-01-17:
Every time i'm using your tools i'm getting the infinity hibernate(suspend option in ubuntu 16.10) issue . At the first time everything is fine(power light is off as it might), waking up as always. But 2nd time(always second) my laptop getting freeze(power light is ON these times, screen is OFF. laptop isn't responde), so i need to  turn it off via power button and then boot normally . i don't want to use any other tools like TLP, because they're not so efficient. With this one my laptop lifetime is gorgeous - about 11 hours compared to ordinary 7.
Laptop model is: acer v3-372-5639. 
If you want to get any other information, feel free to ask.

rickysarraf, 2017-01-18:
Hello Mikhail,

Thank you for the bug report. Can you please provide more details ?

* What version of Laptop Mode Tools are you using ?
* Can you check and confirm if the auto-hibernate module is enabled ? You can check it in `/etc/laptop-mode/conf.d/auto-hibernate.conf`

As I understand your bug report, your laptop goes into infinite sleep cycle, after the first sleep cycle ? On fresh boot, you do not have an automatic suspend being triggered. But after the first suspend, upon resuming, you laptop immediately goes into suspend again. Correct ?

mishkadoma, 2017-01-18:
i do not know initial version, just got it from official rep
`apt-get install laptop-mode-tools`
i think it's pretty old there. 
but after i got a freeze and then delete the package i've downloaded 1.71 version from here and try to install it through `sudo ./install.sh`
[image]
i'm not sure about these messages besause acpid seems installed. 
the result of 
`sudo laptop_mode`
is 
`enabled, active [unchanged]`
but the result of 
`sudo lmt-config-gui`
is blank, nothing.
then i manually put LMT.py and lmt-config-gui in a `usr/share/laptop-mode-tools`, try to run them but it missing some python packages. so i've executed
`apt-get install laptop-mode-tools`
again and choose to leave all new configs at the place. 
and now i can run gui and everything works except suspend option.

checked the auto-hibernate.conf:
`ENABLE_AUTO_HIBERNATION=0`

>But after the first suspend, upon resuming, you laptop immediately goes into suspend again.

after 1st suspend , when i'm touching touchpad or pressing any button, or open the lid , system works fine. the trouble happens only at 2nd suspend. so i can't wake laptop with any of the options i've mentioned. it's not responding to any of them, all i can do is reboot via power button long press. And in addition: the laptop has kinda power indicator, and when it's normally hibernate this light is turned OFF, when laptop is active it is ON, and when that trouble goes, the light is ON too. that is weird.

mishkadoma, 2017-01-19:
i got a news. just now i used powertop and consequently enabling options one after another, and found this one when enable causing the issue i'd desctibe. It is highlighted.
[image]
any thoughts about that?

rickysarraf, 2017-01-19:
Okay! THis information is good. So you are on a Debian based Linux distribution. In this case, I'd recommend you to try out the latest of Laptop Mode Tools using the .deb package, from:
https://packages.debian.org/sid/all/laptop-mode-tools/download

Since you already installed using the source, you should first clean your setup manually.

The device not resuming on 2nd wakeup is clearly a case of Kernel Bug. You should report it upstream, mentioning that when enabling Runtime PM, you've run into this state.

rickysarraf, 2017-01-19:
Regarding your 2nd comment, I'm glad you found out the culprit. Now that you know the offending device, you can Blacklist it under Laptop Mode Tools to not get into power saving mode. To do that, 

In the screenshot above, it is a PCI device. To blacklist a PCI device, run `lspci` to get the devices ID.

For example, let's take this device `00:1c.0 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 4 (rev e4)`

```
rrs@learner:/tmp$ lspci
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 0b)
00:14.0 USB controller: Intel Corporation 8 Series USB xHCI HC (rev 04)
00:16.0 Communication controller: Intel Corporation 8 Series HECI #0 (rev 04)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 4 (rev e4)
00:1d.0 USB controller: Intel Corporation 8 Series USB EHCI #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation 8 Series LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 8 Series SATA Controller 1 [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 8 Series SMBus Controller (rev 04)
01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8723BE PCIe Wireless Network Adapter
2017-01-19 / 16:12:05 ♒♒♒  ☺  
```

The ID for the device is `00:1c.0`. Now we look into sysfs for PCI devices.

```
rrs@learner:/tmp$ ls /sys/bus/pci/devices/
0000:00:02.0@  0000:00:14.0@  0000:00:1b.0@  0000:00:1d.0@  0000:00:1f.2@  0000:01:00.0@
0000:00:03.0@  0000:00:16.0@  0000:00:1c.0@  0000:00:1f.0@  0000:00:1f.3@
2017-01-19 / 16:13:37 ♒♒♒  ☺  
```

So our full device ID for the `00:1c.0` device is: `0000:00:1c.0`

So you take that value (`0000:00:1c.0`) and add it to `/etc/laptop-mode/conf.d/runtime-pm.conf` under `AUTOSUSPEND_RUNTIME_DEVID_BLACKLIST=`

Example: `AUTOSUSPEND_RUNTIME_DEVID_BLACKLIST="0000:00:1c.0"`

With this setting in place, Laptop Mode Tools will ensure to not enable power savings for that particular device, which in your case, hopefully, solved the suspend hang issue.

Note: This is still a kernel bug and perhaps you should report it to the kernel mailing lists.

rickysarraf, 2017-01-19:
@mishkadoma Mikhail, please confirm on the steps and the result. Based on your confirmation, I'll close this bug later.

rickysarraf, 2017-01-23:
I'm closing this report. If you run into the problem, and find it to be a Laptop Mode Tools issues, please feel free to re-open this report.

closed 2017-01-23 by rickysarraf