rickysarraf/laptop-mode-tools #91: Problem with detecting battery and AC logic

IyadKandalaft, 2017-02-17:
The variable FOUND_SYS_CLASS_POWER_SUPPLY_AC is the laptop_mode script is prematurely set to 1 on line 456.  This causes the if statement on 487 to be bypassed and therefore laptops where a battery is present and charging but the AC adapter is not detected will not be put into ON_AC=1.

IyadKandalaft, 2017-02-17:
#92 is the proposed fix.
Note that this was tested on a Dell XPS 9550 with kernel 4.4.  On these laptops, the AC adapter is not detected by kernel 4.4 but the battery is and shows whether it is charging or not.  When the /sys/class/power_supply/battery is **not** discharging, then the laptop is definitely plugged in (could be charging or stops charging at <=100% depending on the BIOS setting).

rickysarraf, 2017-02-18:
I am not excited about working around broken power_supply drivers in Linux. If it was a timing issue, I may consider seeing if we can optimize our checks. But this, from your description, seems to be a clear case of kernel bug, which you should report upstream (or to your distro vendor).

The above said, please provide full details for a bug report. Distro, LMT version, Logs.
Also, looking at: https://github.com/rickysarraf/laptop-mode-tools/blob/lmt-upstream/usr/sbin/laptop_mode#L454, looks like your version is an outdated one.

I am closing this bug report, but please feel free to re-open if you think the bug is a valid one.

rickysarraf, 2017-02-18:
Please look at commit: 0df292d5e9a63552726de43df45bd01622f08e3d

IyadKandalaft, 2017-02-18:
That commit (0df292d) introduced a bug where LMT assumes that all ACPI devices, including batteries, are like an AC power supplies and mistakenly sets ON_AC=1 when only a battery was found.  Hence, even if the battery is discharging, then laptop mode tools thinks there's AC power.  The pull request I provided is one way to fix this issue and merges 2 independent IF statements into more manageable code.  

Distro: Ubuntu 14.04 with HWE stack (not really relevant)
Kernel: 4.4
LMT:  1.71 (ff6362fcdffdce73da6a282eb30debc609036cdd)

rickysarraf, 2017-02-19:
Let's look at it again then. Please also look at: f47bee1c47899f02a11ae9f73176eee2161cb18e on the recent changes.

The (broken) Battery status needs to be handled carefully. What does your BATT/status file report the status as, when power cable is plugged in and otherwise? I am curious what is not working in your case, given that battery is detected in power_supply driver.

Also, I got confused with the line no. references you made, which are not inline with the upstream revision.

The way we determine power status involves broken components, both power and battery driver. In the newer revision, we try harder to conclude the machine's state. So I think the file you may have been referring to may be an outdated revision. Can you please check the latest at:
https://github.com/rickysarraf/laptop-mode-tools/blob/lmt-upstream/usr/sbin/laptop_mode

And better, run LMT in debug mode to know what is going on.

I am reopening this report bug need more details on what is going on.

rickysarraf, 2017-02-19:
Easiest would be if you enable VERBOSE_OUTPUT and only extract journal logs for unit laptop-mode.

IyadKandalaft, 2017-02-21:
Sorry, my comment was referencing wrong line numbers (i was distracted when I put in the issue).  

Before I post the VERBOSE_OUTPUT, take a look on the latest commit (ff6362f) of lmt-upstream in  
/usr/sbin/laptop_mode and walk through the code.

Variables get instantiated:
L454: 	FOUND_SYS_CLASS_POWER_SUPPLY_AC=0
L456: 	BATTERY_NOT_DISCHARGING=0

Then iterate over /sys/class/power_supply and set FOUND_SYS_CLASS_POWER_SUPPLY_AC=1 whether it's a battery or an AC power supply (L459).  If it finds a battery, BATTERY_NOT_DISCHARGING=1 on L476 if the battery is not in state "discharging" or "unknown".

Then L487 checks if [ $FOUND_SYS_CLASS_POWER_SUPPLY_AC != 1 ] && [ $BATTERY_NOT_DISCHARGING = 1 ] and sets ON_AC=1.  This if statement is false if a battery is found without an AC even if the battery is charging because L459 sets FOUND_SYS_CLASS_POWER_SUPPLY_AC=1 when it only finds a battery.

Then, L492 checks if [ $FOUND_SYS_CLASS_POWER_SUPPLY_AC = 1 ] and assumes that we already have set ON_AC properly even though ON_AC is never set to 1 when there's a battery charging without an AC power supply being detected.

rickysarraf, 2017-02-21:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

On Mon, 2017-02-20 at 16:40 -0800, Iyad Kandalaft wrote:
> Sorry, my comment was referencing wrong line numbers (i was distracted when I
> put in the issue).

No problem.

> Before I post the VERBOSE_OUTPUT, take a look on the latest commit (ff6362f)
> of lmt-upstream in
> /usr/sbin/laptop_mode and walk through the code.

You still referenced wrong commit id. :-)
But that's okay. I could relate what you're trying to say.

> Variables get instantiated:
> L454: FOUND_SYS_CLASS_POWER_SUPPLY_AC=0
> L456: BATTERY_NOT_DISCHARGING=0
> Then iterate over /sys/class/power_supply and set
> FOUND_SYS_CLASS_POWER_SUPPLY_AC=1 whether it's a battery or an AC power supply
> (L459). 

Yes. For sure, that choice of variable name has become confusing. But it is just
to determine if power_supply subsystem is good enough, or if we should try other
 power subsystems.

> If it finds a battery, BATTERY_NOT_DISCHARGING=1 on L476 if the battery is not
> in state "discharging" or "unknown".

Yes.

> Then L487 checks if [ $FOUND_SYS_CLASS_POWER_SUPPLY_AC != 1 ] && [
> $BATTERY_NOT_DISCHARGING = 1 ] and sets ON_AC=1. This if statement is false if
> a battery is found without an AC even if the battery is charging because L459
> sets FOUND_SYS_CLASS_POWER_SUPPLY_AC=1 when it only finds a battery.

So like I said above. That variable only is to determine if we need to try other
power supply implementations, like ACPI, APM etc.

So the core of the problem is that your power_supply driver is incapable of
detecting the AC power supply, which technically is in charging state, but
because it is not detected, ON_AC is marked as 0.

> Then, L492 checks if [ $FOUND_SYS_CLASS_POWER_SUPPLY_AC = 1 ] and assumes that
> we already have set ON_AC properly even though ON_AC is never set to 1 when
> there's a battery charging without an AC power supply being detected.

Yes. Understood. That is where the problem is with the kernel.

> So, if an AC power supply is not detected but a battery is detected AND the
> battery is giving the correct status, the code above will only work when the
> battery is discharging but not when it's charging.

We cannot rely on batteries, assuming they give correct status because batteries
are known to give broken statuses.

I just noticed that I hadn't pushed my local changes, part of which do similar
changes to what you've proposed. I'll put specifics into the pull request's
comment.

- -- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."
-----BEGIN PGP SIGNATURE-----

iQIzBAEBCgAdFiEEQCVDstmIVAB/Yn02pjpYo/LhdWkFAlir7yAACgkQpjpYo/Lh
dWn76w/6Assjgs6hZyroRABTInUJsyF0+5AmauJzt6lsc2KPcNRAUq4bYAkgOdyU
cNC8VO6rLt6uK2UJboctEvj37FR7R0/eKQmFcU82ahUsj7D5aA8T/0SgZ/gIyDKD
4YLMNhnc/adINhA/k05jjlfpXsrdv4RDgOnIRdXSXJC6npf3C6IndCwsVJRbqac/
DgHzJu4cPeGixsN2Ao6Sy5fcaIADmO7pgNku2wZKyt2OcFo1eU7GXzD4l/nVZFbQ
JvbdOIxabP6ZoqY/f7PMJl0PyjQ0+MFpAhBsENxKqKwh+Fv/mLcFyQv1qaECBDhx
d7Pls4qiQDCkF6ZnabmrTuD7Vw3pKdKcSIaXNSjY9rewZQXEmIDz3LbV2e85yBmY
pKC1qxoWX/g3g8CeLXGhFecKMaF0bIEKDHG9qokJc/cwIyQsXhidtCsDISaupLsP
b4fSOXX8+D5nfq32qS3M7R86mJNjOe1RwbDRqw/CVcSKQP3ExTTN3KZxZDPFIVtq
K1KBh+YLCJ8k3Jnz4vCXJIkuM6pcmYs/osymOMRgxPgRuU1nelVgmTZAWhiwvYre
4gpu8zygg1qKh38esxU/DIE/kvUBdhl9V9gY/+fOSA6vMFetyvX03YAcI1T78VMe
2kWLSiI3ySNOBwlxIX1K9sFgFgOR1Z00FQ/a05PXXhKa/MgmmQQ=
=GzW/
-----END PGP SIGNATURE-----

IyadKandalaft, 2017-02-22:
LOL I was thinking to myself "what are you talking about. I'M ON THE LATEST COMMIT!".  I'll cancel my pull request because I looked over your last 2 commits and they make the same changes I was proposing.  I can refactor the variable names if you want to avoid future confusion.  I haven't tested the code in my specific case to see if it works though.  I'll do that later.

IyadKandalaft, 2017-03-09:
The latest patches work :).  Sorry for taking so long to test.

`
$ ll /sys/class/power_supply/
total 0
drwxr-xr-x  2 root root 0 Mar  9 18:46 ./
drwxr-xr-x 66 root root 0 Mar  9 18:44 ../
lrwxrwxrwx  1 root root 0 Mar  9 18:44 BAT0 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0/
$ sudo service laptop-mode status
Laptop mode status:

Laptop Mode Tools invoked by parent: 11711
Determining power state from status of battery /sys/class/power_supply/BAT0.
Marking it as non-AC because there's battery /sys/class/power_supply/BAT0 in Discharging state
Not trying other options, already found a power supply.
`

closed 2017-03-09 by IyadKandalaft

rickysarraf, 2017-03-10:
On Thu, 2017-03-09 at 15:47 -0800, Iyad Kandalaft wrote:
> The latest patches work :). Sorry for taking so long to test.

Thank you for confirming it back. :-)

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."