rickysarraf/laptop-mode-tools #93: iw txpower fixed needs an argument

Mortal, 2017-03-09:
In my syslog, I get the message `laptop_mode: Failed to re-set power saving mode for wireless card` which comes from the [iw txpower](https://github.com/resfi/resfi/blob/dce45e893692c9145f2cb2e8eeaff983251652a2/iw-4.3/phy.c#L466) command.

In bca28ce0c8e2cce30e13a26046fef59bb60e3686 a call to `$IW dev $IF set txpower fixed` was added (to disable power savings), but this command (`set txpower fixed`) requires an argument.

closed 2017-03-10 by rickysarraf

andpol, 2017-03-12:
Hey @rickysarraf, although 3644486 _does_ fix a bug in the code, I don't believe that change addresses the issue itself. The issue is with `iw` not being able to set txpower in the same way as `iwconfig`.

See the following verbose logs:
```
Mar 11 21:56:39 xps15 laptop_mode[11354]: ++ log VERBOSE 'On battery: setting power saving mode for wlp2s0 to on.'
...
Mar 11 21:56:39 xps15 laptop_mode[11354]: ++ setPowerSave wlp2s0 on
Mar 11 21:56:39 xps15 laptop_mode[11354]: ++ local IF=wlp2s0
Mar 11 21:56:39 xps15 laptop_mode[11354]: ++ local ONOFF=on
Mar 11 21:56:39 xps15 laptop_mode[11354]: ++ '[' -n /sbin/iw ']'
Mar 11 21:56:39 xps15 laptop_mode[11354]: ++ /sbin/iw dev wlp2s0 set power_save on
Mar 11 21:56:39 xps15 laptop_mode[11354]: ++ '[' on = ON ']'
Mar 11 21:56:39 xps15 laptop_mode[11354]: ++ /sbin/iw dev wlp2s0 set txpower fixed
Mar 11 21:56:39 xps15 laptop_mode[11354]: Missing TX power level argument.
Mar 11 21:56:39 xps15 laptop_mode[11354]: ++ log ERR 'Failed to set power saving mode for wireless card'
```

```
Mar 11 22:13:33 xps15 laptop_mode[16129]: ++ log VERBOSE 'On AC power: setting power saving mode for wlp2s0 to off.'
...
Mar 11 22:13:33 xps15 laptop_mode[16129]: ++ setPowerSave wlp2s0 off
Mar 11 22:13:33 xps15 laptop_mode[16129]: ++ local IF=wlp2s0
Mar 11 22:13:33 xps15 laptop_mode[16129]: ++ local ONOFF=off
Mar 11 22:13:33 xps15 laptop_mode[16129]: ++ '[' -n /sbin/iw ']'
Mar 11 22:13:33 xps15 laptop_mode[16129]: ++ /sbin/iw dev wlp2s0 set power_save off
Mar 11 22:13:33 xps15 laptop_mode[16129]: ++ '[' off = ON ']'
Mar 11 22:13:33 xps15 laptop_mode[16129]: ++ /sbin/iw dev wlp2s0 set txpower fixed
Mar 11 22:13:33 xps15 laptop_mode[16129]: Missing TX power level argument.
Mar 11 22:13:33 xps15 laptop_mode[16129]: ++ log ERR 'Failed to re-set power saving mode for wireless card'
```
From reading the `iw` docs (https://wireless.wiki.kernel.org/en/users/Documentation/iw), I believe the correct thing to do is just leave the txpower config alone (or add a user-configurable power value...). I've created pull request #95, which also fixes another bug in the `iwconfig` case.

andpol, 2017-03-12:
For record keeping purposes, this has been fixed, see #95 for details.