rickysarraf/laptop-mode-tools #94: lmt inconsistently reports that battery is broken

Mortal, 2017-03-09:
On my Thinkpad T460s laptop, which has an AC of type Mains and BAT0 and BAT1 of type Battery in `/sys/class/power_supply`, the status reported for the two batteries is always "Charging"/"Discharging" for one of them and "Unknown" for the other (since the batteries are never charging or discharging together -- they take turns).

The [for-loop in lmt that determines the system `ON_AC` state](https://github.com/rickysarraf/laptop-mode-tools/blob/e7d0665b3e8e7dd387e99783c79f467c5a309fb1/usr/sbin/laptop_mode#L453-L485) breaks as soon as it has found a Mains which is online (i.e. power supply plugged in) or a battery which is Discharging, and it logs an error if along the way it finds a battery which is in the Unknown state.

For my system (where the for-loop visits the devices in the lexicographic order AC, BAT0, BAT1) it means that lmt logs an error when my charger is not plugged in AND I'm currently discharging from BAT1, since in that case BAT0 reports the "Unknown" state. In all other cases, the for-loop breaks before seeing that one of the batteries is in the "Unknown" state.

Should I really consider my system broken and report it to the Linux ACPI developers? If so, lmt should probably always report my system as broken (by not breaking out of the for-loop). Otherwise, lmt should probably never report my system as broken (by only reporting the error if *all* batteries are in the Unknown state).

rickysarraf, 2017-03-10:
Broken batteries have been a PITA to deal with.

LMT does support multiple batteries but I've relied on user reports for tests. In your case, since an "Unknown" state is reported, it makes it harder. You can try reporting it to the upstream folks but I have low hopes of that getting fixed, given there are so many manufacturers of such batteries.

Can you run LMT in debug mode and then share the journal logs ?
I am not sure if we can come with a fix/workaround, but it'd be a nice exercise.

Mortal, 2017-03-10:
Do you want the full output from DEBUG=1 in laptop-mode.conf, or do you just want VERBOSE_OUTPUT=1?

I just toyed a bit and it seems the laptop-mode service is reloaded when I unplug the power supply -- and in one case LMT actually got the "Unknown" state from both batteries. Probably this is something that happens just as the power is unplugged, so maybe there's not a satisfying solution for my system.

Given that you have low hopes of getting my battery driver fixed upstream, as a pragmatic user I would be satisfied with a heuristic akin to UPower's, so I might go and implement that on my local system.

I'd still help debugging for the sake of this project, but please let me know if you want DEBUG=1 or just VERBOSE_OUTPUT=1.

rickysarraf, 2017-03-10:
Sorry for not being specific. Please provide logs with VERBOSE_OUTPUT=1

Yes. LMT is invoked whenever there is a power state change. It also is invoked during swsusp resume and on USB device additions.

Don't take my word on upstream. :-)
You may try. Because the sbs_driver's behavior you see, is a timing issue, but still erratic. There was a similar one, but the other way around. f47bee1c47899f02a11ae9f73176eee2161cb18e

Mortal, 2017-03-13:
Here are [some journal logs with VERBOSE_OUTPUT=1](https://gist.github.com/Mortal/e70243948ce27b747fbdbea650b35267). Unfortunately they don't show the behavior I'm reporting, since the only case in which broken battery is reported is when both batteries are in the Unknown state. I'll send you logs when I trigger the behavior I wrote about (when BAT0 is Unknown and BAT1 is Charging/Discharging).

rickysarraf, 2017-03-13:
On Mon, 2017-03-13 at 08:49 -0700, Mathias Rav wrote:
> Here are some journal logs with VERBOSE_OUTPUT=1. Unfortunately they don't
> show the behavior I'm reporting, since the only case in which broken battery
> is reported is when both batteries are in the [...]

To narrow down the Unknown state, that it may be a timing issue, when power
state changes; could you please, after you have unplugged the adapter, restart
LMT and then check the logs ?

With machine on battery, and an LMT restart invoked, it'd iron out the timing
issue that we are suspecting.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

rickysarraf, 2017-10-12:
No activity. Closing. Please re-open when you have the information.

closed 2017-10-12 by rickysarraf