rickysarraf/laptop-mode-tools #96: Graphical GUI dont work

uNbAs, 2017-03-26:
Hi, the graphical GUI dont work.
 `gksudo /usr/sbin/lmt-config-gui 
Using PyQt4
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0xe3
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x2c00010
X Error: BadPixmap (invalid Pixmap parameter) 4
  Major opcode: 54 (X_FreePixmap)
  Resource id:  0x2c00010
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 2 (X_ShmDetach)
  Resource id:  0x2c00010
X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x163
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0xe3
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x2c00013
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x2c00013
X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x163
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0xe3
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x2c00019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x2c00013
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x2c00019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x2c00019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x2c00019
X Error: BadPixmap (invalid Pixmap parameter) 4
  Major opcode: 54 (X_FreePixmap)
  Resource id:  0x2c00019
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 2 (X_ShmDetach)
  Resource id:  0x2c00019`

Please how I can resolve it?

rickysarraf, 2017-03-26:
Works here. What version of Laptop Mode Tools are you using ? It looks like the MIT-SHM bug, which was fixed in the last 2 of the releases.

[image]

rickysarraf, 2017-04-01:
I am closing it assuming you have it fixed with the latest version. Please feel free to re-open if you still encounter the problem.

closed 2017-04-01 by rickysarraf

uNbAs, 2017-04-04:
Sorry Im not fixed, In home laptop but on a laptop of work yes. I remove laptop-mode-tools and reinstall from ppa

rickysarraf, 2017-04-04:
Okay. Then please feel free to post updates on this bug report. I'll leave it closed, but once you have access to your home laptop, you can share the update here.

Regarding PPA, which one ? The project does not provide any PPA. For Fedora and Suse, the RPMs are provided at: https://github.com/rickysarraf/laptop-mode-tools/releases

For Debian/Ubuntu, you could use the packages from:
http://packages.ubuntu.com/search?keywords=laptop-mode-tools
https://packages.debian.org/search?keywords=laptop-mode-tools