rickysarraf/laptop-mode-tools #97: laptop-mode[1492]: Failed to re-set power saving mode for wireless card

sadid, 2017-04-13:
`journalctl -xb` shows this error in my log while the wireless card has this capability since 
`sudo iwconfig wlp2s0 power on` works without any error and `sudo iw wlp2s0 get power_save` gives `on`.

`laptop-mode[1492]: Failed to re-set power saving mode for wireless card`

I'm using Archlinux on kernel 4.10.9-1

rickysarraf, 2017-04-13:
Can you please run LMT in verbose mode, and then share the journal logs ?
What wireless device is this ?
And also the output of following:
`sudo iwconfig wlp2s0 power off`

sadid, 2017-04-14:
`lspci -k`:
```
02:00.0 Network controller: Intel Corporation Wireless 7265 (rev 59)
	Subsystem: Intel Corporation Dual Band Wireless-AC 7265
	Kernel driver in use: iwlwifi
	Kernel modules: iwlwifi
```

the output for `sudo iwconfig wlp2s0 power on/off` is nothing (without error).

[here ](https://gist.github.com/sadid/34b12320bc2430a5428fa29457ecde23) the verbose log for `journalctl -u laptop-mode -b` can be seen. The error is at line: 252.

rickysarraf, 2017-04-14:
Here lies the problem:

```
Apr 14 10:26:08 uPGA271 laptop_mode[968]: Setting power saving for generic wireless interfaces.
Apr 14 10:26:08 uPGA271 laptop_mode[968]: On AC power: setting power level for wlp2s0 to 0.
Apr 14 10:26:08 uPGA271 laptop_mode[968]: Using /sbin/iwconfig for power mgmt
Apr 14 10:26:08 uPGA271 laptop_mode[968]: On AC power: setting power saving mode for wlp2s0 to off.
Apr 14 10:26:08 uPGA271 laptop_mode[968]: Missing TX power level argument.
Apr 14 10:26:08 uPGA271 laptop_mode[968]: Failed to re-set power saving mode for wireless card
```

The fix is in commit: 791191702b91f7e23b280f35e292869631f5ba22
Can you please try it out ?

sadid, 2017-04-14:
@rickysarraf Thanks
switching to latest commit, I can confirm the problem has been solved.

p.s. I just used the git version from AUR (1.71.16 instead of 1.71.1)

rickysarraf, 2017-04-14:
Thanks for confirming. Accordingly, I am closing this bug.

closed 2017-04-14 by rickysarraf