rickysarraf/laptop-mode-tools #100: laptop-mode.service WantedBy=multi-user.target

mercuriete, 2017-04-24:
https://github.com/rickysarraf/laptop-mode-tools/blob/lmt-upstream/etc/systemd/laptop-mode.service#L21

laptop-mode is in the critical chain.
my boot is taking 6 seconds only in start up this service.

can you modify to start after multi-user.target
After=multi-user.target

```
xxxxxxxxx@localhost ~ $ systemd-analyze critical-chain 
The time after the unit is active or started is printed after the "@" character.
The time the unit takes to start is printed after the "+" character.

graphical.target @9.096s
└─multi-user.target @9.096s
  └─laptop-mode.service @3.031s +6.064s
    └─basic.target @2.883s
      └─sockets.target @2.883s
        └─pcscd.socket @2.883s
          └─sysinit.target @2.882s
            └─systemd-update-utmp.service @2.872s +10ms
              └─systemd-tmpfiles-setup.service @2.820s +50ms
                └─local-fs.target @2.819s
                  └─home.mount @2.769s +46ms
                    └─systemd-fsck@dev-disk-by\x2dpartuuid-xxxxxxxx\xxxxxxxx\xxxxxxxx\xxxxxxxx\xxxxxxxxxxxxxxx.service @2.598s +167ms
                      └─dev-disk-by\x2dpartuuid-xxxxxxxx\xxxxxxxx\xxxxxxxx\xxxxxxxx\xxxxxxxxxxxxxxx.device @2.597s
```

rickysarraf, 2017-04-24:
There may be something wrong on your machine/distribution.

Here, locally, I have:
```
rrs@learner:~$ systemd-analyze critical-chain
The time after the unit is active or started is printed after the "@" character.
The time the unit takes to start is printed after the "+" character.

graphical.target @16.724s
└─multi-user.target @16.724s
  └─exim4.service @16.516s +202ms
    └─network-online.target @16.514s
      └─NetworkManager-wait-online.service @4.423s +12.090s
        └─NetworkManager.service @4.290s +129ms
          └─dbus.service @4.240s
            └─basic.target @4.216s
              └─sockets.target @4.216s
                └─uuidd.socket @4.216s
                  └─sysinit.target @4.214s
                    └─systemd-timesyncd.service @4.078s +135ms
                      └─systemd-tmpfiles-setup.service @3.302s +740ms
                        └─local-fs.target @3.301s
                          └─boot-efi.mount @3.201s +99ms
                            └─systemd-fsck@dev-disk-by\x2duuid-2206\x2d0BF2.service @1.270s +1.
                              └─dev-disk-by\x2duuid-2206\x2d0BF2.device @1.270s
```

And in the full list, where it displays LMT too.

```
         12.090s NetworkManager-wait-online.service
          9.067s systemd-tmpfiles-clean.service
          3.015s iio-sensor-proxy.service
          1.874s systemd-fsck@dev-disk-by\x2duuid-2206\x2d0BF2.service
          1.303s mnt-3AEC0E87EC0E3E1F.mount
          1.291s uml-utilities.service
          1.174s systemd-fsck@dev-disk-by\x2duuid-07502716\x2da1f8\x2d4cec\x2da605\x2d447e151b8
          1.098s systemd-rfkill.service
           778ms tor@default.service
           755ms laptop-mode.service
           740ms systemd-tmpfiles-setup.service
           574ms dev-mapper-sdb_crypt.device
           565ms systemd-fsck@dev-mapper-SSHDCrypt\x2dROOT.service
           541ms boot.mount
           520ms media-SSHD.mount
           425ms systemd-journal-flush.service
           347ms lvm2-monitor.service
           307ms systemd-resolved.service
           303ms swap.img.swap
           259ms libvirtd.service
           217ms dnsmasq.service
           212ms loadcpufreq.service
           206ms var-lib-machines.mount
           202ms exim4.service
           188ms networking.service
           167ms accounts-daemon.service
           161ms upower.service
           144ms ModemManager.service
           135ms phpsessionclean.service
           135ms systemd-timesyncd.service
           131ms mountdebugfs.service
           129ms NetworkManager.service
           114ms dev-mapper-SSHDCrypt\x2dSWAP.swap
            99ms boot-efi.mount
            96ms speech-dispatcher.service
            96ms systemd-udevd.service
            89ms systemd-logind.service
            84ms lm-sensors.service
            80ms openbsd-inetd.service
            77ms apache-htcacheclean.service
```

rickysarraf, 2017-04-24:
> Maybe i am wrong but i dont understand why a service that only changes the governors is preventing me to start the display manager and shows the login screen.

It does more than that. Almost all possible power savings for the Linux kernel.

mercuriete, 2017-04-24:
mine:

´´´
          6.064s laptop-mode.service
          1.920s dev-sda9.device
           978ms NetworkManager.service
           924ms libvirtd.service
           700ms mnt-Datos.mount
           647ms systemd-journal-flush.service
           511ms systemd-udevd.service
           411ms upower.service
           306ms systemd-remount-fs.service
           304ms polkit.service
           279ms systemd-logind.service
           270ms wpa_supplicant.service
           246ms udisks2.service
           224ms systemd-tmpfiles-setup-dev.service
           210ms user@114.service
           196ms colord.service
           195ms systemd-rfkill@rfkill1.service
           195ms systemd-rfkill@rfkill0.service
           193ms systemd-user-sessions.service
           188ms systemd-vconsole-setup.service
           182ms systemd-udev-trigger.service
           174ms systemd-fsck-root.service
           167ms systemd-fsck@dev-disk-by\x2dpartuuid-0e671ced\x2d6b76\x2d4aa9\x2d906e\x2d5d4ceb12614c.service
           154ms systemd-journald.service
           149ms systemd-backlight@backlight:intel_backlight.service
           145ms bluetooth.service
           110ms dev-mqueue.mount
           109ms rtkit-daemon.service
           107ms dev-hugepages.mount
           106ms tmp.mount
           105ms sys-kernel-config.mount
           102ms sys-kernel-debug.mount
           101ms sys-fs-fuse-connections.mount
            56ms systemd-sysctl.service
            51ms user@1000.service
            50ms systemd-tmpfiles-setup.service
            46ms home.mount
            44ms systemd-rfkill@rfkill2.service
            44ms systemd-hostnamed.service
            33ms kmod-static-nodes.service
            30ms systemd-random-seed.service
            21ms dev-sda8.swap
            19ms alsa-restore.service
            11ms systemd-tmpfiles-clean.service
            10ms systemd-update-utmp.service
```

I'm using gentoo, so maybe its something wrong in my config file.
I will see. Thank you for your answer :)

mercuriete, 2017-04-24:
my output:

```
mercuriete@localhost /etc/laptop-mode/conf.d $ systemctl status laptop-mode.service 
● laptop-mode.service - Laptop Mode Tools
   Loaded: loaded (/usr/lib64/systemd/system/laptop-mode.service; enabled; vendor preset: disabled)
   Active: active (exited) since lun 2017-04-24 18:20:55 WEST; 46min ago
     Docs: man:laptop_mode(8)
           man:laptop-mode.conf(8)
           http://github.com/rickysarraf/laptop-mode-tools
  Process: 501 ExecStart=/usr/sbin/laptop_mode init auto (code=exited, status=0/SUCCESS)
  Process: 498 ExecStartPre=/bin/rm -f /var/run/laptop-mode-tools/state (code=exited, status=0/SUCCESS)
  Process: 489 ExecStartPre=/bin/rm -f /var/run/laptop-mode-tools/enabled (code=exited, status=0/SUCCESS)
 Main PID: 501 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/laptop-mode.service

abr 24 18:20:49 localhost systemd[1]: Starting Laptop Mode Tools...
abr 24 18:20:51 localhost laptop_mode[501]: enabled, active
abr 24 18:20:52 localhost.localdomain laptop_mode[501]: Cannot find device "eth0"
abr 24 18:20:55 localhost.localdomain systemd[1]: Started Laptop Mode Tools.
```

how can start the daemon more verbose to try to understand why is taking 6 seconds?

rickysarraf, 2017-04-24:
I believe @alonbl is the maintainer for Gentoo. He may have some input.

The initial invocation of LMT, i.e. the initial startup, does take around 5-6 seconds, which usually gets skipped off of systemd's view because that invocation takes place way way early.

So, in your case, if for whatever reason, that initial invocation before systemd gets missed, systemd will do a proper invocation during regular init. And that is when it'll take around 5-6 seconds.

I have tried my best to keep such invocation logics generic, so that it gets applied to (and leveraged by) all Linux distributions. In `/lib/udev/lmt-udev`, we deliberately call `exec systemctl --no-block reload-or-restart laptop-mode` on a machine with systemd active. 

It helps the case I explained above, wherein LMT was activated by another non-systemd entity. Wherein systemd's invocation ends up becoming a mere reload, unless there's a state change during the boot (which is rare).

mercuriete, 2017-04-24:
the only lines that i have modified were:
in cpufreq.conf

```
BATT_CPU_GOVERNOR=powersave
....
LM_AC_CPU_GOVERNOR=schedutil
....
NOLM_AC_CPU_GOVERNOR=schedutil

```

rickysarraf, 2017-04-24:
> how can start the daemon more verbose to try to understand why is taking 6 seconds?

https://github.com/rickysarraf/laptop-mode-tools/wiki

The wiki should help. Should you read, please do provide feedback if you see anything missing.
To your actual question, you should enable `VERBOSE_OUTPUT=1` in `/etc/laptop-mode/laptop-mode.conf`

mercuriete, 2017-04-24:
Thank you very much :)

rickysarraf, 2017-04-24:
Back to your initial report (about 6 seconds execution time), that is expected. What I'm curious to know is why LMT did not get invoked much earlier during the boot (outside the scope of systemd).

mercuriete, 2017-04-24:
ok.
It is probably bacause:
1) i have everything compiled inside the kernel so the kernel modules are already in the kernel.
2) i dont have initramfs so something that is in the disk should be readed after root is mounted.

I will put my systemd-analyze plot:

[image]

So you can close this issue. because 6 seconds is expected.
Thank you very much :)

rickysarraf, 2017-04-25:
Yes. I think there's not much to do in here.

closed 2017-04-25 by rickysarraf