rickysarraf/laptop-mode-tools #98: Typo in laptop-mode-tools/etc/laptop-mode/conf.d/intel-sata-powermgmt.conf

alrekr, 2017-04-24:
Hi

Line 14 in [intel-sata-powermgmt.conf](https://github.com/rickysarraf/laptop-mode-tools/blob/lmt-upstream/etc/laptop-mode/conf.d/intel-sata-powermgmt.conf#L14) says

"#__COMMENT On newner kernels (4.6+), it can also help enabled AHCI Runtime PM savings"

I'm sure "newner" is a typo. I'm not sure about "enabled", as it might just be my understanding that is faulty.

rickysarraf, 2017-04-24:
Thank you for the report. While English isn't my mother tounge, but I still am confident that both the typos are valid.

closed 2017-04-24 by rickysarraf

alrekr, 2017-04-24:
I would strongly object to passing "newner" as intentional. 
In case you mean "in kernels with versions higher than 4.6" you would use the word "newer".
If the second word is supposed to be "it will help do something" you would use the word "enable". "Enabled" is past tense.

rickysarraf, 2017-04-24:
Maybe my words conveyed the opposite of what I meant. What I meant is that both typos have been fixed. :-)