rickysarraf/laptop-mode-tools #101: Linux 4.11 ASPM powersupersave mode

mlen, 2017-05-05:
4.11 kernel introduced new ASPM mode which should be used in pcie-aspm hook if the support for is detected. Probably it'd be nicer to allow to configure the mode to use on ac and on battery, so that users can choose the right mode for them

rickysarraf, 2017-05-06:
Are you referring to [these](https://www.phoronix.com/scan.php?page=news_item&px=ASPM-PCIE-L1-PM-Sub) recent changes ?

mlen, 2017-05-06:
Yes, exactly.

mlen, 2017-05-06:
Having `ASPM_BATTERY_POLICY` and `ASPM_AC_POLICY` configuration variables would be ideal for me

rickysarraf, 2017-05-09:
IIRC, I did explore those power saving attributes. But the improvements weren't substantial. But I'll revisit it again, soon.

Regarding BATT and AC tunables, yes, it'd be nice to have them. If you already have a patch, please do submit it here.

mlen, 2017-05-09:
I don't have a patch yet, for now I just patched the script locally. I'll take a look at this again some time later (probably in June), because -ENOTIME

robbertkrebbers, 2023-10-29:
> Having `ASPM_BATTERY_POLICY` and `ASPM_AC_POLICY` configuration variables would be ideal for me

I second that it would be great to have these options.

With my laptop I ran into the same issue as described here: https://bugzilla.kernel.org/show_bug.cgi?id=215832#c38 That is, `powersave` results in battery drain in s2idle, but using `powersupersave` solves the problem.

Hence it would be great to tell laptop-mode-tools to use powersupersave.

With the lack of an option to set this policy is there a workaround to tell laptop-mode-tools to use `powersupersave`?