rickysarraf/laptop-mode-tools #103: Automaticly start lmt-config-gui with root access

Limator, 2017-06-14:
There only appears a message to start the laptop-mode-tools as root. For many users this is very confusing. Especially most of the users don't know that the GUI is called different to the name of the programm. In the menu it is shown as Laptop Mode Tools, but to start you need to call lmt-config-gui

It would be great when you could make the graphical start directly with asking for password! Normally this is done with policykit.

Thanks!

rickysarraf, 2017-06-15:
I know of the issue and have thought hard on how to tackle this. But I don't have a better answer so far. Perhaps you can help with it.

The GUI is written in python and needs root privileges. The invoking script, `lmt-config-gui`, is a shell script that calls the GUI. Initially, I thought adding the checks to `euid == 0` right into the script. But they were suboptimal.

LMT GUI already ships the policy file. If you exec it with pkexec, you get the proper result.

I just don't know, yet, how to integrate invocation through pkexec into the .desktop. If you have suggestions, please mention.

closed 2017-06-19 by rickysarraf

rickysarraf, 2017-06-19:
Could you please test the referenced deb ?
https://people.debian.org/~rrs/tmp/laptop-mode-tools_1.71.61-1_all.deb

tuxedovinz, 2017-06-20:
The way you integrated pkexec into .desktop is fine. As an example Gparted does it the same way, going via an additional bash script.

I installed the deb and polkit works fine but the package is missing python3-pyqt5 as an dependency on Ubuntu/Debian. I'd fixed it but couldn't find the packaging sources. ;)

rickysarraf, 2017-06-21:
Thanks for the feedback.

About the dependency, yes, that happened because I recently ported it to Py3PyQT5. So, when the next release happens, that fix will be included.