rickysarraf/laptop-mode-tools #106: battery-level polling solution for systemd results in useless journal/logs spam

smartass101, 2017-08-05:
I I understand correctly, the commit 5fe7a63738fb758f1d5f8809d8fc0bcdce2f1e4a and the related [Debian bug #791412](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=791412) is responsible for log(jounral, syslog) messages like this:

```
Aug 05 16:19:07 argentum systemd[1]: Starting Laptop Mode Tools - Battery Polling Service...
Aug 05 16:19:07 argentum systemd[1]: Reloading Laptop Mode Tools.
Aug 05 16:19:07 argentum systemd[1]: Started Laptop Mode Tools - Battery Polling Service.
Aug 05 16:19:07 argentum laptop_mode[27209]: Laptop mode
Aug 05 16:19:07 argentum laptop_mode[27209]: enabled, not active [unchanged]
Aug 05 16:19:07 argentum systemd[1]: Reloaded Laptop Mode Tools.
Aug 05 16:22:07 argentum systemd[1]: Starting Laptop Mode Tools - Battery Polling Service...
Aug 05 16:22:07 argentum systemd[1]: Reloading Laptop Mode Tools.
Aug 05 16:22:07 argentum systemd[1]: Started Laptop Mode Tools - Battery Polling Service.
Aug 05 16:22:07 argentum laptop_mode[27468]: Laptop mode
Aug 05 16:22:07 argentum laptop_mode[27468]: enabled, not active [unchanged]
Aug 05 16:22:07 argentum systemd[1]: Reloaded Laptop Mode Tools
```

While I do understand that the current solution is technically sound, it produces many somewhat useless messages in the journal. At first sight such behavior looks like a problem. It also spams the log and makes it harder to read.
It could even be argued that it defeats the package's purpose, because it increases the number of writes to logs on disks.
Could something be done about this, please?
At least this "feature" should respect the `CONTROL_BATTERY_LEVEL_POLLING=0` setting. It seems wasteful to run `lmt-udev` and `laptop_mode` through it to do essentially nothing in this case.

smartass101, 2017-08-05:
For now I have disabled the timer responsible for these events by `systemctl disable --now laptop-mode.timer`. BTW, the timer name does not reflect its description and use very well.

rickysarraf, 2017-08-05:
@smartass101 That is exactly what I was going to recommend. To disable the timer. I don't see any issues with it otherwise. Half the logs you mentioned are logged by systemd.

The idea was simple. Given the way systemd performs, the battery polling module could not be run as such under it. Hence, the lmt-poll service was created, which is invoked through laptop-mode.timer. I think this was documented in the release post for that particular release and left for individual packagers to decide how they wanted to integrate it further.

The integration with ```CONTROL_BATTERY_LEVEL_POLLING``` was left out intentionally. Because that works in sysvinit. For systemd users, they need to explicitly enable the timer instead.

Regarding the name and description, let  me know what you think would be a better wording.

```
rrs@priyasi:/usr/share/doc/laptop-mode-tools$ systemctl status laptop-mode.timer 
● laptop-mode.timer - Runs Laptop Mode Tools - Polling Service, every 150 seconds
   Loaded: loaded (/lib/systemd/system/laptop-mode.timer; enabled; vendor preset: enab
   Active: active (waiting) since Fri 2017-08-04 14:07:22 IST; 1 day 7h ago
  Trigger: Sat 2017-08-05 21:59:42 IST; 37s left

Aug 04 14:07:22 priyasi systemd[1]: Started Runs Laptop Mode Tools - Polling Service, 
21:59 ♒♒♒   ☺    
rrs@priyasi:/usr/share/doc/laptop-mode-tools$ systemctl status lmt-poll.service 
● lmt-poll.service - Laptop Mode Tools - Battery Polling Service
   Loaded: loaded (/lib/systemd/system/lmt-poll.service; enabled; vendor preset: enabl
   Active: inactive (dead) since Sat 2017-08-05 22:02:14 IST; 54s ago
     Docs: man:laptop_mode(8)
           man:laptop-mode.conf(8)
           http://github.com/rickysarraf/laptop-mode-tools
  Process: 6420 ExecStart=/lib/udev/lmt-udev auto (code=exited, status=0/SUCCESS)
 Main PID: 6420 (code=exited, status=0/SUCCESS)

Aug 05 22:02:14 priyasi systemd[1]: Starting Laptop Mode Tools - Battery Polling Servi
Aug 05 22:02:14 priyasi systemd[1]: Started Laptop Mode Tools - Battery Polling Servic
22:03 ♒♒♒    ☹  => 3  
```

smartass101, 2017-08-05:
>  I think this was documented in the release post for that particular release and left for individual packagers to decide how they wanted to integrate it further.

I only found some brief mention in changelog and an even briefer mention in changelog.Debian on my system. I definitely wasn't informed about this during the jessie->stretch upgrade as it is not mentionedi n NEWS.Debian. That's why I was so surprised by this behavior.

> For systemd users, they need to explicitly enable the timer instead.

In my case it was enabled by default during my jessie->stretch upgrade, even though I had `CONTROL_BATTERY_LEVEL_POLLING=0`. 

I think the best course of action would be to

1. document this in NEWS.Debian
2. prevent the timer from being enabled by default, unless they have the setting `CONTROL...=1`

>Regarding the name and description, let me know what you think would be a better wording.

I suggest simply `lmt-poll.timer` which better reflects that it starts `lmt-poll.service`.

rickysarraf, 2017-08-06:
yes. I noticed the missing entry for NEWS.Debian. This should have been documented better.

Regarding point 2, I'll see what options systemd provides. I think it does have a way to source such settings from a file.

Regarding the name, is it really that big a problem now, that a name change is important ?

rickysarraf, 2017-08-06:
Does the following suffice for NEWS.Debian, and makes thing much clearer ?
Taken from: https://www.researchut.com/blog/laptop-mode-tools-1_68

```
Starting with the 1.68 release, there are some new files: laptop-mode.service, laptop-mode.timer and lmt-poll.service. lmt-poll.service is the equivalent of the module battery-level-polling. 
To enable battery level polling on a systemd init based machine, please enable **lmt-poll.service and laptop-mode.timer service files**
```

I think this should be enough and can be used to avoid any further name changes. I'm not too excited to change service file names now, after so many releases.

rickysarraf, 2017-08-06:
So systemd has the ```EnvironmentFile``` directive to source external files and their config variables. With this, we could source battery-polling file. But the mentioned directive is part of the service unit, which means the timer will still be triggered, and will invoke the service file, come to know the environment variable and then just exit.

closed 2019-01-18 by rickysarraf

rickysarraf, 2019-01-18:
On Thu, 2019-01-17 at 06:28 -0800, Erich wrote:
> I had to stop laptop with power switch, the consoles and xterminal
> were not accessible.
> ping and Remote login with ssh was not possible , network interface
> was down.
> seems to be serious

I have fixed the issue. But I do not think whatever freeze issue you
would have seen would be caused by this minor extra logging.

The fix for this issue, that to honor the batter polling configuration
setting, is something I wanted to have for quite some time.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

BillyCroan, 2024-01-29:
This is still happening, as of 2024-01-29, seen in Debian Bookworm.

every 150 seconds in the logs, forever repeating these six lines.

I would suggest that at least while the AC power is connected (or once batt is full and AC is connected), this timer should be disabled.

I'm not entirely sure why it needs to be enabled, ever, if CONTROL_BATTERY_LEVEL_POLLING=0 in /etc/laptop-mode/conf.d/battery-level-polling.conf, which it is on my system.  Perhaps the .service could include:
ExecStart=-/bin/sh -c '[ x"$CONTROL_BATTERY_LEVEL_POLLING" = "x1" ] && /lib/udev/lmt-udev auto || systemctl disable lmt-poll.timer'
ExecStart=-/bin/sh -c '[ x"$CONTROL_BATTERY_LEVEL_POLLING" = "x1" ] && /lib/udev/lmt-udev auto || systemctl disable laptop-mode.timer'

Then it would auto-disable itself if the timer fires while CONTROL_LATTERY_LEVEL_POLING was set to 0.  right?

The timer for lmt-poll.service, by convention, should be called lmt-poll.timer, and that its not, adds to the confusion so I also suggest renaming the .timer to match.

BillyCroan, 2024-01-29:
This is still happening, as of 2024-01-29.  seen in Debian 12.

the .timer is set to 150 seconds.  and the .timer name does not match the .service name it calls, which is unnecessarily confusing.

rickysarraf, 2024-02-20:
>The timer for lmt-poll.service, by convention, should be called lmt-poll.timer, and that its not, adds to the confusion so I also suggest renaming the .timer to match.

Well. When I first was integrating LMT with systemd, it started with `laptop-mode.service` itself. Thus the `laptop-mode.timer`. Then, later, it was evident that having the polling service separate would be better; so `lmt-poll.service`. So I don't want to change it no more.

Maybe if `systemd` provides some sort of aliasing, that something that can be added.