rickysarraf/laptop-mode-tools #108: Runtime power management AUTOSUSPEND_TIMEOUT=0 is confusing

IyadKandalaft, 2017-08-10:
In the runtime_pm module, a device's power/autosuspend is set to 0 when disabling autosuspend.  Subsequently, power/control is set to on, which also disables autosuspend. See
https://github.com/rickysarraf/laptop-mode-tools/blob/5f43794fb0d7b238387089939326c25581121a17/usr/share/laptop-mode-tools/modules/runtime-pm#L146

This behavior used to be correct but is no longer the case. Setting autosuspend to 0 temporarily enables autosuspend with a timeout of 0 seconds.  thereafter, setting the power/control to on disables the autosuspend.

The kernel docs indicate:
"In 2.6.21 writing "0" to power/autosuspend would prevent the device from being autosuspended; the behavior was changed in 2.6.22.  The power/autosuspend attribute did not exist prior to 2.6.21, and the power/level attribute did not exist prior to 2.6.22.  power/control was added in 2.6.34, and power/autosuspend_delay_ms was added in 2.6.37 but did not become functional until 2.6.38"

IyadKandalaft, 2017-08-13:
See pull request #109 for a proposed fix.

rickysarraf, 2017-10-10:
Can you please point me to the docs mentioning so ?
When the last time I investigated Runtime PM, it was this combination that allowed the setting to kick in.
The kernels you're mentioning are very old.

rickysarraf, 2017-10-10:
Okay. I think you aren't concerned with the combination being written together. That is a different behavior, for which I can't recollect the reference. But the values needed to be written to both.

Anyways, regarding your PR #109 , I think it is good. What do you think about not depending on string comparison ? Instead lets just rely on the sysfs interface, from newer to older.

rickysarraf, 2017-10-12:
Closing, as discussed in #109

closed 2017-10-12 by rickysarraf