rickysarraf/laptop-mode-tools #111: GUI not found

Maxiride, 2017-09-18:
Premise: I'm on Ubuntu 16.04 LTS, I've downloaded the latest 1.71 release tarball and installed it with `sudo checkinstall ./install.sh`. Checkinstall is like make file but with a builtin monitoring of what goes where in the system for easier removal.

The installation went successfully, in the shell `laptop_mode` is an available command however when heading to `/usr/sbin/`, the end path `lmt-config-gui` isn't available as stated in the [wiki](https://github.com/rickysarraf/laptop-mode-tools/wiki/Laptop-Mode-Tools-Graphical-Configuration-Tool)

rickysarraf, 2017-10-10:
If `checkinstall ./install.sh` invokes `install.sh` correctly, then it should be there.

Looking at the line https://github.com/rickysarraf/laptop-mode-tools/blob/6eede656bf7dfd2cb6708fb3c51de21b83ab6097/install.sh#L202, I fail to see why it shouldn't be there.

Could you please provide the logs for checkinstall ?

Maxiride, 2017-10-16:
Well I eventually installed directly using the install.sh and the GUI now it is there. I suppose checkinstall didn't worked as expected but I'm afraid I don't have the logs anymore.

closed 2017-10-16 by Maxiride