rickysarraf/laptop-mode-tools #112: Syntax error in /usr/sbin/laptop_mode

cduston, 2017-10-10:
When starting the service for the first time, I get the following:
```
# /etc/rc.d/rc.laptop-mode start 
/usr/sbin/laptop_mode: line 205: WIRELESS_POWER~_DEBUG: syntax error in expression (error token is "~_DEBUG")
Laptop mode
enabled, active
```
I installed with the latest package here, running Slackware 14.2 on a MacBook Pro 9,2. I've also tried this using a package from SlackBuilds, and both have the same issue. Should I worry about this message?

rickysarraf, 2017-10-10:
Yes. This looks like a bug to me. But I can only be certain with the full debug logs.

From the looks of it, it seems you have a backup configuration file in `/etc/laptop-mode/conf.d/` directory. Something left over by an editor.

Can you please show me the listing of `/etc/laptop-mode/conf.d/` ?

rickysarraf, 2018-01-19:
No response from OP. Closing bug report. Please re-open if you have more information.

closed 2018-01-19 by rickysarraf