rickysarraf/laptop-mode-tools #114: laptop_mode service stops.

NewbProgrammer101, 2017-10-22:
Prerequisites: Have a laptop with OpenRC as init system.
Step 1: Turn on the laptop either on A/C or Battery.
Step 2: Log in either as user or root.
Step 3: top | grep laptop_mode # do this IMMEDIATELY after logging in.
Step 4: Watch the service disappear.

A few seconds after logging in, the laptop_mode service will quit without warning. 
When you type rc-service laptop_mode restart, it will restart, but then quit again without warning.

rickysarraf, 2017-10-23:
I am no expert with OpenRC, but you can have a look at the SysV init and the systemd service/timer files.

Regarding the way it performs, LMT is not a daemon. It is a bunch of scripts that act on event, perform their task and exit. There is no process to spawn in the background.

closed 2017-10-23 by rickysarraf

NewbProgrammer101, 2017-10-23:
I see. The difference between a daemon and things that act as an event still baffles me.

But nevertheless, I understand.