rickysarraf/laptop-mode-tools #116: Power supply via USB not detected

luke-jr, 2017-11-10:
`$POWER_SUPPLY/type` is only checked for Mains or Battery, but not USB.

IMO the current behaviour is reasonable, however, since USB might not provide sufficient power to guarantee the battery isn't still discharging.

luke-jr, 2017-11-10:
Correction, current behaviour fails to determine when on battery at all, so not reasonable.

luke-jr, 2017-11-10:
(For reference, the hardware is a GPD Win.)

rickysarraf, 2017-11-24:
Do you mean the current behaviour, while on USB charging? Otherwise, LMT has been working well for many users

I have no insight into that hardware so I'm willing to review a patch.

luke-jr, 2017-11-24:
This is the patch I am running with that works:

```diff
diff -ur laptop-mode-tools_1.71.orig/etc/rules/99-laptop-mode.rules laptop-mode-tools_1.71/etc/rules/99-laptop-mode.rules
--- laptop-mode-tools_1.71.orig/etc/rules/99-laptop-mode.rules  2017-01-12 03:06:23.000000000 -0500
+++ laptop-mode-tools_1.71/etc/rules/99-laptop-mode.rules   2017-11-10 06:28:58.424673661 -0500
@@ -1,3 +1,3 @@
-ACTION=="change", SUBSYSTEM=="power_supply", ENV{POWER_SUPPLY_NAME}=="|AC|ACAD", RUN+="lmt-udev auto"
+ACTION=="change", SUBSYSTEM=="power_supply", RUN+="lmt-udev auto"
 ACTION=="add|remove", SUBSYSTEM=="machinecheck", RUN+="lmt-udev auto"
 ACTION=="add", SUBSYSTEM=="usb", RUN+="lmt-udev force"
diff -ur laptop-mode-tools_1.71.orig/usr/sbin/laptop_mode laptop-mode-tools_1.71/usr/sbin/laptop_mode
--- laptop-mode-tools_1.71.orig/usr/sbin/laptop_mode    2017-01-12 03:06:23.000000000 -0500
+++ laptop-mode-tools_1.71/usr/sbin/laptop_mode 2017-11-10 06:28:34.623491400 -0500
@@ -456,8 +456,8 @@
    BATTERY_NOT_DISCHARGING=0
    for POWER_SUPPLY in /sys/class/power_supply/* ; do
        if [ -f $POWER_SUPPLY/type ] ; then
-           FOUND_SYS_CLASS_POWER_SUPPLY_AC=1
            if [ "$(cat $POWER_SUPPLY/type)" = "Mains" ]; then
+               FOUND_SYS_CLASS_POWER_SUPPLY_AC=1
                log "VERBOSE" "Determining power state from $POWER_SUPPLY/online."
                if [ "$(cat $POWER_SUPPLY/online)" = 1 ]; then
                    ON_AC=1
@@ -529,7 +529,7 @@
        fi
    else
        log "VERBOSE" "No /sys/class/power_supply, ACPI, APM or PMU power management information found -- assuming AC power is present."
-       ON_AC=1
+       #ON_AC=1
    fi
 
 
@@ -759,6 +759,7 @@
                        log "ERR" "Cannot determine design_capacity_warning"
                        log "ERR" "Disabling hibernation"
                        ENOUGH_CHARGE=1
+                       ALARM_LEVEL=0
                    fi
 
                    if [ "$ALARM_LEVEL" -ne 0 ] && [ "$DISABLE_BATTERY_ALARM_CHECK" -ne 1 ]; then
```

It is probably not suitable for general usage, however.

Some sysfs contents:

* http://luke.dashjr.org/tmp/code/power_supply_full.tgz
* http://luke.dashjr.org/tmp/code/power_supply_discharging.tgz
* http://luke.dashjr.org/tmp/code/power_supply_charging.tgz

rickysarraf, 2017-12-26:
This needs more information on the hardware side. From the power supply, we need status like 'AC|ACAD'. We can't just generalize and listen for all events.

Also, we try to remain on the safer side when there's no power supply subsystem available. So we mark ON_AC=1 and assume that there's AC power. Changing this would break a lot of use cases.

rickysarraf, 2018-05-01:
So Linux 4.17 has some new features in regard to USB power supply.
https://lkml.org/lkml/2018/4/24/424