rickysarraf/laptop-mode-tools #119: default power on DIS with AC online,  howto disable it?

andysendo, 2017-11-24:
my system is debian9, kernel 4.13(backports), installed LMT v.172. 

After LMT started with default configure, in AC mode the second video card (Radeon) is enabled. It is no need for me, kernel DRI_PRIME control is good enough, usually DIS stauts is DynOff (ac on/off), if i do want to use DIS for some app i just start the app with ENV DRI_PRIME=1.

I tried modity /etc/laptop-mode/conf.d/vgaswitcheroo.conf:
LM_AC_ENABLE_VGASWITCHEROO=1
NOLM_AC_ENABLE_VGASWITCHEROO=1
restart LMT service the DIS still powerON.

So howto disable LMT auto control vgaswitcheroo?
Reguards.

rickysarraf, 2017-11-24:
This modules isn't enabled by default. Have you ensured that you've enabled the same ?

'''

# Control video output settings?
CONTROL_VGASWITCHEROO=0
'''

andysendo, 2017-11-24:
i installed LMT by apt, NO change any default conf, then the second video card enabled(ac on). 

CONTROL_VGASWITCHEROO=0 is defualt.

[laptop-mode-tools.deb.tar.gz](https://github.com/rickysarraf/laptop-mode-tools/files/1500680/laptop-mode-tools.deb.tar.gz)
That is the deb by apt install from debian repo.

rickysarraf, 2017-11-24:
Yes. You need to enable that module by hand, or using the GUI tool. It has
been disabled by default because AMD Radeon is not a very common VGA

andysendo, 2017-11-24:
I don't want LMT auto control vgaswitch so i did not change CONTROL_VGASWITCHEROO=0 to 1, but LMT auto enable DIS after install?

rickysarraf, 2017-11-25:
In that case, this must be a bug in LMT. Can you run the vgaswitcheroo
module in debug mode?

To do so, you need to set DEBUG=1 for this module and then capture the
system logs.

andysendo, 2017-11-25:
[debug.txt](https://github.com/rickysarraf/laptop-mode-tools/files/1502952/debug.txt)
 i set vgaswitcheroo.conf & laptop-mode.conf DEBUG=1 then run  command 'sudo laptop_mode' got this debug. Please check it.

rickysarraf, 2017-11-25:
The changes are good. But please plug and then unplug the power for LMT to
trigger.

s3nt fr0m a $martph0ne, excuse typ0s

andysendo, 2017-11-26:
--- plug and then unplug the power for LMT to trigger.
Don't know what you mean. 

Atfer intall with default settings, If AC Off LMT will auto disable DIS, AC On LMT will auto enable DIS, even if i stop LMT service by systemctl LMT vgaswitch still working that way.

How do i log it?

rickysarraf, 2017-11-26:
@andysendo After you have made your settings, just do an LMT restart. This will log all debug messages to syslog/journal. Then you can filter and share the relevant log messages.

andysendo, 2017-11-26:
newly apt intall LMT, AC On, DIS enabled (disabled before install) 
$ sudo journalctl | grep laptop
 debian laptop_mode[970]: enabled, not active
 debian laptop_mode[1573]: Laptop mode
 debian laptop_mode[1573]: enabled, not active [unchanged]

change setting /etc/laptop-mode/conf.d/vgaswitcheroo.conf DEBUG=1, restart LMT, AC On, DIS enabled, log as 

[debug1.txt](https://github.com/rickysarraf/laptop-mode-tools/files/1503605/debug1.txt)

AC Off, DIS auto disabled, log as 
[debug2.txt](https://github.com/rickysarraf/laptop-mode-tools/files/1503607/debug2.txt)

then AC back On, DIS auto enabled again, log as 
[debug3.txt](https://github.com/rickysarraf/laptop-mode-tools/files/1503608/debug3.txt)

Please check it.

rickysarraf, 2017-11-26:
Thanks for the logs. They are clear enough to show that LMT isn't invoking
the module.

There may be other tools fiddling with the power savings for the device.

s3nt fr0m a $martph0ne, excuse typ0s

andysendo, 2017-11-26:
I just install cpufreqd for powersaving, then LMT. I am sure LMT cause this problem.
Maybe LMT/systemd service cause it?

rickysarraf, 2017-11-26:
I don't see that happening. That's why I asked you to run it into debug mode.

Looking back at the module documentation:

```
# Modes of operation for Hybrid Graphics
# The default settings will switch off discrete graphics, when on battery
# and not in use
BATT_ENABLE_VGASWITCHEROO=1
LM_AC_ENABLE_VGASWITCHEROO=0
NOLM_AC_ENABLE_VGASWITCHEROO=0
VGASWITCHEROO_FILE="/sys/kernel/debug/vgaswitcheroo/switch"
~                                                                                      
```

We only fiddle with the ON and OFF values. Nothing else.

Now back to the vgaswitcheroo implementation:

```
Driver power control
In this mode of use, the discrete GPU automatically powers up and down at the discretion of the driver’s runtime pm. On muxed machines, the user may still influence the muxer state by way of the debugfs interface, however the ON and OFF commands become a no-op for the discrete GPU.

This mode is the default on Nvidia HybridPower/Optimus and ATI PowerXpress. Specifying nouveau.runpm=0, radeon.runpm=0 or amdgpu.runpm=0 on the kernel command line disables it.

When the driver decides to power up or down, it notifies vga_switcheroo thereof so that it can (a) power the audio device on the GPU up or down, and (b) update its internal power state representation for the device. This is achieved by vga_switcheroo_set_dynamic_switch().

```

From the symptoms you've mentioned, it seems the default mode of of power savings operation is left out to the device driver itself. You may want to check your kernel logs to confirm.

There's nothing much LMT does here. Hence, I'm closing this bug. If you have additional information showing LMT to be the cause of the problem, please feel free to re-open this bug.

closed 2017-11-26 by rickysarraf