rickysarraf/laptop-mode-tools #120: Partitions are remounted when USB is connected

dschrempf, 2017-11-24:
I have a short question. Is it normal that all partitions are remounted when I connect a USB device (a DAC)? 

I get the following output in `dmesg`:

    [402000.297303] usb 1-1.3: new full-speed USB device number 4 using ehci-pci
    [402001.405941] EXT4-fs (sda2): re-mounted. Opts: data=ordered,commit=360
    [402001.498949] amdgpu: [powerplay] min_core_set_clock not set
    [402001.570299] EXT4-fs (sdb1): re-mounted. Opts: data=ordered,commit=360
    [402001.640137] EXT4-fs (sdb1): re-mounted. Opts: data=ordered,commit=360
    [402001.709000] EXT4-fs (sdc1): re-mounted. Opts: data=ordered,commit=360
    [402005.928252] EXT4-fs (sda2): re-mounted. Opts: data=ordered,commit=360
    [402006.015125] amdgpu: [powerplay] min_core_set_clock not set
    [402006.130828] EXT4-fs (sdb1): re-mounted. Opts: data=ordered,commit=360
    [402006.188702] EXT4-fs (sdb1): re-mounted. Opts: data=ordered,commit=360
    [402006.232294] EXT4-fs (sdc1): re-mounted. Opts: data=ordered,commit=360
    [402010.539761] EXT4-fs (sda2): re-mounted. Opts: data=ordered,commit=360
    [402010.604265] amdgpu: [powerplay] min_core_set_clock not set
    [402010.750664] EXT4-fs (sdb1): re-mounted. Opts: data=ordered,commit=360
    [402010.790189] EXT4-fs (sdb1): re-mounted. Opts: data=ordered,commit=360
    [402010.851073] EXT4-fs (sdc1): re-mounted. Opts: data=ordered,commit=360
    [402014.971215] EXT4-fs (sda2): re-mounted. Opts: data=ordered,commit=360
    [402015.063730] amdgpu: [powerplay] min_core_set_clock not set
    [402015.155186] EXT4-fs (sdb1): re-mounted. Opts: data=ordered,commit=360
    [402015.235480] EXT4-fs (sdb1): re-mounted. Opts: data=ordered,commit=360
    [402015.282952] EXT4-fs (sdc1): re-mounted. Opts: data=ordered,commit=360

It might be that this remount is triggered by `mpd` because a sound card is
attached and part of the database is on sda2 and on sdb1. sdc1 is a backup drive
that should not be remounted in any case. Do you have any ideas?

rickysarraf, 2017-11-25:
To apply power savings for the newly attached USB device, we invoke LMT. That in turn runs through all the modules, resulting in the remounted that you are seeing.

This is normal behaviour. Is it causing you any trouble?

PS: There is an option in LMT to invoke on a module basis. This behaviour is still used in chromium os. But it had particular corner case bugs, hence I disabled it for upstream

rickysarraf, 2017-11-25:
Some context is available in https://github.com/rickysarraf/laptop-mode-tools/issues/80

rickysarraf, 2017-11-25:
Unless you have an erratic behaviour, I'd leave it as is.  I'm closing this bug report but please feel free to reopen it if it is causing any trouble

closed 2017-11-25 by rickysarraf

dschrempf, 2017-11-25:
Thank you for your fast reply. No, it doesn't lead to errativ behavior. The only problem is that the drives spin up because they are remounted and I'd rather not trigger a spin-up every time I attach the DAC. Is there a way to not invoke LMT for a specific USB device because I manually manage the power supply of the DAC anyways.

Thanks a lot!

rickysarraf, 2017-11-25:
I wasn't aware that a remount, that just fiddles with the commit internal,
triggers an fsync.

You could change the udev invocation for subsystem type usb, to limit
itself just for the runtime-pm module

s3nt fr0m a $martph0ne, excuse typ0s

rickysarraf, 2017-11-26:
@dschrempf  You could change the USB type invocation to below.

```
ACTION=="add|remove", SUBSYSTEM=="usb", RUN+="lmt-udev force modules=runtime-pm
devices=%k"
```

This would force execution of the **runtime-pm** module only.

dschrempf, 2017-11-29:
Hi, thank you very much for you help. I changed the line to

    ACTION=="add", SUBSYSTEM=="usb", RUN+="lmt-udev force modules=runtime-pm devices=%k"

Like this it only triggers the runtime-pm module when USB devices are attached, not removed. Works like a charm, thank you!