rickysarraf/laptop-mode-tools #122: wireless power error L79

danarchy85, 2017-12-20:
I'm running laptop-mode-tools 1.71 on Gentoo and have run into an error with wireless-power upon daemon startup.

```
 :~ $ sudo rc-service laptop_mode restart
 * Stopping laptop_mode ...
/usr/share/laptop-mode-tools/modules/wireless-power: line 79: dev: command not found
Failed to set power saving mode for wireless card          [ ok ]
 * Starting laptop_mode ...
/usr/share/laptop-mode-tools/modules/wireless-power: line 79: dev: command not found
Failed to set power saving mode for wireless card`
```

 I have found a discrepancy between the two branches here 'lmt-upstream' and 'master': both appear to have the same 1.71 version in laptop-mode-tools.spec, but this bug seems to be fixed in 'lmt-upstream', but not in 'master'.

I'm assuming that Gentoo is pulling from the 'master' branch, and so has pulled in a faulty 1.71 version: 

Line 79 is definitely making the wrong call to `iw` rather than `iwconfig` and the following change fixes the error:

```
/usr/share/laptop-mode-tools/modules/wireless-power
@@ -76,7 +76,7 @@
                 if [ $ONOFF = "ON" ]; then
                     $IWCONFIG $IF txpower auto
                 else
-                    $IW dev $IF txpower fixed
+                    $IWCONFIG $IF txpower fixed
                 fi
        fi
 }
```

I see that was fixed within 'lmt-upstream', though. Is there a reason for why 'master' appears to be behind 'lmt-upstream', or should Gentoo be pulling from 'lmt-upstream'? 

Full bug details on Gentoo's end can be found here: https://bugs.gentoo.org/641834

rickysarraf, 2017-12-22:
The master branch is for Debian GBP packaging.

I had documented this in the README file, that all development is done in the lmt-upstream branch.
https://github.com/rickysarraf/laptop-mode-tools/blob/lmt-upstream/README.md#developers

@alonbl I believe you are packaging for Gentoo ?

closed 2017-12-22 by rickysarraf

alonbl, 2017-12-22:
yes, I will cherry-pick 3644486