rickysarraf/laptop-mode-tools #123: runtime_pm issues with ata disks

rickysarraf, 2018-01-16:
These are serious error occurring because our devices aren't really capable of LPM. Device blacklisting is an issue here because of the same device is referenced through multiple persistent names.

We need a better blacklist mechanism. I've had the following in my runtime-pm blacklist, but it still picks the devices.
```
# The list of Device IDs that should not use autosuspend. Use system commands or
# look into sysfs to find out the IDs of your devices.
# Example: AUTOSUSPEND_RUNTIME_DEVID_BLACKLIST="046d:c025 0123:abcd"
AUTOSUSPEND_RUNTIME_DEVID_BLACKLIST="2717:ff40 0bda:0129 0000:00:1f.2 1:0:0:0 target0:0:0 target1:0:0"
```

because

```
root@priyasi:/sys/bus/scsi# ls -l devices/
total 0
lrwxrwxrwx 1 root root 0 Jan 16 10:59 0:0:0:0 -> ../../../devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:0/0:0:0:0
lrwxrwxrwx 1 root root 0 Jan 16 10:59 1:0:0:0 -> ../../../devices/pci0000:00/0000:00:1f.2/ata2/host1/target1:0:0/1:0:0:0
lrwxrwxrwx 1 root root 0 Jan 16 11:27 host0 -> ../../../devices/pci0000:00/0000:00:1f.2/ata1/host0
lrwxrwxrwx 1 root root 0 Jan 16 11:27 host1 -> ../../../devices/pci0000:00/0000:00:1f.2/ata2/host1
lrwxrwxrwx 1 root root 0 Jan 16 11:27 host2 -> ../../../devices/pci0000:00/0000:00:1f.2/ata3/host2
lrwxrwxrwx 1 root root 0 Jan 16 11:27 host3 -> ../../../devices/pci0000:00/0000:00:1f.2/ata4/host3
lrwxrwxrwx 1 root root 0 Jan 16 11:27 target0:0:0 -> ../../../devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:0
lrwxrwxrwx 1 root root 0 Jan 16 11:27 target1:0:0 -> ../../../devices/pci0000:00/0000:00:1f.2/ata2/host1/target1:0:0
```

I only uncovered this issue after applying #118 Thanks @yardenac 

```
Jan 16 12:32:28 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Jan 16 12:32:28 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Jan 16 12:32:28 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Jan 16 12:32:33 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Jan 16 12:32:33 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Jan 16 12:32:33 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Jan 16 12:32:38 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Jan 16 12:32:38 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Jan 16 12:32:38 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Jan 16 12:32:40 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Jan 16 12:32:40 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Jan 16 12:32:49 priyasi kernel: ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
Jan 16 12:32:49 priyasi kernel: ata2.00: configured for UDMA/133
Jan 16 12:32:49 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Jan 16 12:32:49 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Jan 16 12:32:51 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Jan 16 12:32:51 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Jan 16 12:32:52 priyasi kernel: ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
Jan 16 12:32:52 priyasi kernel: ata2.00: configured for UDMA/133
Jan 16 12:32:52 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Jan 16 12:32:52 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Jan 16 12:32:54 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Jan 16 12:32:54 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Jan 16 12:33:04 priyasi kernel: ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
Jan 16 12:33:04 priyasi kernel: ata2.00: configured for UDMA/133
Jan 16 12:33:04 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Jan 16 12:33:04 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Jan 16 12:33:06 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Jan 16 12:33:06 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Jan 16 12:33:09 priyasi kernel: ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
Jan 16 12:33:09 priyasi kernel: ata2.00: configured for UDMA/133
Jan 16 12:33:09 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Jan 16 12:33:09 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Jan 16 12:33:12 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Jan 16 12:33:12 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Jan 16 12:33:12 priyasi kernel: ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
Jan 16 12:33:12 priyasi kernel: ata2.00: configured for UDMA/133
Jan 16 12:33:12 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Jan 16 12:33:12 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Jan 16 12:33:14 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Jan 16 12:33:14 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Jan 16 12:33:15 priyasi kernel: ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[... 113 lines cut ...]
Jan 16 12:34:48 priyasi kernel: ata2.00: configured for UDMA/133
Jan 16 12:34:48 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Jan 16 12:34:48 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Jan 16 12:34:50 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Jan 16 12:34:50 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Jan 16 12:34:53 priyasi kernel: ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 320)
Jan 16 12:34:53 priyasi kernel: ata1.00: configured for UDMA/133
Jan 16 12:34:53 priyasi kernel: sd 0:0:0:0: [sda] Starting disk
Jan 16 12:34:55 priyasi systemd[1]: Starting Laptop Mode Tools - Battery Polling Service...
Jan 16 12:34:55 priyasi systemd[1]: Reloading Laptop Mode Tools.
Jan 16 12:34:55 priyasi systemd[1]: Started Laptop Mode Tools - Battery Polling Service.
Jan 16 12:34:56 priyasi kernel: ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
Jan 16 12:34:56 priyasi kernel: ata2.00: configured for UDMA/133
Jan 16 12:34:56 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Jan 16 12:34:56 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Jan 16 12:34:56 priyasi laptop_mode[32672]: Laptop mode
Jan 16 12:34:56 priyasi laptop_mode[32672]: enabled, active [unchanged]
Jan 16 12:34:56 priyasi systemd[1]: Reloaded Laptop Mode Tools.
Jan 16 12:34:59 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Jan 16 12:34:59 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Jan 16 12:34:59 priyasi kernel: ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
Jan 16 12:34:59 priyasi kernel: ata2.00: configured for UDMA/133
Jan 16 12:34:59 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Jan 16 12:34:59 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
```

rickysarraf, 2018-02-03:
@alonbl @copyninja
The 1.72 release may uncover this bug for some of the users. If such is the case, please report back on this issue. If this issue is generally seen, we can add device types `AHCI` into runtime-pm default blacklist.

copyninja, 2018-02-03:
Because of this issue my load average started spiking with no command returning I had to force switch off the laptop. When I booted to recovery mode also this issue persisted. Finally I removed laptop-mode-tools with great difficulty. It is not possible to use system with this issue forget opening file to blacklist module

liskin, 2018-02-20:
Isn't this https://www.mail-archive.com/search?l=linux-block@vger.kernel.org&q=subject:%22Re%5C%3A+linux%5C-next+scsi%5C-mq+hang+in+suspend%5C-resume%22&o=newest&f=1 all over again? I saw this a few months ago when I played with block multiqueue in some then new kernel, but found out someone forgot to implement power management into it and their fix (https://patchwork.kernel.org/patch/9891133/) wasn't enough to prevent LMT from hanging my machine. I resorted to "dm_mod.use_blk_mq=0 scsi_mod.use_blk_mq=0".

rickysarraf, 2018-02-21:
I don't think the issues are the same. Our runtime-pm module wasn't executing completely. This got fixed in the last release with PR #118 , which then uncovered this issue.

The issue is that 'sd' type devices get offlined completely, when runtime pm is applied. So this issue should be reproducible even during normal use case. As was such in my tests and from what @copyninja reported. Similar reports were seen from a Debian user on the Debian BTS too.

It could very well be that they are related. But from the mentioned thread, I can't conclude that. As you can see from the logs I've shared, the disks are stopped and never come back.

rickysarraf, 2018-02-21:
So, by the way, does using `dm_mod.use_blk_mq=0 scsi_mod.use_blk_mq=0` and removing `sd` from the runtime-pm.conf blacklist work proper for you ?

liskin, 2018-02-21:
Well it seems this isn't really related to runtime-pm, sorry for the noise, I commented in the middle of debugging. `use_blk_mq=0` doesn't help any more, I had to `CONTROL_AHCI_RUNTIME_PM=0`. Funny thing is that with older kernels, this problem only happened with blk mq, but when I dist-upgraded yesterday, this started happening even with old queue code. Anyway, this is my wife's laptop, so I'm not sure I'll have time to dig deeper. :-/

rickysarraf, 2018-02-22:
You can have `CONTROL_AHCI_RUNTIME_PM=1`. That'll work. You just need to ensure that your runtime-pm blacklist includes: `AUTOSUSPEND_RUNTIME_DEVTYPE_BLACKLIST="sd hub usbhid usb-storage"`

`sd` is the important device class here, that needs to be blacklisted.

liskin, 2018-02-22:
You obviously haven't tried unstable's 4.15.4 yet. :-)

rickysarraf, 2018-02-22:
I am on it, since yesterday.

s3nt fr0m a $martph0ne, excuse typ0s

liskin, 2018-02-22:
And do you see "Stopping disk" immediately followed by "Starting disk" in dmesg every time you pull out the charger? Like 3 or 4 times we experienced it not being followed by "Starting disk". :-(

copyninja, 2018-02-22:
This was exact same thing happened to me but I don't remember kernel
version. It goes on till finally everything freezes terminal looks ok but
none of command returns.

rickysarraf, 2018-02-22:
I have seen that a couple of times since yesterday. My initial thought was that my hardware was going bad. But now that you've reported the same, I have some relief.

I tried a couple of [un]plugs today and couldn't reproduce it. I see this issue is intermittent and it is still unclear of what chain of events uncover this bug.

```
Feb 22 15:55:18 priyasi kernel: rfkill: input handler disabled
Feb 22 15:55:38 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Feb 22 15:55:38 priyasi kernel: sd 0:0:0:0: [sda] Synchronizing SCSI cache
Feb 22 15:55:38 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Feb 22 15:55:38 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Feb 22 15:55:38 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Feb 22 15:55:38 priyasi kernel: sd 0:0:0:0: [sda] Stopping disk
Feb 22 15:55:38 priyasi kernel: EXT4-fs (dm-2): re-mounted. Opts: errors=remount-ro,data=ordered,commit=600
Feb 22 15:55:38 priyasi kernel: BTRFS warning (device loop0): excessive commit interval 600
Feb 22 15:55:38 priyasi kernel: BTRFS info (device loop0): disk space caching is enabled
Feb 22 15:55:38 priyasi kernel: EXT4-fs (sda6): re-mounted. Opts: data=ordered,commit=600
Feb 22 15:55:38 priyasi kernel: BTRFS warning (device sda5): excessive commit interval 600
Feb 22 15:55:38 priyasi kernel: BTRFS info (device sda5): disk space caching is enabled
Feb 22 15:55:39 priyasi kernel: sd 0:0:0:0: [sda] Starting disk
Feb 22 15:55:39 priyasi kernel: EXT4-fs (dm-5): re-mounted. Opts: errors=remount-ro,stripe=32747,data=ordered,commit=600
Feb 22 15:55:39 priyasi kernel: EXT4-fs (dm-5): re-mounted. Opts: errors=remount-ro,stripe=32747,data=ordered,commit=600
Feb 22 15:55:48 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Feb 22 15:55:48 priyasi kernel: NMI watchdog: Enabled. Permanently consumes one hw-PMU counter.
Feb 22 15:55:48 priyasi kernel: EXT4-fs (dm-2): re-mounted. Opts: errors=remount-ro,data=ordered,commit=0
Feb 22 15:55:48 priyasi kernel: BTRFS info (device loop0): using default commit interval 30s
Feb 22 15:55:48 priyasi kernel: BTRFS info (device loop0): disk space caching is enabled
Feb 22 15:55:48 priyasi kernel: EXT4-fs (sda6): re-mounted. Opts: data=ordered,commit=0
Feb 22 15:55:48 priyasi kernel: BTRFS info (device sda5): using default commit interval 30s
Feb 22 15:55:48 priyasi kernel: BTRFS info (device sda5): disk space caching is enabled
Feb 22 15:55:48 priyasi kernel: EXT4-fs (dm-5): re-mounted. Opts: errors=remount-ro,stripe=32747,data=ordered,commit=0
Feb 22 15:55:48 priyasi kernel: EXT4-fs (dm-5): re-mounted. Opts: errors=remount-ro,stripe=32747,data=ordered,commit=0
Feb 22 15:56:02 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Feb 22 15:56:02 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Feb 22 15:56:02 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Feb 22 15:56:02 priyasi kernel: EXT4-fs (dm-2): re-mounted. Opts: errors=remount-ro,data=ordered,commit=600
Feb 22 15:56:02 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Feb 22 15:56:02 priyasi kernel: BTRFS warning (device loop0): excessive commit interval 600
Feb 22 15:56:02 priyasi kernel: BTRFS info (device loop0): disk space caching is enabled
Feb 22 15:56:02 priyasi kernel: EXT4-fs (sda6): re-mounted. Opts: data=ordered,commit=600
Feb 22 15:56:02 priyasi kernel: BTRFS warning (device sda5): excessive commit interval 600
Feb 22 15:56:02 priyasi kernel: BTRFS info (device sda5): disk space caching is enabled
Feb 22 15:56:02 priyasi kernel: EXT4-fs (dm-5): re-mounted. Opts: errors=remount-ro,stripe=32747,data=ordered,commit=600
Feb 22 15:56:02 priyasi kernel: EXT4-fs (dm-5): re-mounted. Opts: errors=remount-ro,stripe=32747,data=ordered,commit=600
Feb 22 15:56:20 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Feb 22 15:56:20 priyasi kernel: NMI watchdog: Enabled. Permanently consumes one hw-PMU counter.
Feb 22 15:56:20 priyasi kernel: EXT4-fs (dm-2): re-mounted. Opts: errors=remount-ro,data=ordered,commit=0
Feb 22 15:56:21 priyasi kernel: BTRFS info (device loop0): using default commit interval 30s
Feb 22 15:56:21 priyasi kernel: BTRFS info (device loop0): disk space caching is enabled
Feb 22 15:56:21 priyasi kernel: EXT4-fs (sda6): re-mounted. Opts: data=ordered,commit=0
Feb 22 15:56:21 priyasi kernel: BTRFS info (device sda5): using default commit interval 30s
Feb 22 15:56:21 priyasi kernel: BTRFS info (device sda5): disk space caching is enabled
Feb 22 15:56:21 priyasi kernel: EXT4-fs (dm-5): re-mounted. Opts: errors=remount-ro,stripe=32747,data=ordered,commit=0
Feb 22 15:56:21 priyasi kernel: EXT4-fs (dm-5): re-mounted. Opts: errors=remount-ro,stripe=32747,data=ordered,commit=0
Feb 22 15:56:28 priyasi kernel: sd 0:0:0:0: [sda] Synchronizing SCSI cache
Feb 22 15:56:28 priyasi kernel: ahci 0000:00:1f.2: port does not support device sleep
Feb 22 15:56:28 priyasi kernel: sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Feb 22 15:56:28 priyasi kernel: sd 1:0:0:0: [sdb] Stopping disk
Feb 22 15:56:28 priyasi kernel: sd 1:0:0:0: [sdb] Starting disk
Feb 22 15:56:28 priyasi kernel: sd 0:0:0:0: [sda] Stopping disk
Feb 22 15:56:28 priyasi kernel: EXT4-fs (dm-2): re-mounted. Opts: errors=remount-ro,data=ordered,commit=600
Feb 22 15:56:28 priyasi kernel: BTRFS warning (device loop0): excessive commit interval 600
Feb 22 15:56:28 priyasi kernel: BTRFS info (device loop0): disk space caching is enabled
[... cut ...]
Feb 22 15:57:04 priyasi kernel: EXT4-fs (sda6): re-mounted. Opts: data=ordered,commit=600
Feb 22 15:57:04 priyasi kernel: BTRFS warning (device sda5): excessive commit interval 600
Feb 22 15:57:04 priyasi kernel: BTRFS info (device sda5): disk space caching is enabled
Feb 22 15:57:04 priyasi kernel: sd 0:0:0:0: [sda] Starting disk
Feb 22 15:57:04 priyasi kernel: EXT4-fs (dm-5): re-mounted. Opts: errors=remount-ro,stripe=32747,data=ordered,commit=600
Feb 22 15:57:04 priyasi kernel: EXT4-fs (dm-5): re-mounted. Opts: errors=remount-ro,stripe=32747,data=ordered,commit=600
```

rickysarraf, 2018-02-22:
@copyninja  What you saw was devices being offlined and never recovering, followed by all disk related processes getting to `DI` state. That issue was resolved by adding `sd` to the blacklist, fixed in 1.72.2 (or in Debian 1.72-2).

liskin, 2018-02-22:
Hm, @rickysarraf you have a Stopping followed by remount followed by Starting. But with an SSD, I thought remounting for commit=600 is unnecessary, so I configured that bit out. Not sure if that's what keeps your disks on and my wife's off, it's just an idea.

liskin, 2018-02-22:
(and just for completeness, my own laptop only has NVMe disks, so that's why I can no longer experiment with any of this locally)

rickysarraf, 2018-02-22:
Well. Holding off the I/O at the file systems layer can be a good thing. I know SSDs are fast and don't have the drawback as in rotational disks, but still a large chunk of sequential writes to the SSD are much better.

I think I documented it in the FAQs.
https://github.com/rickysarraf/laptop-mode-tools/wiki/FAQ#i-have-a-solid-state-disk-ssd-in-my-machine-should-i-enable-any-of-the-disk-related-parts-of-laptop-mode-tools-or-are-they-irrelevant

copyninja, 2018-02-22:
But I've seen them recovering. I could literally hear disk switching off
and on even when it froze completely.

rickysarraf, 2018-02-22:
@liskin The ordering of the messages about Start/Stop/Remount must be mixed. I have 2 disks, one ssd and the other sshd.

rickysarraf, 2018-02-22:
I hit the bug again and it is pretty annoying. I'm sure there must be data loss involved too, when used in combination with LMT. For now I'm going to revert back to 4.14.

@liskin Do you want to report it to the Debian Kernel Team ? I'm curious what best `steps to reproduce` can be provided for this bug.

liskin, 2018-02-22:
well I haven't managed to reproduce it controllably, the only reproducer I have is "let her run around the house with a laptop for half a day" :-(

rickysarraf, 2018-02-23:
[image]

This is the best I've been able to capture so far. Since the disk (or the file system), at this point, is inaccessible, the full trace never gets logged to the journal/syslog. 

And to make things worse, the machine goes into a freeze at this point so I can't even scroll back to the buffer. But it does respond to SysRq. So the kernel is alive and blocked, Where ? Probably the disk is gone, or the file system went haywire.

rickysarraf, 2018-02-23:
[image]
[image]

I tried to extract some more information. The issue is intermittent, as we've talked before. From the looks of it, it seems the disks do recover. But by the time they do, the block/fs is locked.

liskin, 2018-02-23:
Try ssh. I was able to ssh into a machine with nonresponsive GNOME just fine and even sudo su -.

rickysarraf, 2018-07-26:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=904441
and
https://lists.debian.org/debian-kernel/2018/07/msg00425.html

liskin, 2018-07-30:
@rickysarraf https://patchwork.kernel.org/patch/10548975/

l29ah, 2018-11-27:
I have an SSD on Thinkpad X230 on 1.72.2, and sometimes when i plug the power cable in, i observe in dmesg `Stopping disk`, `Starting disk`, and all the disk I/O dies. Is there a workaround for it?

rickysarraf, 2018-11-29:
> I have an SSD on Thinkpad X230 on 1.72.2, and sometimes when i plug the power cable in, i observe in dmesg `Stopping disk`, `Starting disk`, and all the disk I/O dies. Is there a workaround for it?

With `1.72.2`, all defaults should be sane. What kernel are you on ? What distribution ?

l29ah, 2018-11-29:
4.19.0-rc6, also observed with earlier releases; Gentoo.

rickysarraf, 2018-11-29:
In that case I'm not sure. It could be specific to your device. Can you enable `verbose logging` in laptop-mode-tools and set `DEBUG=1` in `runtime-pm.conf` and then try to generate some debug logs ?

The problem was inherited by `sd` device types, which in `1.72.2` have been added to the blacklist. For modern block (NVMe) device types, this issue is not seen, at least for 2 people. Me and @liskin

liskin, 2018-11-29:
Also, https://patchwork.kernel.org/patch/10548975/ has been merged some time ago so blacklisting of `sd` shouldn't be needed with recent kernels...

PF4Public, 2019-04-04:
Hi there, I'm probably experiencing the same or similar issue. 

Recently (after installing LMT) I've noticed that my laptop hangs upon resume when on battery. Through trial and error I found that it takes one switch from AC to battery to make this happen. After that even stopping LMT does not help.

Turns out, LMT do set `/sys/bus/scsi/devices/0:0:0:0/power/autosuspend_delay_ms` unconditionally to `0`. It has not trouble, if `control` is `on`, but should control become `auto`, it for some reason hangs the whole system upon restore. By default `autosuspend_delay_ms` is set to `-1`. With this setting suspending and restoring works fine with AC or battery. For some reason, even if the `sd` is set to blacklist, LMT still put `0` in `autosuspend_delay_ms` , which then hangs the system with `auto` in `control`.

I have added another check for `if ! blacklisted $runtime_device; then USE_DEVICE=1` in order to keep `autosuspend_delay_ms` set to `-1`. Is there any better solution?

Let me know, if this issue should be filled as a separate bug.

rickysarraf, 2019-04-04:
> Hi there, I'm probably experiencing the same or similar issue.
>
> Recently (after installing LMT) I've noticed that my laptop hangs upon resume when on battery. Through trial and error I found that it [...]

You must be having a different issue altogether. What kernel are you on ? The fixes on the kernel side have all been upstream lately.

On my machine, I suspend/resume more than 10 times in a day, without issues.

The best in your case, since you have exact steps to reproduce/trigger the bug, is to enable debug logs and then share them here.

PF4Public, 2019-04-04:
Do I open a separate issue then?

rickysarraf, 2019-04-05:
> Do I open a separate issue then?

Yes, Please. I am inclined to close this issue as the issue on the kernel side seems to have been fixed. But I've kept this open so far, so that others, who face the same issue, can find the information here relevant.

aruslantsev, 2019-07-16:
Hi! 
It seems, that I have the same issue on Gentoo Linux with kernel 5.2.1 and laptop-mode-tools-1.72.2-r1. I have Lenovo Thinkpad X230 with two SATA SSD: Samsung 850 EVO 500GB (/home) and Samsung 860 EVO mSATA 250GB (/, swap). I have messages like 
[21845.388162] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[21845.388845] sd 0:0:0:0: [sda] Stopping disk
[21845.396629] sd 0:0:0:0: [sda] Starting disk
when detaching power cable.

I have this string in runtime-pm.conf:
AUTOSUSPEND_RUNTIME_DEVTYPE_BLACKLIST="sd hub usbhid usb-storage"

https://bugs.gentoo.org/689970

rickysarraf, 2019-07-24:
Yes. It is a kernel side issue. Please refer to the link that @liskin has mentioned.
I don't have `sd` blacklisted anymore and I have even forgotten when last I had seen this issue.
I am surprised you are seeing this issues on 5.xx kernel

closed 2020-01-11 by rickysarraf

BeanieBen9990, 2020-12-22:
I'm having an issue like this on kernel 5.9.x. My separate, unmounted hard drive is being power managed by laptop_mode_tools and won't stop clicking constantly. I see that this may not be an issue with this module/code in particular but I just wanted to add a comment saying that I'm getting these dmesg messages when laptop_mode_tools is set to manage my hard drive along with constant clicking. Setting `hdparm -B [255 254]` and `hdparm -[y Y]` makes it stop but resumes once the power cable is taken out or put back in

rickysarraf, 2020-12-23:
@BeanieBen9990 Not sure why you are seeing this issue. On one of my other machine, where I have `sd` class device, I haven't run into this symptom in quite some time.

While on AC:
```
rrs@chutzpah:~$ sudo laptop_mode status
Mounts:
   /dev/mapper/LocalCrypt-ROOT on / type ext4 (rw,relatime,errors=remount-ro)
   /dev/mapper/LocalVG-EXDATA on /media/rrs/EXDATA type ext4 (rw,relatime)
   /dev/sda1 on /mnt/sda1 type ext4 (rw,relatime)
   /dev/sda3 on /boot type ext4 (rw,relatime)
 
Drive power status:
   
   /dev/sda:
    drive state is:  active/idle
   
   /dev/sdb:
    drive state is:  active/idle
 
(NOTE: drive settings affected by Laptop Mode cannot be retrieved.)
 
Readahead states:
   /dev/mapper/LocalCrypt-ROOT: 128 kB
   /dev/loop0: 128 kB
   /dev/mapper/LocalVG-EXDATA: 128 kB
   /dev/sda1: 128 kB
   /dev/sda3: 128 kB
 
Laptop Mode Tools is allowed to run: /var/run/laptop-mode-tools/enabled exists.
 
/proc/sys/vm/laptop_mode:
   0
 
/proc/sys/vm/dirty_ratio:
   40
 
/proc/sys/vm/dirty_background_ratio:
   10
 
/proc/sys/vm/dirty_expire_centisecs:
   3000
 
/proc/sys/vm/dirty_writeback_centisecs:
   500
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu2/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu3/cpufreq/scaling_governor:
   schedutil
 
/proc/acpi/button/lid/LID/state:
   state:      open
 
/sys/class/scsi_host/host0/link_power_management_policy:
   max_performance
 
/sys/class/scsi_host/host1/link_power_management_policy:
   max_performance
 
/sys/class/scsi_host/host2/link_power_management_policy:
   max_performance
 
/sys/class/scsi_host/host3/link_power_management_policy:
   max_performance
 
/sys/class/power_supply/AC/online:
   1
 
/sys/class/power_supply/BAT0/status:
   Unknown
 
13:15 ♒♒♒   ☺    
```

While on BATT:
```
rrs@chutzpah:~$ sudo laptop_mode status
Mounts:
   /dev/mapper/LocalCrypt-ROOT on / type ext4 (rw,relatime,errors=remount-ro,commit=600)
   /dev/mapper/LocalVG-EXDATA on /media/rrs/EXDATA type ext4 (rw,relatime,commit=600)
   /dev/sda1 on /mnt/sda1 type ext4 (rw,relatime,commit=600)
   /dev/sda3 on /boot type ext4 (rw,relatime,commit=600)
 
Drive power status:
   
   /dev/sda:
    drive state is:  active/idle
   
   /dev/sdb:
    drive state is:  active/idle
 
(NOTE: drive settings affected by Laptop Mode cannot be retrieved.)
 
Readahead states:
   /dev/mapper/LocalCrypt-ROOT: 3072 kB
   /dev/loop0: 3072 kB
   /dev/mapper/LocalVG-EXDATA: 3072 kB
   /dev/sda1: 3072 kB
   /dev/sda3: 3072 kB
 
Laptop Mode Tools is allowed to run: /var/run/laptop-mode-tools/enabled exists.
 
/proc/sys/vm/laptop_mode:
   2
 
/proc/sys/vm/dirty_ratio:
   60
 
/proc/sys/vm/dirty_background_ratio:
   1
 
/proc/sys/vm/dirty_expire_centisecs:
   60000
 
/proc/sys/vm/dirty_writeback_centisecs:
   60000
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_max_freq:
   2700000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_min_freq:
   800000
 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu2/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu3/cpufreq/scaling_governor:
   schedutil
 
/proc/acpi/button/lid/LID/state:
   state:      open
 
/sys/class/scsi_host/host0/link_power_management_policy:
   med_power_with_dipm
 
/sys/class/scsi_host/host1/link_power_management_policy:
   med_power_with_dipm
 
/sys/class/scsi_host/host2/link_power_management_policy:
   med_power_with_dipm
 
/sys/class/scsi_host/host3/link_power_management_policy:
   med_power_with_dipm
 
/sys/class/power_supply/AC/online:
   0
 
/sys/class/power_supply/BAT0/status:
   Discharging
 
13:17 ♒♒♒   ☺    
```