rickysarraf/laptop-mode-tools #124: wrong gui path

Fabioamd87, 2018-02-04:
Hello,

I've just installed Laptop Mode Tools 1.72 by the ./install script on ubuntu 17.10 64bit but when i run lmt-config-gui I get:

```
fabio@a514:/usr/share/laptop-mode-tools$ lmt-config-gui
python3: can't open file '/usr/share/laptop-mode-tools/lmt.py': [Errno 2] No such file or directory
```

```
fabio@a514:/usr/share/laptop-mode-tools$ ls
LMT.py  module-helpers  modules
fabio@a514:/usr/share/laptop-mode-tools$
```
solved with a:

`fabio@a514:/usr/share/laptop-mode-tools$ sudo mv LMT.py lmt.py`

is there an uppercase mistake somewhere?

rickysarraf, 2018-02-05:
Thank you for the report. Yes. There was  a case sensitive naming mistake.

closed 2018-02-05 by rickysarraf

FrostKnight, 2019-06-01:
I just had the same problem, thank you for that solution fabio!